gnss-sdr/gnss-sdr #575: Compilation error when building on the main branch

agrenier96, 2021-12-15:
Hello everyone,
I am new to GNSS-SDR, I am trying to build the package from sources to later use my RTL-SDR V3 dongle with it in real time. Right now I am trying to build to basic package on the main branch first. 

I followed the steps available on the Github and website page, adding the listing references first. The CMake correctly state that the code is ready to be built (see attached log file). However, when I run the make command, it get stuck on step:
``[ 58%] Building CXX object src/algorithms/libs/CMakeFiles/algorithms_libs.dir/pass_through.cc.o``

And the following error is highlighted. You can find the complete log in the attached files.
`error: could not convert ‘((Pass_Through*)this)->Pass_Through::conjugate_cc_’ from ‘conjugate_cc_sptr’ {aka ‘boost::shared_ptr<conjugate_cc>’} to ‘gr::basic_block_sptr’ {aka ‘std::shared_ptr<gr::basic_block>’}`

I am working on Windows 10, compiling using the Windows Subsystem Linux version 2, with an Ubuntu 20.04. 

As I couldn't find any related issue on the Github, I am wandering what can cause this error? Please let me know if you need additional information. Thank you for your help.

[log_cmake.txt](https://github.com/gnss-sdr/gnss-sdr/files/7718128/log_cmake.txt)
[log_build.txt](https://github.com/gnss-sdr/gnss-sdr/files/7718133/log_build.txt)

jwmelto, 2021-12-15:
Hi @agrenier-gnss .

The error message is quite informative, if you can navigate through all the detail. The key part is this:

`error: could not convert … ‘boost:shared_ptr<…>’ to … ‘std::shared_ptr<…>’`

What version of GNU Radio do you have installed? 3.7 is no longer supported, and 3.8 is end-of-life.  The newer versions use `std::shared_ptr`, but it looks like your GNSS-SDR is configured to use `boost::shared_ptr`. These two template types are not interchangeable.

It’s usually a good idea to start with the **next** branch, since that branch is updated more often.

carlesfernandez, 2021-12-19:
Dear @agrenier-gnss,

I've reproduced the build of the main branch in an ubuntu:focal docker image, and it seems to work well. Were you starting from a fresh build? CMake should detect that you are using GNU Radio v3.8.1 and select the appropriate type for GNU Radio blocks.

agrenier-gnss, 2021-12-20:
Dear @jwmelto, @carlesfernandez,

I think I found out the issue. The GNU Radio library recently pushed a 3.10 version on the APT package branch. I compiled GNU Radio 3.8 from source and installed it, reset my build folder and it seems fixed now. 

Thank you for your answers and sorry for the late reply. I believe this issue is closed now.

closed 2021-12-20 by agrenier-gnss