gnss-sdr/gnss-sdr #578: Building on M1 Mac using the 'apple-m1' branch

mmtaksuu, 2021-12-25:
Hi everyone, I have currently been trying to install gnss-sdr by building from the source code. When I try to build the gnss-sdr the compilation finished successfully. But when I try to run the code it couldn't run.

I have an issue about dynamic libraries couldn't loaded. 
How can I solve the problem. Is there any way to solve the problem?

[image]

[image]

jwmelto, 2021-12-27:
Did you install lapack using Homebrew? If so, the project will need modification because on M1 Macs Homebrew installs in /opt/homebrew, not /usr/local. See the FAQ <https://docs.brew.sh/FAQ#why-is-the-default-installation-prefix-opthomebrew-on-apple-silicon> for more detail.

Also, you can look at https://github.com/BZFlag-Dev/bzflag/pull/299/files <https://github.com/BZFlag-Dev/bzflag/pull/299/files> for a PR on another project dealing with potentially the same issue.
--
Jim Melton

mmtaksuu, 2021-12-27:
Hi, Thanks for your answer first. 
But I am confusing to change the default installation location of homebrew. How can I change it for some specific packages. How can I do this? I will be happy if you recommend me any way to do this for running gnss-sdr on my computer.

Thank you so much, in advance.

jwmelto, 2021-12-27:
You don't want to change the default install location for Homebrew, but you do need to search there. 

First, you mentioned an `apple-m1` branch. I don't see that branch here, so I'm not sure it is supported. 

Second, your Xcode project needs to be modified to search in `/opt/homebrew` for include files and libraries. There are two places where this needs to be done. The PR I linked for you shows how. I recommend putting `/opt/homebrew` before `/usr/local` since the M1 has an Intel emulation capability. 

If you are trying to port to M1, these are important things to understand. If there is already a port, I'd talk to the people who did it.