gnss-sdr/gnss-sdr #579: Pseudoterminal issue

sv1, 2022-01-04:
I am trying to use gnss-sdr with gpsd on a Debian Linux 11

For that I have to create a pseudoterminal. I have set the following options on my config file

PVT.flag_nmea_tty_port=true
PVT.nmea_dump_devname=/dev/pts/3

but /dev/pts/3 is never created.

pts/3 does not exist either before or after starting gnss-sdr.

minecraft2048, 2022-03-16:
Bumping this, also interested in getting gnss-sdr to work with gpsd

sv1, 2022-07-16:
I do not know on which distribution you are on, but on Debian 11:

$ sudo gpsd -b -n -N /dev/pts/3
gpsd:ERROR: SER: device open of /dev/pts/3 failed: Permission denied - retrying
read-only
gpsd:ERROR: SER: read-only device open of /dev/pts/3 failed: Permission denied
gpsd:ERROR: initial GPS device /dev/pts/3 open failed
gpsd:ERROR: can't run with neither control socket nor devices open

On Wed, 2022-05-18 at 11:40 -0700, vladisslav2011 wrote:
> Hello.
> 1. Create a pair of pseudoterminals:
> ```
> socat -d -d pty,raw,echo=0 pty,raw,echo=0
> ```
> It should output something like
> ```
> 2022/05/18 21:29:06 socat[17517] N PTY is /dev/pts/26
> 2022/05/18 21:29:06 socat[17517] N PTY is /dev/pts/27
> 2022/05/18 21:29:06 socat[17517] N starting data transfer loop with FDs [5,5]
> and [7,7]
> ```
> 2. Add/edit gnss-sdr config file
> ```
> PVT.flag_nmea_tty_port=true
> PVT.nmea_dump_devname=/dev/pts/26
> ```
> 3. Run gnss-sdr with your config file
> 4. Run gpsd
> ```
> gpsd -b -n -N -D 255 /dev/pts/27
> ```
> 5. Run xgps or any other gpsd client
> 6. It works!
>

vladisslav2011, 2022-07-16:
Any distribution should work. You may have to install socat first:

```
sudo apt-get install socat
```

Have you done all steps?
If so, paste here an output from all steps, not the last one.
Every step should be done in a separate virtual terminal. Use xterm/gnome-terminal/konsole/etc. One window/tab per step. Do not close any terminal window until you are done with GNSS-SDR.
gpsd does not require sudo. It runs well under unprivileged user.

sv1, 2022-09-01:
I wish things where so easy!

In most linux distributions gpsd is constrained by Apparmor. gpsd apparmor
profile does not allow access to /dev/pts/*. So in order to access anything else
except a physical serial port you have change gpsd apparmor profile.

OR

you can use a "device" that already is included in gpsd apparmor profile:

# mkfifo /tmp/gpsfake-0.sock

set 

PVT.nmea_dump_devname=/tmp/gpsfake-0.sock

start gnss-sdr and gpsd

Apostolos

On Sat, 2022-07-16 at 15:15 -0700, vladisslav2011 wrote:
> Any distribution should work. You may have to install socat first:
>
> ```
> sudo apt-get install socat
> ```
>
> Have you done all steps?
> If so, paste here an output from all steps, not the last one.
> Every [...]