gnss-sdr/gnss-sdr #588: GPS L5 health status issue?

jmfriedt, 2022-02-24:
I am addressing real time processing of GPS L5 signals collected from an XTRX embedded SDR board. Whenever gnss-sdr locks on the L5 signal, I get
```
New GPS L5 CNAV message received in channel 4: ephemeris from satellite GPS PRN 10 (Block IIF)
Satellite GPS PRN 10 (Block IIF) is not healthy, not used for navigation
New GPS L5 CNAV message received in channel 2: ephemeris from satellite GPS PRN 24 (Block IIF)
Satellite GPS PRN 24 (Block IIF) is not healthy, not used for navigation
```
Considering that so many unhealthy satellites is suspicious, I fetched an old record from December 2020 that was successfully decoded at the time (was version 0.0.13 of gnss-sdr at the time) and ran the same configuration file on this same record and surely enough this time the PVT solution failed with the same
```
New GPS L5 CNAV message received in channel 1: ephemeris from satellite GPS PRN 03 (Block IIF)
Satellite GPS PRN 03 (Block IIF) is not healthy, not used for navigation
New GPS L5 CNAV message received in channel 4: ephemeris from satellite GPS PRN 09 (Block IIF)
Satellite GPS PRN 09 (Block IIF) is not healthy, not used for navigation
```
when processed with v0.0.15 (cloned Jan. 2020). I am not familiar with the NAV messages transmitted by GPS but it seems to me that
```
commit d80d33ad915da76d7fb6655a92749b8160e31fca
Author: Carles Fernandez <carles.fernandez@gmail.com>
Date:   Sat Jul 24 10:21:25 2021 +0200
    Do not use satellites marked as not healthy in the nav message
```
erroneously removes L5 messages that should be kept in ``algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc``
Thanks.

jmfriedt, 2022-02-24:
Replacing in src/algorithms/PVT/gnuradio_blocks/rtklib_pvt_gs.cc line 2026 from
```cpp
if ((prn_aux == in[i][epoch].PRN) && (((std::string(in[i][epoch].Signal) == std::string("2S")) || (std::string(in[i][epoch].Signal) == std::string("L5"))) && (tmp_eph_iter_cnav->second.signal_health == 0)))
```
to
```cpp
if ((prn_aux == in[i][epoch].PRN) && (((std::string(in[i][epoch].Signal) == std::string("2S")) && (tmp_eph_iter_cnav->second.signal_health == 0)) || (std::string(in[i][epoch].Signal) == std::string("L5"))))
```
recovers the ability to decode the L5 signal (tested on the same record mentioned in the first issue description). However the unhealthy satellite message remains: we are not clear whether this part of the issue should be handled at the flag setting level (not setting the flag decoded in the message in case of L5 decoding since this flag seems meaningless from the advisory cited earlier) or at the message display level.

carlesfernandez, 2022-02-24:
Thanks for spotting this issue. Since we usually perform tests against synthetic signals, this didn't show up.

The unhealthy satellite message (and the condition at rtklib_pvt_gs.cc) are there to avoid using satellites for PVT that are reporting themselves as unhealthy. Here it seems that is a matter of interpretation of the message. In the advisory note you sent, it says:

```
 THE CNAV MESSAGE HEALTH BITS MAY FUNCTION INDEPENDENTLY FROM THE LEGACY MESSAGE HEALTH BITS AND SHOULD NOT BE CONSIDERED RELIABLE INDICATORS OF SIGNAL QUALITY OR PERFORMANCE DURING PRE-OPERATIONAL USE.
```

And today, at https://www.gps.gov/systems/gps/modernization/civilsignals/, it reads:

```
However, L5 remains pre-operational and should be employed at the user's own risk until it is declared operational.
```

so it seems that we shouldn't discard L5 signals based on their health flag status, as you did in rtklib_pvt_gs.cc

jmfriedt, 2022-02-24:
So should we leave the unhealthy message statement as an indication to the gnss-sdr user that L5 is not yet officially commissioned?
Thanks

noelpedro, 2022-03-29:
@jmfriedt you mind sharing your L5 config file?

currently not able to track with FPGA offloading.

thanks

jmfriedt, 2022-03-30:
This is my configuration file for an XTRX board running the whole processing on the general purpose CPU, no offloading. Indeed with this setup I could reach reliably decoding of NAV messages from 4 satellites but no PVT solution, which I assumed might be due to a poor constellation geometry considering the few satellites transmitting L5.
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=11230000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=11230000
SignalSource.freq=1176450000
SignalSource.osmosdr_args=xtrx,dev=/dev/xtrx0,loglevel=0 ; ,extclk=21408000
SignalSouce.antenna=RXH
SignalSource.gain_mode=manual
SignalSource.dump=false
SignalSource.gain=63
SignalSource.AGC_enabled=false
;SignalSource.dump_filename=../data/signal_source.dat
;SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############ 
SignalConditioner.implementation=Pass_Through

;######### SIGNAL_ ############ 

;######### CHANNELS GLOBAL CONFIG ############ https://gnss-sdr.org/docs/sp-blocks/channels/
Channels_L5.count=05                    ;optional, default 0 : number of channes targeting GPS LA C/A signals
Channels.in_acquisition=1               ;optional, default to the total number of channels : Maximum number of channels performing signal acquisition at the same time. The recommended value is 1.
                                        ; In the case of having assigned a channel to a given satellite (e.g., with Channel0.satellite=1), it is recommended to increase this number in order to 
                                        ; always have at least one channel searching for new satellites
Channel.signal=L5
;######### ACQUISITION GLOBAL CONFIG ############ https://gnss-sdr.org/docs/sp-blocks/acquisition/
Acquisition_L5.implementation=GPS_L5i_PCPS_Acquisition  ;mandatory
Acquisition_L5.item_type=gr_complex                             ;optional, default gr_complex : set the sample data type expected at the block input
Acquisition_L5.threshold=2.75                                   ;optional, default 0.0 : Decision threshold gama from which a signal will be considered present
Acquisition_L5.doppler_max=12000                                ;optional, default 5000 Hz : maximum doppler value in the search grid
Acquisition_L5.doppler_step=500                                 ;optional, default 500 Hz : frequency step in the search in the search grid

;######### TRACKING GLOBAL CONFIG ############ https://gnss-sdr.org/docs/sp-blocks/tracking/
Tracking_L5.implementation=GPS_L5_DLL_PLL_Tracking      ;mandatory
Tracking_L5.item_type=gr_complex                        ;optional, default gr_complex : set the sample data type expected at the block input
Tracking_L5.pll_bw_hz=20.0;                             ;optional, default 50 Hz : bandwith of the PLL low pass filter
Tracking_L5.dll_bw_hz=4.0;                              ;optional, default 2Hz : bandwith of the DLL low pass filter

;######### TELEMETRY DECODER GPS CONFIG ############ https://gnss-sdr.org/docs/sp-blocks/telemetry-decoder/
TelemetryDecoder_L5.implementation=GPS_L5_Telemetry_Decoder     ;mandatory

;######### OBSERVABLES CONFIG ############ https://gnss-sdr.org/docs/sp-blocks/observables/
Observables.implementation=Hybrid_Observables   ;mandatory

;######### PVT CONFIG ############ https://gnss-sdr.org/docs/sp-blocks/pvt/
PVT.implementation=RTKLIB_PVT   ;mandatory
PVT.averaging_depth=100         ;non trouvé
PVT.flag_averaging=true         ;non trouvé
PVT.output_rate_ms=10           ;optional, default 500ms must be multiple of 20 ms : rate at which PVT solutions will be computed
PVT.display_rate_ms=500         ;optional, default 500ms, must be multiple of output_rate_ms : rate at:
```

noelpedro, 2022-03-30:
@jmfriedt Its possible to get an L5 only solution? Thanks for sharing the config.

jmfriedt, 2022-03-30:
It must be possible since with post-processing I managed to obtain some time ago (Oct 10 2020 from the file creation date)
```
Current receiver time: 1 min 43 s
Current receiver time: 1 min 44 s
Current receiver time: 1 min 45 s
New GPS L5 CNAV message received in channel 2: ephemeris from satellite GPS PRN 04 (Block U
nknown)
Current receiver time: 1 min 46 s
Current receiver time: 1 min 47 s
Current receiver time: 1 min 48 s
Current receiver time: 1 min 49 s
Current receiver time: 1 min 50 s
Current receiver time: 1 min 51 s
New GPS L5 CNAV message received in channel 4: ephemeris from satellite GPS PRN 09 (Block IIF)
New GPS L5 CNAV message received in channel 1: ephemeris from satellite GPS PRN 03 (Block IIF)
Current receiver time: 1 min 52 s
New GPS L5 CNAV message received in channel 0: ephemeris from satellite GPS PRN 01 (Block IIF)
First position fix at 2020-Oct-10 11:26:19.400000 UTC is Lat = 47.2514 [deg], Long = 5.99311 [deg], Height= 425.513 [m]
Position at 2020-Oct-10 11:26:19.500000 UTC using 4 observations is Lat = 47.251351988 [deg], Long = 5.993112379 [deg], Height = 425.016 [m]
```
and the SDR (B210 receiver in this case) does not have enough bandwidth to collect multiple GPS bands. This result was achieved with GNSS-SDR v0.0.13 though and I cannot comment with the latest release which has so far failed to provide a position, but I have not reproduced the experiment of storing the raw IQ files and post-processing with the latest gnss-sdr, so I cannot separate poor constellation geometry from inability of gnss-sdr to converge on a L5 PVT solution or the XTRX LMS7002 providing lower gain than the B210 AD9361 frontend. I have only tried real time decoding, and between switch from GNU Radio 3.7 to 3.9 or 3.10, evolution of VOLK with its updated SIMD detection mechanism, and updates to gnss-sdr itself, I am unable to separate various causes of failure to lock on the L5 solution. The beauty of opensource ... everything changing all the time and consistent results being challenged all the time :( But keep up with the amazing work anyway.

noelpedro, 2022-03-30:
Nice! Thats super useful if L1 and L2 bands are jammed.

noelpedro, 2022-03-30:
That position fix with L5 is with track_pilot=false correct?

jmfriedt, 2022-03-31:
Since this option is not defined in the configuration file, I assume it is set to its default value which from https://gnss-sdr.org/docs/sp-blocks/tracking/ seems to be True. So this option seems to be enabled by default, although I have no idea about its potential impact on L5 decoding. This same documentation cites E5 so I am not sure this option has any impact on L5.

ZiglioUK, 2022-03-31:
> I cannot separate poor constellation geometry from inability of gnss-sdr to converge on a L5 PVT solution

It's actually rtk-lib that fails silently. It's impossible to know when that happens. It certainly not printed to the screen, it's in the log files

jwmelto, 2022-03-31:
For debugging, I find it helpful to add the `--logtostderr` command line option. That causes the log to be interleaved with console output.

Also, I added a config option to expose the RTK debug [`.rtk_trace_level`](https://github.com/gnss-sdr/gnss-sdr/blob/832759e9e55590b436a65b09766bdc308be2f015/src/algorithms/PVT/adapters/rtklib_pvt.h#L58) in the PVT config. That might prove helpful?

noelpedro, 2022-04-05:
Getting L5 nav messages PRN 8 and 14 both unhealthy.

You try an E5a only position fix?

commit b7f8a7d5 (2022-02-26): Compute PVT solutions when using GPS L5 signals even if the satellite is reported as not healthy in the CNAV message (Fixes: #588)
closed 2022-04-20 by carlesfernandez