gnss-sdr/gnss-sdr #590: Multiple channels track the same satellite

dryairship, 2022-03-15:
I set the number of channels to 31, and maximum number of channels with concurrent signal acquisition also to 31. I expected that each channel will try to track a different satellite. However, the logs reveal that multiple channels start tracking the same satellite.

Relevant part of the config file:
```
;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=31
Channels.in_acquisition=31
```

Relevant part of the logs:
```
Current receiver time: 2 min 36 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 23 (Block III)
Current receiver time: 2 min 37 s
Tracking of GPS L1 C/A signal started on channel 18 for satellite GPS PRN 30 (Block IIF)
New GPS NAV message received in channel 17: subframe 4 from satellite GPS PRN 10 (Block IIF)
New GPS NAV message received in channel 0: subframe 4 from satellite GPS PRN 10 (Block IIF)
New GPS NAV message received in channel 2: subframe 4 from satellite GPS PRN 23 (Block III)
Current receiver time: 2 min 38 s
GPS L1 C/A tracking bit synchronization locked in channel 6 for satellite GPS PRN 23 (Block III)
Enabled 10 ms extended correlator in channel 6 for satellite GPS PRN 23 (Block III)
Current receiver time: 2 min 39 s
Current receiver time: 2 min 40 s
Loss of lock in channel 25!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 25 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 21 for satellite GPS PRN 23 (Block III)
Current receiver time: 2 min 41 s
Current receiver time: 2 min 42 s
Loss of lock in channel 28!
Current receiver time: 2 min 43 s
Tracking of GPS L1 C/A signal started on channel 14 for satellite GPS PRN 22 (Block IIR)
Loss of lock in channel 3!
New GPS NAV message received in channel 0: subframe 5 from satellite GPS PRN 10 (Block IIF)
New GPS NAV message received in channel 17: subframe 5 from satellite GPS PRN 10 (Block IIF)
New GPS NAV message received in channel 2: subframe 5 from satellite GPS PRN 23 (Block III)
New GPS NAV message received in channel 6: subframe 5 from satellite GPS PRN 23 (Block III)
```

As can be seen, at `2 min 37 s`, satellite 10 is being tracked by 2 channels (0 and 17), and satellite 23 is being tracked by channel 2. Then at `2 min 38 s`, channel 6 also locks to satellite 23, and we see that at `2 min 43 s`, channels 0 and 17 are tracking satellite 10, and channels 2 and 6 are tracking satellite 23.

ZiglioUK, 2022-03-15:
There's no point in setting the number of channels to more than 12 if you're only tracking GPS. Also 1 acquisition channel is enough.

Check that your acquisition threshold is not set too low. Sometimes that happens due to false acquisitions. There's no global check that makes sure the same satellite is not being tracked by two channels at once as far as I know.
The only way to avoid this problem is really to hardcode each channel to track a specific satellite.

dryairship, 2022-03-16:
@ZiglioUK 

> There's no point in setting the number of channels to more than 12 if you're only tracking GPS. Also 1 acquisition channel is enough.

I agree with the first part, I was just experimenting. But I get the same issue when I set both Channels_1C.count and Channels.in_acquisition to 12. So it probably happens whenever the number of acquisition channels is >1.

> Check that your acquisition threshold is not set too low

How can I do that (and what would be the optimal acquisition threshold)?

ZiglioUK, 2022-03-16:
> How can I do that (and what would be the optimal acquisition threshold)?

There are a couple of ways, it's either
`Acquisition_1C.pfa`
or
`Acquisition_1C.threshold`

If your antenna is outside with a clearview of the sky, I normally set the threshold to 2.5.