gnss-sdr/gnss-sdr #592: Acquisiton and Tracking

truff77, 2022-03-27:
Hi all, I'm pretty new to GNSS and I'm having the "Loss of lock" message. I just want to ensure I understand correctly the flow of operation. 
When I get this king of message (bellow), does it means that the acquisition block has detected a satellite in the signal and passed the doppler shift and phased shift to the tracking component ? 
"Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 08 (Block IIF)"

"Loss of lock" resulting from the Tracking module ? 

thank you in advance.

jwmelto, 2022-03-27:
That's pretty much it.

This doesn't seem an "issue" as much as a general question? I can't promise any better response, but you might also consider the [mailing list](https://lists.sourceforge.net/lists/listinfo/gnss-sdr-developers) or our [Matrix channel](https://matrix.to/#/#gnss-sdr:gnuradio.org) for questions.

Not an expert, by any means, but "Loss of lock" can be an unstable timing source in your front end, dropped data, or some other signal quality issues.

truff77, 2022-03-28:
Thank you @jwmelto ! 
You are right, this is not an issue, I'll use the mailing and/or Matrix channel for other questions.

truff77, 2022-05-17:
Hi all,
yes I did solve my issue. It was actually a mistake I did. The signal I was working on was recorded at a center frequency of 2600 Mhz when the GPS L1 signal is at 1 575,42 MHz. 
I had to use InputFilter.IF=-580000 in the InputFilter section to shift the frequency on the right L1 band.