gnss-sdr/gnss-sdr #595: BladeRF 2.0 Set up to get Gps location

KimlongSeng, 2022-04-14:
Hi,
I have complete setup bladeRF with gnss-sdr, I want to get current GPS location(lat-lon), but can't seem to get it working. It printout tracking of GPS L1 but no new position(not like when I do the "my first fix position guide by gnss-sdr"). I am new to this, and really appreciate any help.

-Bladerf config bash file:
#########################
# Use this to load fpga via bladeRF-cli -i, if not using flash to auto-load 
#fpga /home/donkelly/bladeRF/firmwareLoads/hostedxA4. rbf

# Run the following on Linux to auto-load flash fpga 
# bladerf-cli -L /home/donkelly/bladeRF/firmwareLoads/hostedxA4. rbf

# Set basic parms
bladeRF-cli -e 'set frequency rx 1575.42e6'
bladeRF-cli -e 'set frequency tx 47e6' 
bladeRF-cli -e 'set agc off'
bladeRF-cli -e 'set samplerate 4e6' 
# Not sure following helps 
bladeRF-cli -e 'set bandwidth 3e6' 

# Set gains 
# lnagain can be bypass, 0, 3, or 6 
# rxvgal can be 5 to 30 
# rxvga2 can be o to 30 
#lnagain 3 
#rxvga1 30 
#rxvga2 30 
bladeRF-cli -e 'set gain rx 48' 
bladeRF-cli -e 'set gain tx 0' 

# Perform calibrations 30 
# bladeRF-cli -e 'set cal lms 
# bladeRF-cli -e 'set caldc rx 

# Enable external 10M Hz reference clock signal
bladeRF-cli -e 'set clock_ref enable'

# Print configuration 
#bladeRF-cli -e 'print' 

# Enable bias-tee for rx, enable with either true 
# For some reason, must run this last in this script to keep it enabled. 
bladeRF-cli -e 'set biastee rx on'
#########################

-My Gnss-sdr conf (w/gnss-monitor) file:
#########################################
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000
;# mod/edit here
GNSS-SDR.telecommand_enabled=true
;GNSS-SDR.telecommand_tcp_port=3333
GNSS-SDR.telecommand_tcp_port=8888
;##############

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.rf_gain=40
SignalSource.if_gain=30
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
;# Next line enables the internal HackRF One bias (3.3 VDC)
;#SignalSource.osmosdr_args=hackrf,bias=1
;###original
;SignalSource.osmosdr_args=bladerf,biastee=1	
;####

;## mod
SignalSource.osmosdr_args=bladerf,biastee=rx
;###

SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.early_late_space_chips=0.5
Tracking_1C.early_late_space_narrow_chips=0.15
Tracking_1C.pll_bw_hz=40
Tracking_1C.dll_bw_hz=2.0
Tracking_1C.pll_bw_narrow_hz=5.0
Tracking_1C.dll_bw_narrow_hz=1.50
Tracking_1C.fll_bw_hz=10
Tracking_1C.enable_fll_pull_in=true
Tracking_1C.enable_fll_steady_state=false
Tracking_1C.dump=false
Tracking_1C.dump_filename=tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
;# edit/modify
PVT.enable_monitor=true
PVT.monitor_client_addresses=127.0.0.2
PVT.monitor_udp_port=1113
;###########
PVT.implementation=RTKLIB_PVT

;PVT.positioning_mode=Single
PVT.positioning_mode=PPP_Static

PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
;#add
PVT.rtcm_MT1045_rate_ms=5000
PVT.rtcm_MT1097_rate_ms=1000
;##
PVT.rinex_version=2

;#MONITOR CONFIG#
Monitor.enable_monitor=true
Monitor.decimation_factor=1
Monitor.client_addresses=127.0.0.2
Monitor.udp_port=1114
#################################################

-output: 

gnss-sdr --config_file=/home/cild/Documents/Hackrf-position/position.conf 
Initializing GNSS-SDR v0.0.16.git-next-f38cfb7e4 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf,bias=1
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio v3.8.5.0-6-g57bd109d
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2018.01.1
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
Actual RX Gain: 14 dB...
Actual Bandwidth: 6e+06 [Hz]...
RF Channels: 1
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Current receiver time: 1 s
Current receiver time: 2 s
Current receiver time: 3 s
Current receiver time: 4 s
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 26 (Block IIF)
......
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 18 (Block III)

etc....
################################################################

vladisslav2011, 2022-05-02:
Hello.
You have internal sampling rate set to 2M:

> GNSS-SDR.internal_fs_sps=2000000

But your BladeRF sampling rate is set to 4M:

> SignalSource.sampling_frequency=4000000

And input filter decimation is not enabled:

> InputFilter.decimation_factor=1

Try setting either `InputFilter.decimation_factor` to 2 or `GNSS-SDR.internal_fs_sps` to 4000000 or `SignalSource.sampling_frequency` to 2000000. 
Good luck.

marcospanghero, 2022-06-22:
Also, I am not sure the amplifier passes through the BiasT, you might want to remove it or add one in series after the amp. check the schematic

ark-1313, 2022-09-23:
Hi,
I too am trying with the same code KimLongSen, but unable to get any lock & with the same setup. Any updates ?

ark-1313, 2022-09-23:
Was able to get the position. It was the issue that the antenna was not able to get sufficient LOS