gnss-sdr/gnss-sdr #596: Error while loading shared libraries

cipriancircu, 2022-04-21:
Hello.
I recently built gnss-sdr on the latest available version (0.0.17). I can run it from the terminal, but if I close the terminal and re-open it I’m being given the following message: “gnss-sdr: error while loading shared libraries: libgnuradio-uhd.so.3.8.5git:can not open shared object file: No such file or directory”. 
I have to mention that I run the tool in the same way that I was doing on previous version.
Since I couldn’t find this message posted anywhere, I was hoping that anyone could point out a possible solution. 
Thank you very much!

carlesfernandez, 2022-04-21:
Has anything changed from the successful run until you opened a new terminal? It seems a change in the environment. Are you using PyBOMBS, maybe?

carlesfernandez, 2022-04-21:
You need to set the PyBOMBS environment variables when you open a new terminal, before running gnss-sdr (or anything else installed with PyBOMBS)

```
source ~/{base_folder}/setup_env.sh
```

cipriancircu, 2022-04-26:
Thank you. This has fixed the error. 
I think this thread can be closed now.
Hopefully, it will be of use to others!