gnss-sdr/gnss-sdr #603: Acquisition and tracking dump problems

flyinhawk, 2022-05-07:
I come across various problems while using GNSS-SDR regarding dump files.

**1-)**Acquisition dump files are not created. 
_Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 02 (Block IIR)
Unable to create or open Acquisition dump file_
[image]

**2-)** Tracking, telemetry and observable files are created as .dat files but I should be able to create them as .mat files as it is stated on the website.
[image]
I wrote following commands for tracking on config file but it is still generated as .dat files.
_Tracking_1C.dump=true
Tracking_1C.dump_mat=false_

I want to try to read 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat file, therefore I copied and configured config file on https://gnss-sdr.org/my-first-fix/

You can see the config file below
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/media/user/gnss_sdr/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat
SignalSource.item_type=ishort
SignalSource.sampling_frequency=4000000
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=4000000
Resampler.sample_freq_out=2000000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=8
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250
Acquisition_1C.blocking=true
Acquisition_1C.dump=true
Acquisition_1C.dump_filename=acq_spain
Acquisition_1C.dump_channel=0

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;
Tracking_1C.dump=true
Tracking_1C.dump_mat=true
Tracking_1C.dump_filename=./trk_spain_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=true
TelemetryDecoder_1C.dump_mat=true
TelemetryDecoder_1C.dumo_filename=tele_spain_

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true
Observables.dump_mat=true
Observables.dump_filename=obs_spain

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=true
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

fener1efes2, 2022-06-19:
I have the same problem

Imtiaz08, 2023-07-06:
any progress with that? please how did you fix it if you have already found a solution

carlesfernandez, 2024-12-05:
We have verified that the dump files work well with the current code base (`next` branch). Maybe it was a permission problem?