gnss-sdr/gnss-sdr #609: Configuration file for GPS L5

MohsenNSh, 2022-05-29:
Hi guys

I am trying to navigate by the L5 signal of the GPS. Unfotuanality, I can not do it because I think my used configuration is wrong, especially in the Input filter, acquisition, and tracking blocks. anybody can share a correct configuration for the L5 signal.

MohsenNSh, 2022-06-15:
> This config works for me. [gps5.txt](https://github.com/gnss-sdr/gnss-sdr/files/8817579/gps5.txt) But not every GPS satellite caries an L5 payload. It may be [possible,]([url](url)) that you will not get 4+ satellites in good positions to get a PVT. Galileo or combined Galileo+GPS configurations are more reliable. Galileo config: [gal5.txt](https://github.com/gnss-sdr/gnss-sdr/files/8817603/gal5.txt) Combined GPS+Galileo config: [gal5+gps5.txt](https://github.com/gnss-sdr/gnss-sdr/files/8817606/gal5%2Bgps5.txt)

Thanks for your reply. I have tried over and over to run the L5 dataset that has been locked by NTlab1065. Unfortuanlity. it did not work. To continue, you can see my config. Is it correct?
[GNSS_SDR_L5_NTLab1065.txt](https://github.com/gnss-sdr/gnss-sdr/files/8906874/GNSS_SDR_L5_NTLab1065.txt)

vladisslav2011, 2022-06-22:
Your config file does not look correct.
1. According to NTlab1065 datasheet, the maximum sampling rate at L2 is about 77MHz, but your setting is 99.375MHz. That's wrong. Are you using an external ADC?
2. Freq_Xlating_Fir_Filter should have more taps and about 7MHz passband. You are not cancelling images...
3. GNSS-SDR.internal_fs_sps may be decreased to 10Msps. Use resampler to interpolate to lower sampling rate.

MohsenNSh, 2022-06-25:
> 

> Your config file does not look correct.
>
> 1. According to NTlab1065 datasheet, the maximum sampling rate at L2 is about 77MHz, but your setting is 99.375MHz. That's wrong. Are you using an external [...]

thanks for your reply. the answers of the mentioned issues are as the following:

1- we have used the NTlab1065-USB EVK so that according to its configuration file, the sample rate to acquire the L5 signal on ch4 and config02 is 99375000 Hz. So, I am sure that it is correct.
2-it has been modified. But, it was not effective.
3-it has been modified. But, it was not effective.

vladisslav2011, 2022-06-26:
OK. If you have tuned PLL-A to somewhere between Glonass and GPS L1, then it may be possible to choose PLL-A as ADC clock source and get 99.375MHz clock.
It looks, like you have set PLL B N-divider to 238 and PLL B R-divider to 2 to get 13.550MHz IF.

4. Check your antenna. It should have at least 20db LNA. It should work with HackRF. If you have a HackRF, you may check your antenna using one of config files, I have posted.
5. Have you checked your files? The input file should have samples from CH4 only, 4 samples, packed to 1 byte, but it looks, like NTlab1065-USB EVK software is capturing all channels interleaved by default.

MohsenNSh, 2022-06-26:
> OK. If you have tuned PLL-A to somewhere between Glonass and GPS L1, then it may be possible to choose PLL-A as ADC clock source and get 99.375MHz clock. It looks, like you have set PLL B N-divider [...]

Thanks. It is noteworthy that the PLL B and A are set by the file configurations that have been provided by the NTlab. So, I do not interferer in it.

1-I have used an A43 antenna with 30 dB gain. we have tested before. Please, do not worry about it.
2-It is correct. Please, do not worry about it.

To tackle this problem, we have used an LNA with 30 dB gain after antenna so that it amplifies the range between 1 GHz to 2 GHz. But, it did not work. Do you have any suggestion?