gnss-sdr/gnss-sdr #616: Manual build success but unusable binary on archlinux

rom-burner, 2022-06-04:
Either building 0.0.17 manually on Archlinux or using the git built package from AUR, both result in nominally successful builds, but running the my-first-fix example results in the following error:

```gnss-sdr --config_file=./my-first-GNSS-SDR-receiver.conf
Initializing GNSS-SDR v0.0.17 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
RF Channels: 1
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Processing file /home/user/work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 800000000 samples (1600000000 bytes)
GNSS signal recorded time to be processed: 99.95 [s]
/usr/include/c++/12.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::complex<float>; _Alloc = volk_gnsssdr::alloc<std::complex<float> >; std::vector<_Tp, _Alloc>::reference = std::complex<float>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
/usr/include/c++/12.1.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::complex<float>; _Alloc = volk_gnsssdr::alloc<std::complex<float> >; std::vector<_Tp, _Alloc>::reference = std::complex<float>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[1]    19064 IOT instruction (core dumped)  gnss-sdr --config_file=./my-first-GNSS-SDR-receiver.conf
```
The only way to avoid the size assertion failure is to build and install the volk_gnsssdr library separately first, and then build and install gnss-sdr.  Is there a linking step missing in the cmake recipe if you don't already have volk_gnsssdr library installed on your system?

carlesfernandez, 2022-06-11:
Dear @rom-burner,

I've just built the software in a docker container and I can't reproduce this problem:

```
docker run -it --rm archlinux:latest
pacman -Syu --noconfirm
pacman -S --noconfirm gcc make automake m4 libtool autoconf cmake pkgconf \
  git boost boost-libs log4cpp libunwind libvolk gnuradio libpcap blas lapack \
  gflags google-glog openssl python-mako pugixml hdf5 gtest protobuf libsndfile
git clone https://github.com/gnss-sdr/gnss-sdr
cd gnss-sdr/build
git checkout next
cmake ..
make -j2
make install
```

… and the binary is usable. You can test it by doing:

```
cmake -DENABLE_SYSTEM_TESTING_EXTRA=ON ..
make -j2
../install/position_test
```

rom-burner, 2022-06-12:
Thanks for taking the time to look at this @carlesfernandez.

I confirm building it in docker also works on the `main` branch that i was using.

Weirdly you can replicate the problem in docker archlinux:latest if you pull and build the package at https://aur.archlinux.org/packages/gnss-sdr-git as a nonroot user.

The only difference I can see with that package and your recipe seems to be it builds from `main`, it has a couple of cmake cache flags (`-DENABLE_OSMOSDR=ON -DENABLE_UNIT_TESTING=OFF`) and it forces some additional (unnecessary?) dependencies over and above those you list (`armadillo`, `libmatio`, `airspy`, `bladerf`, `gnuradio-iqbal`, `hackrf`, `rtl-sdr` and `gnuradio-osmosdr`).

Before I get in touch with the package maintainer do you know if it's likely to be the cmake flags or some dependency conflict that could cause the issue?

carlesfernandez, 2022-06-13:
Hi @rom-burner ,

`volk_gnsssdr` uses a vector template with its own allocator (see [src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/include/volk_gnsssdr/volk_gnsssdr_alloc.h](https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/include/volk_gnsssdr/volk_gnsssdr_alloc.h)):

```
namespace volk_gnsssdr
{
...
template <class T>
using vector = std::vector<T, alloc<T> >;
}
```
The library passes the ASAN checks, but thanks for the hint. I'll reproduce this, adding `-Wp,D_GLIBCXX_ASSERTIONS`, and try to narrow-down the cause.

rom-burner, 2022-06-13:
here's the offender

https://github.com/gnss-sdr/gnss-sdr/blob/bba80cb920ea79e4989d77352f9f7dd4bb3e4d2a/src/algorithms/acquisition/gnuradio_blocks/pcps_acquisition.cc#L973-L975

i know it's just a pointer to the start of the `volk_gnsssdr::vector` but it is out of bounds and technically undefined behaviour.  a bit more cplusplussey with `std::copy` instead, or at least `d_data_buffer.data()` should fix it?

carlesfernandez, 2022-06-13:
Something like this should work, and it writes the data in the right position:

```
std::move(in, in + buff_increment, d_data_buffer.begin() + d_buffer_count);
```

Could you please confirm that replacing the `memcpy(...)` line by that one fixes the problem?  Then I'll make that fix in all the applicable instances.

rom-burner, 2022-06-13:
confirmed - that fixes the issues. thks!

carlesfernandez, 2022-06-13:
It does compile and run under macOS (so it actually _moves_ the data). It can be replaced by

```
std::copy(in, in + buff_increment, d_data_buffer.begin() + d_buffer_count);
```

which maybe is more portable.

carlesfernandez, 2022-06-14:
The `next` branch should be ok now.

commit e8fb655f (2022-06-14): Fixes binary on ArchLinux (Fixes: #616)

Replace some memcpy by std:::copy or std::copy_n
closed 2023-04-06 by carlesfernandez