gnss-sdr/gnss-sdr #618: Using GNSS-SDR with multiple input files

cipriancircu, 2022-06-08:
Hello.
The signal that I’m working with is split into multiple binary files.
I was wonder if it’s possible to use multiple files as input for gnss-sdr. 
Thank you!

nifx, 2022-07-13:
I just saw this by searching answers to my issue -> https://github.com/gnss-sdr/gnss-sdr/issues/633

There, I came across
https://github.com/gnss-sdr/gnss-sdr/issues/410

which might help you with multiple binary files

fener1efes2, 2022-07-23:
> I know `Labsat_Signal_Source` can (see [docs](https://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-labsat_signal_source)). I'm not sure about a generic `File_Signal_Source`

Is there a specific `Signal_Source` or a config file to process [TEXBAT](https://radionavlab.ae.utexas.edu/texbat/) files?