gnss-sdr/gnss-sdr #620: Failed to read GPS ephemeris data from SUPL server

BoGuTU, 2022-06-14:
Hello everyone. I'm trying to read GPS ephemeris data from SUPL server,but it failed.
About the config file, here it is:

;######### SUPL RRLP GPS assistance configuration #####
; Check http://www.mcc-mnc.com/
; On Android: https://play.google.com/store/apps/details?id=net.its_here.cellidinfo&hl=en
GNSS-SDR.SUPL_gps_enabled=true
GNSS-SDR.SUPL_read_gps_assistance_xml=false
GNSS-SDR.SUPL_gps_ephemeris_server=supl.google.com
GNSS-SDR.SUPL_gps_ephemeris_port=7275
GNSS-SDR.SUPL_gps_acquisition_server=supl.google.com
GNSS-SDR.SUPL_gps_acquisition_port=7275
GNSS-SDR.SUPL_MCC=262
GNSS-SDR.SUPL_MNC=7
GNSS-SDR.SUPL_LAC=41728
GNSS-SDR.SUPL_CI=28059821

Here are the results:

ERROR: SUPL client request for ephemeris data returned -2
Please check your network connectivity and SUPL server configuration

ZaquL, 2022-06-29:
> GNSS-SDR.SUPL_MNC=7

There is no such thing. All MNCs are two and/or 3 numbers under MCC. Yes, one MCC can have 3 numbers are two numbers MNC simultaniously, that do not overlap, e.g. 001 and 01 are two different MNC under one MCC.

>Check http://www.mcc-mnc.com/

It is far from complete. Use Wireshark's list.

Assely11, 2024-01-04:
when you use your browser to open the URL, can it be opened successfully?

ValeZAA, 2024-01-04:
No the play Google app is gone. You can use LTE Discovery app or many others