gnss-sdr/gnss-sdr #621: no BDS satellites with PRN > 30 can be acquired and tracked

peiyzhou, 2022-06-17:
Hi! I was using gnss-sdr to decode my collected raw IF datasets with BDS B1I frequency. For the channels settings:

Channels_B1.count=40
Channels.in_acquisition=20
Channel.signal=B1

I was able to decode BDS B1I satellites for PRN smaller than 30 (from both BDS-2&3), but was not able to acqusite BDS satellites with PRN >30, which should be available there.  I was wondering if there are some settings related to PRNs which should be increased to processing satellites with larger PRNs?

Thank you!

peiyzhou, 2022-06-17:
it seems the PRNs are set to <32 in the function beidou_b1i_code_gen_int(), which is used for local PRN code generation.

vladisslav2011, 2022-06-22:
There are no specific settings.
You can see PRN 32 tracked at B1I here:
https://www.youtube.com/watch?v=BIf4bZthvUE
And PRN 33, PRN 40, PRN 56 at B3I here:
https://www.youtube.com/watch?v=jwFO56G27vk

minecraft2048, 2022-06-25:
> 

Off topic, but I wonder.. is the redacted coordinates an upstream feature?

peiyzhou, 2022-06-25:
> There are no specific settings. You can see PRN 32 tracked at B1I here: https://www.youtube.com/watch?v=BIf4bZthvUE And PRN 33, PRN 40, PRN 56 at B3I here: https://www.youtube.com/watch?v=jwFO56G27vk

B1I and B3I use different methods for generating PRN code. PRN code for B1I is only implemented for BDS PRN<=32 (function beidou_b1i_code_gen_int()), and PRN codes for all BDS B3I satellites are implemented.

peiyzhou, 2022-06-25:
> There are no specific settings. You can see PRN 32 tracked at B1I here: https://www.youtube.com/watch?v=BIf4bZthvUE And PRN 33, PRN 40, PRN 56 at B3I here: https://www.youtube.com/watch?v=jwFO56G27vk

Anyways, I use the current code to process B1I observations, including code and carrier phase, and use rtklib to obtain a BDS B1I only short baseline RTK solution. I was not able to fix the interger ambiguity, although the accuracy of the baseline solution is at cm-level. did you try to use the B1I carrier phase for RTK?? 
Thanks

vladisslav2011, 2022-06-25:
@minecraft2048
Just a joke. Nevermind.

vladisslav2011, 2022-06-25:
> I was wondering if there are some settings related to PRNs which should be increased to processing satellites with larger PRNs?

I've checked the code. Only PRNs 1-33 are supported for now: https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/libs/beidou_b1i_signal_replica.cc#L53
Wait some time, please. I'll open a PR to fix this issue.

peiyzhou, 2022-06-26:
@vladisslav2011 thanks for the pr, it did fix the acq and track of satellites with prn >32.

It is noted that B1I also has carrier phase observations, even without the use of B1c, you may have a look at the dumped rinex file.

vladisslav2011, 2022-07-08:
> It is noted that B1I also has carrier phase observations

Of course, I'm wrong. I thought about pilot (that is not present in B1I signal) while writing about carrier (that is always present).
The carrier phase measurement are always done and passed to RTKLIB. I can confirm that RTKLIB gives only baseline solutions based on pseudoranges and does not take carrier phase measurements into account in PPP_Static/PPP_Kinematic modes. I think it would be better to open a separate issue about not working PPP modes... And another issue about bad telemetry decoding and another one about different tracking parameters for GEO satellites...

vladisslav2011, 2022-07-11:
PPP_static mode works in multiple system setup (BeiDou+GPS+Galileo)
https://www.youtube.com/watch?v=ZQs2sFchJ6w
That's strange. I think there is a bug in rtklib.

vladisslav2011, 2022-07-17:
@peiyzhou
My PR got merged. Could you checkout https://github.com/gnss-sdr/gnss-sdr/tree/next branch and retest.

ValZapod, 2022-08-07:
> That's strange. 

Why do you think it is stange that PPP_static works?

vladisslav2011, 2022-08-07:
The PPP_static mode did not work in single system setup (BeiDou B1I only), but it works in triple system setup. And it looks like BeiDou satellites are actually used by RTKLIB to calculate a PVT.
How could it be possible if phase information, supplied to RTKLIB is not accurate enough?
I think, RTKLIB should ignore BeiDou satellites after switching to precise mode and use only GPS and Galileo satellites to improve the accuracy. But it uses BeiDou satellites and shows improved accuracy. So, there is a bug in RTKLIB, that prevents it from switching to precise mode in single system configuration and phase information from B1I tracking block is good enough to improve the PVT.

peiyzhou, 2022-09-01:
@vladisslav2011 thanks for the PR. I tested with the next branch, the issue has been fixed. Now GNSS-SDR can recognize all BDS satellites.

closed 2022-09-01 by peiyzhou