gnss-sdr/gnss-sdr #623: Two signal sources to single channel

fener1efes2, 2022-06-19:
I want to simulate spoofing of GPS receivers.
Therefore, I need to implement two signal sources.
One of them will be authentic GPS signal, the other one will be spoofing signal.

Can I assign two signal sources to a single RF channel?

ZiglioUK, 2022-06-20:
Pretty sure that's not possible, each signal will be processed by an indipendent Tracking thread. You will need to mix the two signals yourself offline (if files) or with a Y cable if in real time.

jwmelto, 2022-06-22:
@fener1efes2 whenever this kind of question pops up on the GNU Radio chat, the gurus gently nudge the person to learn some basic DSP skills. It may seem tempting to just skip that part and run right to coding, but in the long run, you will be better off if you understand what you are doing. A great place to start is https://pysdr.org.

Since this question isn't really specific to GNSS-SDR, you might consider closing it. And to learn more about GNU Radio and SDR, see https://matrix.to/#/#gnuradio:gnuradio.org. For GNSS-SDR, you can try https://matrix.to/#/#gnss-sdr:gnuradio.org

Hope this helps.

closed 2022-06-22 by fener1efes2