gnss-sdr/gnss-sdr #633: SegFault by using 2 RF channels and Glonass L1/L2

nifx, 2022-07-12:
[gnss_sdr_glonass_l1_l2.txt](https://github.com/gnss-sdr/gnss-sdr/files/9092568/gnss_sdr_glonass_l1_l2.txt)

Hi all,

first of all, thanks a lot for this amazing tool. I really enjoy using it.
Currently, I do some tests with Glonass samples.
It is my first multi-frequency configuration. It uses Glonass L1 / L2.
Please find the configuration attached.

If I start gnss_sdr 0.0.17 (Docker) with the attached configuration, I get a segfault:

> Initializing GNSS-SDR v0.0.17 ... Please wait.
> Logging will be written at .
> RF Channels: 2
> Processing file ./IqStream, which contains 175383063552 samples (350766127104 bytes)
> GNSS signal recorded time to be processed: 158.69 [s]
> Segmentation fault (core dumped)
 
If I downstrip the configuration to separate L1 and L2 into separate configuration files, both are working.
But if I use 2 RF channels, I cannot get it running.

Note: I minimized the number of channels in the attached configuration to get a feeling if I did something wrong here.
I also tested with more channels without success.

From the logfiles, I cannot find anything suspicious. Only .INFO is generated, unfortunately without any error messages.

Thanks a lot for any ideas.

BR Nico

jwmelto, 2022-07-13:
A `File_Signal_Source` does not provide two channels of output.

The most likely situation is that you have recorded G1 in one file and G2 in another. Simply specify two `File_Signal_Sources` as `SignalSource0` and `SignalSource1`

(I should point out that the `File_Signal_Source` does not support the `sampling_frequency` attribute, so that does nothing (except perhaps serve as documentation).

You can also use the `Multichannel_File_Signal_Source` which is similar to two file sources (It allows you to specify any number of files/channels).

nifx, 2022-07-13:
Thanks @jwmelto for your feedback.
Indeed, I have a high-bandwidth sample file (~550MHz sample rate, complex samples) which contains L1 and L2.
But nevertheless, I was able to define 2 signal sources pointing to the same file which is working better, thanks to your hint.
Thus, I do not get a segmentation fault with this solution.

But what I don't understand now is the following:
- if I configure channels for 1G (Glonass L1 C/A) and 2G (Glonass L2 C/A), I see only console output for Glonass L1 (tracking started, NAV decoded)
- if I set the number of channels for Glonass L1 C/A to 0, I see the console output for Glonass L2
- but why does it not work for both L1 / L2 simultaneously?

Please find the configuration attached.
[gnss_sdr_glonass_l1_l2.txt](https://github.com/gnss-sdr/gnss-sdr/files/9099906/gnss_sdr_glonass_l1_l2.txt)

I played a lot of different combinations of:

- GNSS-SDR.num_sources
- Receiver.sources_count
- RF_channel_ID (I think this not the correct one for multiple signal sources)
- Signal_Source_ID (I think this is the one to use)

I would expect to see tracking and NAV messages in the console output for Glonass L1 and L2.
Do you have any insights why this does not seem to work?

nifx, 2022-07-13:
I also tried it with Multichannel_File_Signal_Source
also in combination with RF_channel_ID in channel assignments. Still no luck

nifx, 2022-07-13:
Funny thing is that I get GPS L1 C/A + Glonass L2 C/A running.
But when I replace GPS L1 C/A with Glonass L1 C/A, Glonass L2 C/A seems dead.

vladisslav2011, 2022-07-13:
Hello everyone.
Backtrace:
```
Thread 3 "valve92" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe6eec700 (LWP 13603)]
__memcpy_ssse3 () at ../sysdeps/x86_64/multiarch/memcpy-ssse3.S:132
132	../sysdeps/x86_64/multiarch/memcpy-ssse3.S: Нет такого файла или каталога.
(gdb) bt
#0  __memcpy_ssse3 () at ../sysdeps/x86_64/multiarch/memcpy-ssse3.S:132
#1  0x000055555571151a in Gnss_Sdr_Valve::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#2  0x00007ffff6e85107 in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.11
#3  0x00007ffff6e436d2 in gr::block_executor::run_one_iteration() ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.11
#4  0x00007ffff6e8fc86 in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, boost::shared_ptr<boost::barrier>, int) ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.11
#5  0x00007ffff6e80824 in ?? () from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.11
#6  0x00007ffff6e2d332 in ?? () from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.11
#7  0x00007ffff78dabcd in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#8  0x00007ffff4ed06db in start_thread (arg=0x7fffe6eec700) at pthread_create.c:463
#9  0x00007ffff2efe61f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```