gnss-sdr/gnss-sdr #636: Multiple source approach does not work if the input sources are having different frequencies

cipriancircu, 2022-07-15:
When trying to run the tool with multiple sources (File signal source implementation) that are on different frequencies, the tool only detects L1 band

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=25000000
GNSS-SDR.num_sources=2   ; starting from v0.0.15

;######### SIGNAL_SOURCE CONFIG ############

SignalSource0.implementation=File_Signal_Source
SignalSource0.sampling_frequency=25000000
SignalSource0.filename=
SignalSource0.item_type=ishort
SignalSource0.samples=0
SignalSource0.repeat=false
SignalSource0.enable_throttle_control=false

SignalSource1.implementation=File_Signal_Source
SignalSource1.filename=
SignalSource1.sampling_frequency=25000000
SignalSource1.item_type=ishort
SignalSource1.samples=0
SignalSource1.repeat=false
SignalSource1.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner0.implementation=Signal_Conditioner

DataTypeAdapter0.implementation=Ishort_To_Complex

InputFilter0.implementation=Pass_Through ; or Fir_Filter
InputFilter0.input_item_type=gr_complex
InputFilter0.output_item_type=gr_complex
InputFilter0.taps_item_type=float
InputFilter0.number_of_taps=5
InputFilter0.number_of_bands=2
InputFilter0.band1_begin=0.0
InputFilter0.band1_end=0.44
InputFilter0.band2_begin=0.55
InputFilter0.band2_end=1.0
InputFilter0.ampl1_begin=1.0
InputFilter0.ampl1_end=1.0
InputFilter0.ampl2_begin=0.0
InputFilter0.ampl2_end=0.0
InputFilter0.band1_error=1.0
InputFilter0.band2_error=1.0
InputFilter0.filter_type=bandpass
InputFilter0.grid_density=16
InputFilter0.sampling_frequency=25000000
InputFilter0.IF=0
InputFilter0.dump=false
InputFilter0.dump_filename=../data/input_filter.dat

Resampler0.implementation=Pass_Through

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner1.implementation=Signal_Conditioner

DataTypeAdapter1.implementation=Ishort_To_Complex

InputFilter1.implementation=Pass_Through ; or Fir_Filter
InputFilter1.input_item_type=gr_complex
InputFilter1.output_item_type=gr_complex
InputFilter1.taps_item_type=float
InputFilter1.number_of_taps=5
InputFilter1.number_of_bands=2
InputFilter1.band1_begin=0.0
InputFilter1.band1_end=0.44
InputFilter1.band2_begin=0.55
InputFilter1.band2_end=1.0
InputFilter1.ampl1_begin=1.0
InputFilter1.ampl1_end=1.0
InputFilter1.ampl2_begin=0.0
InputFilter1.ampl2_end=0.0
InputFilter1.band1_error=1.0
InputFilter1.band2_error=1.0
InputFilter1.filter_type=bandpass
InputFilter1.grid_density=16
InputFilter1.sampling_frequency=25000000
InputFilter1.IF=0
InputFilter1.dump=false
InputFilter1.dump_filename=../data/input_filter.dat

Resampler1.implementation=Pass_Through

;######### CHANNELS L1 GLOBAL CONFIG ############
Channels.in_acquisition=2

Channels_1C.count=5
Channel0.signal=1C
Channel0.SignalSource_ID=0
Channel1.signal=1C
Channel1.SignalSource_ID=0
Channel2.signal=1C
Channel2.SignalSource_ID=0
Channel3.signal=1C
Channel3.SignalSource_ID=0
Channel4.signal=1C
Channel4.SignalSource_ID=0

Channels_1B.count=5
Channel5.signal=1B
Channel5.SignalSource_ID=0
Channel6.signal=1B
Channel6.SignalSource_ID=0
Channel7.signal=1B
Channel7.SignalSource_ID=0
Channel8.signal=1B
Channel8.SignalSource_ID=0
Channel9.signal=1B
Channel9.SignalSource_ID=0

;######### CHANNELS L5 GLOBAL CONFIG ############
Channels_L5.count=5
Channel10.signal=L5
Channel10.SignalSource_ID=1
Channel11.signal=L5
Channel11.SignalSource_ID=1
Channel12.signal=L5
Channel12.SignalSource_ID=1
Channel13.signal=L5
Channel13.SignalSource_ID=1
Channel14.signal=L5
Channel14.SignalSource_ID=1

Channels_5X.count=5
Channel15.signal=5X
Channel15.SignalSource_ID=1
Channel16.signal=5X
Channel16.SignalSource_ID=1
Channel17.signal=5X
Channel17.SignalSource_ID=1
Channel18.signal=5X
Channel18.SignalSource_ID=1
Channel19.signal=5X
Channel19.SignalSource_ID=1

;# GPS L1 ACQUISITION #
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.threshold=2;
Acquisition_1C.pfa=0.01;
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.max_dwells=2
Acquisition_1C.repeat_satellite=false
Acquisition_1B.implementation=Galileo_E1_PCPS_Ambiguous_Acquisition
Acquisition_1B.item_type=gr_complex
Acquisition_1B.threshold=2;
Acquisition_1B.pfa=0.01;
Acquisition_1B.doppler_max=6000
Acquisition_1B.doppler_step=100
Acquisition_1B.coherent_integration_time_ms=1
Acquisition_1B.max_dwells=2
Acquisition_1B.repeat_satellite=false

;# GPS L5 ACQUISITION #
Acquisition_L5.implementation=GPS_L5i_PCPS_Acquisition
Acquisition_L5.item_type=gr_complex
Acquisition_L5.threshold=2.5;
Acquisition_L5.pfa=0.01;
Acquisition_L5.doppler_max=6000
Acquisition_L5.doppler_step=100
Acquisition_L5.coherent_integration_time_ms=1
Acquisition_L5.max_dwells=2
Acquisition_L5.repeat_satellite=false

Acquisition_5X.implementation=Galileo_E5a_Noncoherent_IQ_Acquisition_CAF
Acquisition_5X.item_type=gr_complex
Acquisition_5X.coherent_integration_time_ms=1
Acquisition_5X.threshold=0.009
Acquisition_5X.doppler_max=5000
Acquisition_5X.doppler_step=125
Acquisition_5X.bit_transition_flag=false
Acquisition_5X.max_dwells=1
Acquisition_5X.CAF_window_hz=0 ;  **Only for E5a** Resolves doppler ambiguity averaging the specified BW in the winner code delay. If set to 0 CAF filter is desactivated. Recommended value 3000 Hz
Acquisition_5X.Zero_padding=0 ; **Only for E5a** Avoids power loss and doppler ambiguity in bit transitions by correlating one code with twice the input data length, ensuring that at least one full code is present without transitions. If set to 1 it is ON, if set to 0 it is OFF.
Acquisition_5X.dump=false
Acquisition_5X.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL L1 CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0
Tracking_1C.dll_bw_hz=4.0
Tracking_1C.dump=true
Tracking_1C.dump_filename=./tracking_ch_
Tracking_1C.dump_mat=false

Tracking_1B.implementation=Galileo_E1_DLL_PLL_VEML_Tracking
Tracking_1B.item_type=gr_complex
Tracking_1B.pll_bw_hz=40.0
Tracking_1B.dll_bw_hz=4.0
Tracking_1B.dump=true
Tracking_1B.dump_filename=./tracking_ch_
Tracking_1B.dump_mat=false

;######### TRACKING GLOBAL CONFIG ############
Tracking_L5.implementation=GPS_L5_DLL_PLL_Tracking
Tracking_L5.item_type=gr_complex
Tracking_L5.track_pilot=true
Tracking_L5.pll_bw_hz=15.0;
Tracking_L5.dll_bw_hz=2.0;
Tracking_L5.pll_bw_narrow_hz=4.0;
Tracking_L5.dll_bw_narrow_hz=1.0;
Tracking_L5.order=2;
Tracking_L5.early_late_space_chips=0.5;
Tracking_L5.dump=false
Tracking_L5.dump_filename=./tracking_ch_

Tracking_5X.implementation=Galileo_E5a_DLL_PLL_Tracking
Tracking_5X.item_type=gr_complex
Tracking_5X.track_pilot=true
Tracking_5X.pll_bw_hz=15.0;
Tracking_5X.dll_bw_hz=2.0;
Tracking_5X.pll_bw_narrow_hz=5.0;
Tracking_5X.dll_bw_narrow_hz=1.0;
Tracking_5X.order=2;
Tracking_5X.early_late_space_chips=0.5;
Tracking_5X.dump=false
Tracking_5X.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER L1 CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1B.implementation=Galileo_E1B_Telemetry_Decoder

;######### TELEMETRY DECODER L5 CONFIG ############
TelemetryDecoder_5X.implementation=Galileo_E5a_Telemetry_Decoder
TelemetryDecoder_L5.implementation=GPS_L5_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true
Observables.dump_filename=./obs
Observables.dump_mat=false
#Observables.remove_dat=true

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.averaging_depth=1000
PVT.flag_averaging=true
PVT.output_rate_ms=10000
PVT.display_rate_ms=1000
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.rinex_output_enabled=true
PVT.rinex_output_path=./Rinex_output
PVT.rinexobs_rate_ms=1000
PVT.dump=true
PVT.dump_filename=./pvt
#PVT.dump_mat=false
PVT.enable_monitor=true
PVT.monitor_client_addresses=127.0.0.1
PVT.monitor_udp_port=1111

Monitor.enable_monitor=true
Monitor.decimation_factor=1
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1112

Also, although the run should last 10 minutes, the receiver time never exceeds 2 minutes.
Another weird behavior is the way in which gnss-sdr handles the channels. As you can see in the lines attached, each channel has its own source and frequency band attributed (as described in the guide). But when running it, the channels seem to be randomly populated with data which is not from the correct frequency, or even constelation (check the attached image).
[image]

jwmelto, 2022-07-15:
Try throttle on. 

With two files, the first file is read to completion, triggering a shutdown, before the second is processed sufficiently. 

I think this is why I use the multi file source for this type of processing.

vladisslav2011, 2022-07-17:
`Channel#.SignalSource_ID` configuration parameter is not used in the source code. You should use `Channel#.RF_channel_ID` parameter or `Channels_1C.RF_channel_ID` and similar parameters if you are using "next" branch.
This config
[gal5+gal1.txt](https://github.com/gnss-sdr/gnss-sdr/files/9128531/gal5%2Bgal1.txt)
works well (at least tracking if you want PVT, replace `PVT.iono_model=Iono-Free-LC` with `PVT.iono_model=Broadcast` or be lucky enough to track 6+ satellites on 5X).

Annemariev-zwol, 2022-07-21:
Hi all,

We unfortunately experience the same issue. Thanks for creating the ticket and for the already given suggestions. Up to now we have tried the following tests and/or potential solutions, unfortunately with no success:

- We have compared the time solutions (TOW_at_start_symbol_ms) obtained for the same satellites in both bands (by performing separate runs with the two input files). Although there are very small differences (order of 1-10 ns) comparison to a run with a previously generated input file which contained both bands did show time differences of the same order so this doesn’t seem to be the problem.

- We reversed the signal source order (source 0 lower L-band, source 1 upper L band). Although I could still see messages being received no data was stored to the observables dump at all. (I somewhat expected data of one of the two bands to still be store to the observables dump, and therefore to get a bit more insight in what the problem might be)

- We tried the suggestion given below your ticket to use the throttle on function, this made no difference. Outcome was the same as without throttle on

- I already included Channel#.RF_channel_ID in the configuration file, this doesn’t solve the problem. (although I assigned every channel individually instead of all signals of a certain type at once, I will check whether this makes a difference)

More suggestions for possible causes or solutions are of course welcome (and thanks in advance for anyone contributing to this issue)

Best regards,
Annemarie