gnss-sdr/gnss-sdr #637: RTL-SDR .conf file

fener1efes2, 2022-07-18:
I have built test setup for RTL-SDR mentioned in GNSS-SDR's web page.
[https://gnss-sdr.org/docs/tutorials/gnss-sdr-operation-realtek-rtl2832u-usb-dongle-dvb-t-receiver/](GNSS-SDR)

I have used an active antenna. I know that the antenna works very well since I connected the antenna to a U-BLOX receiver and the receiver successfully has given navigation solutions.

But when it comes to GNSS-SDR it does not give position solutions with active antenna.
I have used the same config file.
Where do I get wrong?

wkazubski, 2022-07-30:
It is not so easy. I have some experience with receiving GPS, Galileo and BeiDou with rtl-sdr dongle.
First, not all dongles receive at L1 frequency. Old ones with Elonics E4000 RF chip were the best but are very hard to find now. Most recent ones use RM820T chps usually work up to 13 or 1.4GHz, at least the ones I have. Latest RM820T2 are somewhat better (working to around 1.5 to 1.6GHz), so there are some chances for success. If you have a RF signal generator covering L1 frequency, try to check the dongle using GQRX, SDR# or similar software.
Second, commercial GPS antennas are usually actve and require 3.3 or 5V bias. Only original rtl-sdr v3 dongles have working biads option. Alternatively, use external bias tee.
Third, dongles usually have substntial frequency offset, sometimes as high as 100ppm.Only original V3 dongles have working TCXO and do not need frequecy correction. Some non original dongles sold as v3 do not have TCXO or have it but not connected. Gnss-sdr can tolerate small frequency offset, but  greaer one must be compensated by setting apropriate sampling frequency and IF fequency.

fener1efes2, 2022-08-21:
Thank you for your comment @wkazubski 
I have used an external bias tee to feed the antenna. Bias tee works well.
I have an original RTL-SDR dongle.

I connected a GPS simulator to RTL-SDR and it works well, it has navigation solutions.
But when it comes to open sky, it does not work.

vladisslav2011, 2022-08-21:
You can check for the presence of the signal with generic SDR software. Gqrx works well. Set FFT size to 1M and the signal will be visible. If you see something like this https://www.youtube.com/watch?v=A9a5G_WPAsM&t=57s then your setup should work very well. Covering the antenna with hand should make noise bursts disappear.
If you have original HF-capable dongle, you should be able to calibrate the frequency offset using time station signals at 10.0MHz or using narrow Inmarsat channels at L-band.