gnss-sdr/gnss-sdr #639: Arch Linux linker error

bmcq55, 2022-07-22:
In order to avoid the following error message:
```
[100%] Built target core_receiver
[100%] Linking CXX executable gnss-sdr
/usr/bin/ld: ../../glog-0.6.0/libglog.a(utilities.cc.o): undefined reference to symbol '_ULx86_64_init_local'
/usr/bin/ld: /usr/lib/libunwind.so.8: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/main/CMakeFiles/gnss-sdr.dir/build.make:188: src/main/gnss-sdr] Error 1
make[1]: *** [CMakeFiles/Makefile2:2641: src/main/CMakeFiles/gnss-sdr.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

I had to change the following files:
```

./gnss-sdr/build/src/main/./CMakeFiles/gnss-sdr.dir/link.txt
./gnss-sdr/build/src/tests//CMakeFiles/run_tests.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/flowgraph_test.dir/link.txt
./gnss-sdr/build/src/tests//CMakeFiles/gnss_block_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/gnuradio_block_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/run_tests.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/flowgraph_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/gnss_block_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/gnuradio_block_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/matio_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/acq_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/trk_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/control_thread_test.dir/link.txt
./gnss-sdr/build/src/tests/CMakeFiles/utils/front-end-cal/CMakeFiles/front-end-cal.dir/link.txt

```
Without making those modifications, the program does not compile successfully.

To include the system /usr/lib/libunwind.so library as follows:
`
/usr/bin/c++ -O3 -DNDEBUG "CMakeFiles/gnss-sdr.dir/main.cc.o" -o gnss-sdr /usr/lib/libunwind.so
`
(Note that I'm including the system libunwind.so).

Without this change, the program won't compile

jwmelto, 2022-07-23:
The issue is here:
> `/usr/bin/ld: ../../glog-0.6.0/libglog.a(utilities.cc.o): undefined reference to symbol '_ULx86_64_init_local'
`
It is an issue with glog (in particular, the GetStackTrace function in utilities.o). I can't figure out the proper CMake magic just now; one solution would be for you to build and install your own GFlags library, then build GNSS-SDR with `-DENABLE_OWN_GLOG` If the problem persists, then I'd take it up with glog.

jwmelto, 2022-07-23:
This appears to be duplicate of #562

carlesfernandez, 2022-07-25:
Please let me know if the current `next` branch fixes this issue.