gnss-sdr/gnss-sdr #643: Compiled Releases

ShadowJonathan, 2022-07-30:
While the build instructions on this repo are very good, I think that for the software to reach a wider audience (and to be more easily usable), the releases should contain pre-compiled executables, and/or you can have compiled versions on other distribution networks, such as Apt (via distro lists) or Homebrew.

For homebrew notably it'd be very useful to have a compiled app, as installing xcode plus dependencies is a very uncommon method of installation for popular or established apps. (I'm not asking for it to get put on the play store)

ShadowJonathan, 2022-07-30:
It looks like [this page](https://gnss-sdr.org/build-and-install/#macos--mac-os-x-using-macports) contains some installation releases, however, these are not immidiately noted in the readme, and is behind a link that can be easily missed. I recommend merging them into the readme, or moving the build instructions out of the readme into a seperate file, so that the links mentioning "building" and "installation" are more clearly put together. Right now anyone can scroll down to brute-force installing it, while the above method would force them to look a little further.

carlesfernandez, 2024-12-05:
Maybe you can use the [Docker image](https://hub.docker.com/r/carlesfernandez/docker-gnsssdr), which will save you from having to build the software.

closed 2024-12-05 by carlesfernandez