gnss-sdr/gnss-sdr #650: Converting cshort to gr_complex without filtering

minecraft2048, 2022-08-09:
The table in https://gnss-sdr.org/docs/tutorials/understanding-data-types/#from-the-signal-source-to-the-processing-channels seems to show that you can use `pass_through` filter to convert cshort data type to gr_complex, but when I tried it by setting this in the config:
```
InputFilter0.implementation=Pass_Through
InputFilter0.input_item_type=cshort
InputFilter0.output_item_type=gr_complex
```
it crashes with `E0809 14:56:44.607372 3083725 gnss_flowgraph.cc:630] Can't connect signal conditioner block internally: itemsize mismatch: Invalid input/ouput data type configuration for the DataTypeAdapter/InputFilter connection`

Is there a way to convert cshort to gr_complex without using any of the FIR filters?

vladisslav2011, 2022-08-10:
Hello.
No.
If you want to convert your sample to gr_complex without filter, you should use DataTypeAdapter, not InputFilter.

```
DataTypeAdapter.implementation=Ishort_To_Complex
```

instead of

```
DataTypeAdapter.implementation=Ishort_To_Cshort
```

minecraft2048, 2022-08-24:
> Hello. No. If you want to convert your samples to gr_complex without filtering, you should use DataTypeAdapter, not InputFilter.
> 
> ```
> DataTypeAdapter.implementation=Ishort_To_Complex
> ```
> 
> instead of
> 
> ```
> DataTypeAdapter.implementation=Ishort_To_Cshort
> ```

The problem is that there is no `Cshort_To_Complex` implementation for DataTypeAdapter

jwmelto, 2023-01-06:
@minecraft2048 is there really a difference between `cshort` and `ishort`? They both have 16-bit integer I (real) followed by 16-bit integer Q (imaginary). If you use `Ishort_To_Complex`, does it not work?