gnss-sdr/gnss-sdr #653: CMake issues building with gnuradio 3.10

jwmelto, 2022-08-16:
GNU Radio moved from the unsupported log4cpp to spdlog in 3.10. The addressing of this in GNSS-SDR was, I fear, over-engineered. The current configuration will not compile on my CentOS 7 configuration. CentOS suffers from missing or old packages, so many things have to be built from source.

In the **FindSPDLOG.cmake** module, there is a blind assertion of a compile definition `-DSPDLOG_FMT_EXTERNAL=1`. This is incorrect. However, it's a symptom of a deeper issue.

When **spdlog** is installed, it provides CMake support files, such that `find_package(spdlog CONFIG)` should set all necessary variables. That we do `find_package(SPDLOG REQUIRED)` creates a need to write a redundant custom FindXXX.cmake module.

More to the point, as a consumer of GNU Radio, we shouldn't be specifying GNU Radio dependencies in the first place. It is the obligation of GNU Radio to expose their library dependencies through _their_ CMake modules.

I see a large ripple through the GNSS-SDR code-base where CMakeLists.txt have been expanded to add `spdlog::spdlog fmt::fmt` as link dependencies in all of our code that uses GNU Radio. I don't think this is needed. I'll be experimenting and see if I can reduce the tentacles caused by this approach.

ryanvolz, 2022-08-17:
As someone who has packaged GNU Radio and already had to figure out what's happening with `spdlog` and `libfmt`, hopefully I can offer some guidance. There was a period of time early on when GNU Radio 3.10 was not depending on `spdlog` correctly, and I think working around that bug is why `gnss-sdr` is having problems now.

If everything is working as intended (e.g with GR 3.10.3 at least), `gnss-sdr` should not need to add `-DSPDLOG_FMT_EXTERNAL=1`, and doing so causes problems for users with an `spdlog` that was built with the built-in `libfmt`. Also, there should be no circumstances where it would be necessary to independently link with `libfmt` or require it via CMake. `spdlog` will either depend on the external version or not, and be linked accordingly.

What I'm unsure of is whether or not it is necessary for `gnss-sdr` to depend on and link against `spdlog` directly. I think it is possible for a program to be built against GNU Radio and not itself link with `spdlog`, but I think it entirely depends on which GR headers it consumes. Most likely it is necessary.

Finally, if `gnss-sdr` needs the `fmtlib` headers for some reason (my brief search shows it doesn't, but maybe I missed it), it is absolutely critical to include `<spdlog/fmt/fmt.h>` and not `<fmt/fmt.h>`. Doing the latter is what GNU Radio initially did that caused headaches with internal/external linkage and requiring `libfmt` regardless of how `spdlog` was built.

jwmelto, 2022-08-17:
My initial attempt to remove all the explicit dependencies in the CMakeLists.txt files is promising. I was able to build to completion.

It's good to know that there was a legacy reason why things are the way they are, but I think I'll have a pull request soon that will undo all that and make it easier.

carlesfernandez, 2022-08-23:
Dear @jwmelto,

Thanks for raising this issue. The main root cause of this problem is, in my opinion, that we are not using GNU Radio’s CMake config files but a custom, quite heuristic FindGNURADIO.cmake module. This is due to the fact that old GNU Radio versions not always did a good job in creating those CMake files, and this caused problems in a variety of environments.

There is quite an effort in maintaining the code base buildable in ancient OS such as Debian 8 or Ubuntu 14.04 (C++11, CMake 2.8.12, GNU Radio v3.7.x, etc.). It wouldn’t be a catastrophe to drop that support, but it should be for a compelling reason. Some extra work with the current CMake files should allow for building the code both in ancient setups and in most modern ones (Including those using a custom building of GNU Radio), and this is a nice feature I would like to keep as far as it requires a reasonable amount of work. 

The bare minimum that should be supported by the code in order to allow for buildings in some relevant embedded systems are dictated by the [Geniux Rocko](https://gnss-sdr.org/introducing-geniux-v2206/#geniux-rocko-22060) environment, which is based on the Yocto Project 2.4.4 (CMake v3.8.2, GCC v7.3.0, GNU Radio v3.7.13, Boost v1.64.0, Python v2.7.14).

I’ve just pushed to the `next` branch [some changes](bef029cb50584058923dd624a472df85ba64dea8) that *should* fix your building issues while keeping that backward compatibility. Please let me know if those changes do not work for you.

jwmelto, 2022-08-24:
Thanks for that. I was struggling in my regressions, because some difference between the GNSS-SDR `FindVOLK.cmake` file and the Volk one caused Fedora to throw a "stack smashing abort". I'm building with your changes now. If all is well, I'll close this issue and close my PR.

I recognize the need for the custom `FindGNURADIO.cmake` file; I initially tried to remove ours, but found that there is much metadata asserted throughout. I also discovered that upstream sources don't always do you favors with the delivered CMake files (or packages).

There is a tension caused by trying to support old build environments on one end and newer upstreams on the other. I can't answer for this project where that balance lies. Given that GNU Radio now requires C++17 and Python 2.7 is end-of-life, _any_ rev-locked environment is destined for obsolescence. Whether new development on **this** project needs to be backported to the older ones is a choice to be made. It appears the current choice is to do so. Other projects create `maint` branches around major releases; new development is forward-facing but features may be backported if there is interest/desire in doing so.

closed 2023-01-06 by jwmelto