gnss-sdr/gnss-sdr #663: BUG: BeiDou B3I tracking block sometimes locks to wrong correlation peak

vladisslav2011, 2022-09-06:
This results in wrong pseudoranges calculated and, as a result, incorrect PVT.
Channels, that have locked to a wrong correlation peak are showing too low C/N0 in GNSS-SDR Monitor. Sometimes that channels recover and switch to highest peak, sometimes the lock gets lost and correct peak is chosen after reacquisition.

Steps to reproduce:
1. Prepare a baseband recording using good L2/L5 antenna placed in a good location. Strong signals with high SNR are required to trigger this bug reliably.
2. Prepare a configuration file using a recording from step 1 with monitoring enabled
3. Start GNSS-SDR monitor
4. Start GNSS-SDR
5. Check monitor for position jumps and abnormal C/N0 graph

Expected behavior:
Satellites at good elevation angles should show high C/N0 values (>40) and C/N0 graph should be smooth without strong fluctuation.
There should be no position jumps on a position plot.

Observed behavior:
Some satellites at good elevation angles are showing abnormal low C/N0 values (~30) and C/N0 graph is showing ~10db fluctuation.
There are position jumps visible on a position plot.

Demonstration video: in progress.
PR: in progress.

jwmelto, 2022-09-07:
I don't know what the storage limits are for this repo, but it would be great to have a data file that demonstrates the issue attached to the bug report. Of course, I speak only for me.

vladisslav2011, 2022-09-07:
```
$ ls -lh /home/vlad/ftp/gqrx_20220805_185829_1268520000_8000000_8.raw
-rw-rw-r-- 1 vlad vlad 5.9G Aug  5 22:05 /home/vlad/ftp/gqrx_20220805_185829_1268520000_8000000_8.raw
```
I don't think uploading 6Gb to github to be a good idea...
All my B3I IQ recordings demonstrate the issue. I can even put the antenna near the open window, start realtime config and get this issue in 2..3 minutes.

jwmelto, 2022-09-07:
I suspected as much. How well does the data compress? Even so, test case data files for this project are perpetually a problem.

vladisslav2011, 2022-09-07:
GNSS signal looks like noise, so it does not compress well.
It should not be hard to record a sample. 1 antenna (cloverleaf, DIY 1 evening) +1 LNA (1 transistor/mmic, 2 PCB SMA connectors, a piece of proto PCB and some passives, DIY 2 evenings or buy it from ebay/aliexpress) + 1 frontend (HackRF works well. Even new bad chinese clones.) and a good place to put an antenna (river bank is good in my QTH, a window of a city building is not as good, but works too).