gnss-sdr/gnss-sdr #666: Why is GNSS-SDR so big?

alexmealster19, 2022-09-08:
So I’ve been working with GNSS-SDR for a little while now, mainly trying to get it to install on single board computers. I have run into many problems trying to get it to install on single board computers and have been working on cross compiling from a more powerful device. However my question is, why is GNSS-SDR so large in the first place? I’m not going to pretend that I understand how exactly GNSS-SDR works but I have seen videos with other devices (Adafruit ultimate GPS breakout board and RTL-SDR) that are small and don’t seem to require strong processors with lots of storage. So what about GNSS-SDR makes it so large and requires so much RAM/storage to compile? And is there a way to strip it down to just receive positioning data? I would appreciate advice/insight from anyone who has experience with this software.

ValZapod, 2022-09-11:
> requires so much RAM/storage to compile?

Cause C++. Simple.

jwmelto, 2022-09-11:
> Cause C++. Simple.

That's not a very helpful answer.

On the other hand, I'm not sure what the OP intends with this issue. How much is "so much"? How little would you find acceptable? To reduce the size of the runtime program, you would first need to minimize Volk and GNU Radio to only build/install the portions you need. Then you would need to be careful to only configure the parts of GNU Radio that you need.

Optimization (both performance and space) is a complex process, and should only be performed once the logic is working  and to address a measurable performance problem. I would submit that GNSS-SDR is not (particularly) optimized at all; it is at Beta release maturity (as evidenced by the 0.0.x release numbering).

If you have specific optimizations to offer, [patches welcome](https://github.com/gnss-sdr/gnss-sdr/blob/main/CONTRIBUTING.md) 😃

ValZapod, 2022-09-11:
> That's not a very helpful answer.

True though. One removal of std::string: https://github.com/bastibl/gr-rds/pull/30

jwmelto, 2022-09-11:
> True though. One removal of std::string: [bastibl/gr-rds#30](https://github.com/bastibl/gr-rds/pull/30)

Again, not a particularly helpful answer. `std::string` provides a number of benefits over raw pointers (pointers are generally evil and the source of many exploits. The PR you cite has a 1280x4 array statically declared in a header file. That is the real issue, and not that the type of the array element is `std::string` vs `char const*`. `static` values in a header are instantiated in every single compilation unit where they are seen, multiplying the number of instances of this array.

Sure, the memory footprint of 1280x4x(sizeof(char const*)) is less than 1280x4x(sizeof(std::string)), but it is inconsequential until you start multiplying it by the number of compilation units in which it appears. 

The correct optimization (without giving up the type-safety of `std::string` would have been to remove the instantiation of this array from a header to a single compilation unit. I'd probably also use `std::string_literal` to declare strings, rather than C-string initialization, and I'd also use `std::array` to provide safety over the array of messages. My guess is that the performance between the modern C++ implementation and the C-style implementation would be inconsequential, when the naive code bloat was removed.

Optimization is a balancing act between concerns. Maintainability should be given precedence in all but the most performance-critical code. `std::string` is in almost all ways better than raw C-strings (most other languages implement quoted literals as a native string type; C++ was crippled in this regard by backwards-compatibility with C), and the library implementation is highly optimized by the compiler writers.

I don't think removing `std::string` will be positively entertained.

ValZapod, 2022-09-30:
>Sure, the memory footprint of 1280x4x(sizeof(char const*)) is less than 1280x4x(sizeof(std::string)),

Can you say 100% that something like this does not exist in this codebase? Even simple printf() that is optimised by compiler to puts() can be further optimised by write() and even then is very very slow. Just saying. fmtlib may be faster... But that is like boost: out of tree. https://github.com/fmtlib/fmt

I suppose PREEMT_RT of linux kernel should make it faster.

wow-awsl-qwq, 2022-11-10:
I dont think it is too big

closed 2024-10-11 by carlesfernandez