gnss-sdr/gnss-sdr #677: Assertion '!__overflow' failed in arch linux

sensille, 2022-10-10:
Running the my-first-GNSS-SDR-receiver example fails on current archlinux (next branch, commit:72b7311c6c):

`Initializing GNSS-SDR v0.0.17.git-next-72b7311c6 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
RF Channels: 1
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Processing file /home/arne/sdr/gnss1.dat, which contains 470679552 samples (941359104 bytes)
GNSS signal recorded time to be processed: 58.7849 [s]
/usr/include/c++/12.2.0/numeric:197: constexpr std::common_type_t<_Mn, _Nn> std::lcm(_Mn, _Nn) [with _Mn = long unsigned int; _Nn = long unsigned int; common_type_t<_Mn, _Nn> = long unsigned int]: Assertion '!__overflow' failed.
Aborted (core dumped)
` 

Stack trace:
`#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0)
    at pthread_kill.c:44
#1  0x00007ffff4f626b3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff4f12958 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff4efc53d in __GI_abort () at abort.c:79
#4  0x00007ffff529d112 in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, 
    function=<optimized out>, condition=<optimized out>)
    at /usr/src/debug/gcc/libstdc++-v3/src/c++11/debug.cc:60
#5  0x00007ffff6a239f0 in gr::flat_flowgraph::allocate_block_detail(std::shared_ptr<gr::basic_block>) ()
   from /usr/lib/libgnuradio-runtime.so.3.10.4
#6  0x00007ffff6a23c38 in gr::flat_flowgraph::setup_connections() () from /usr/lib/libgnuradio-runtime.so.3.10.4
#7  0x00007ffff6a56895 in gr::top_block_impl::start(int) () from /usr/lib/libgnuradio-runtime.so.3.10.4
#8  0x00007ffff6a56c85 in gr::top_block::start(int) () from /usr/lib/libgnuradio-runtime.so.3.10.4
#9  0x000055555560c7d8 in GNSSFlowgraph::start (this=0x555556aaa690)
    at /home/arne/sdr/gnss-sdr/src/core/receiver/gnss_flowgraph.cc:287
#10 0x00005555555de6e4 in ControlThread::run (this=0x555556aa7560)
    at /home/arne/sdr/gnss-sdr/src/core/receiver/control_thread.cc:365
#11 0x00005555555d4401 in main (argc=1, argv=0x7fffffffea20) at /home/arne/sdr/gnss-sdr/src/main/main.cc:130
`

I tried the AUR gnss-sdr-git as well as building from source myself, both master and next branch. All with the same result. gcc version is 12.2.0.

benedekt, 2022-11-14:
This is still an issue, I encountered the same problem.

benedekt, 2022-11-14:
I've encountered the same error message and trace in my own application.

After some trial and error, the source was traced back to some sample rate mismatch. If the signal source rate matches the system rate, GNSS-SDR runs well. If not, that strict assertion fails in libgnuradio. (A recompilation of GNSS-SDR won't fix the problem, since it comes from a lower layer.)

GeorgeTownsendd, 2022-12-01:
Just commenting to add that I've had the same experience: ran into this problem following the "My first position fix" guide on Arch Linux, with the provided data/config.

carlesfernandez, 2022-12-01:
Thanks @sensille for raising this issue and the detailed info. The problem seems related to `-D_GLIBCXX_ASSERTIONS`, which triggers the overflow in the internals of GNU Radio, specifically [here](https://github.com/gnuradio/gnuradio/blob/463c3477549b26b32d9b73eef30044e97c4eee64/gnuradio-runtime/lib/flat_flowgraph.cc#L127). 

This problem appeared when ArchLinux bumped to GCC 12.2.0, and it also happens on Fedora Rawhide. However, the issue does not show on OpenSUSE Tumbleweed, which also uses GCC 12.2.1.

We are chasing this bug, but still no luck. @benedekt could you elaborate a little bit more on that "If the signal source rate matches the system rate, GNSS-SDR runs well"?

sensille, 2022-12-02:
Thanks @carlesfernandez for looking into this.

carlesfernandez, 2022-12-04:
Hi,

could you check if the current `next` branch solves the issue for you?

GeorgeTownsendd, 2022-12-04:
Hi @carlesfernandez,

The next branch solved this issue for me- the 'my first fix' guide now executes as expected.

Many thanks for the fix and the project!

sensille, 2022-12-07:
I can also confirm that it works now. Thanks!

commit 6a2ac9ea (2022-12-04): Fix runtime error if GNU Radio was built with -D_GLIBCXX_ASSERTIONS. Fixes #677
closed 2022-12-07 by sensille