gnss-sdr/gnss-sdr #681: Segmentation Fault on Raspbian Bullseye

alexmealster19, 2022-10-28:
I've been working with GNSS-SDR on a raspberry pi and I was able to do everything I wanted on the previous Raspberry Pi OS: buster. However, when I upgrade to the newest OS, bullseye, I run into a segmentation fault. I was able to entirely install GNSS-SDR. I installed the software packages using sudo apt-get install and then I cloned the gnss-sdr repository and built from source. I did everything normal, except I used cmake -DENABLE_OSMOSDR=ON ../ .   

Everything seems to install properly, but once I try to run any commands (including gnss-sdr --version) I get a segmentation fault. 

Like I said before, I've completed this exact process using buster and it worked perfectly. I read that segmentation faults can mean a corrupted file system or bad microSD card but I've been able to run gnss-sdr on this microSD card since I got the segmentation fault, just not while using Bullseye. I've tried both the main and next branch for GNSS-SDR. I've tried upgrading from buster to bullseye without flashing so gnss-sdr can remain in place but I still get a segmentation fault. 

I would really appreciate any insight possible as to how I can navigate this issue!

vladisslav2011, 2022-10-28:
Hello.
Could you obtain a backtrace and post it here?

1. Run gnss-sdr under gdb
2. Wait until it crashes
3. Issue a `bt` command
4. Post output here

alexmealster19, 2022-10-31:
Thank you for the quick response! I'm sorry I didn't get back to you sooner I didn't have access to a computer this weekend. I followed your instructions and have pasted the back trace below. 

`For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gnss-sdr...
(No debugging symbols found in gnss-sdr)
(gdb) run
Starting program: /usr/local/bin/gnss-sdr 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0xb50023c8 in boost::filesystem::detail::directory_iterator_increment(boost::filesystem::directory_iterator&, boost::system::error_code*) ()
   from /lib/arm-linux-gnueabihf/libboost_filesystem.so.1.71.0
(gdb) bt
#0  0xb50023c8 in boost::filesystem::detail::directory_iterator_increment(boost::filesystem::directory_iterator&, boost::system::error_code*) ()
   from /lib/arm-linux-gnueabihf/libboost_filesystem.so.1.71.0
#1  0xb4a1df24 in boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, unsigned int, boost::system::error_code*) () from /lib/arm-linux-gnueabihf/libboost_filesystem.so.1.74.0
#2  0xb6233444 in gr::prefs::_sys_prefs_filenames[abi:cxx11]() () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#3  0xb6234978 in gr::prefs::prefs() () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#4  0xb6234c54 in gr::prefs::singleton() () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#5  0xb6229664 in gr::configure_default_loggers(log4cpp::Category*&, log4cpp::Category*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#6  0xb62984a0 in ?? () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#7  0xb629b868 in ?? () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#8  0xb629534c in ?? () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#9  0xb61cba74 in ?? () from /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2
#10 0xb6fdd328 in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0xbefff2d4, env=env@entry=0xbefff2dc) at dl-init.c:72
#11 0xb6fdd430 in call_init (env=<optimized out>, argv=<optimized out>, argc=<optimized out>, l=<optimized out>) at dl-init.c:30
#12 _dl_init (main_map=0xb6fff9c8, argc=1, argv=0xbefff2d4, env=0xbefff2dc) at dl-init.c:119
#13 0xb6fcca94 in _dl_start_user () from /lib/ld-linux-armhf.so.3
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 

`

I also got the following warning when using make 

`/usr/bin/ld: warning: libboost_serialization.so.1.71.0, needed by /usr/lib/arm-linux-gnueabihf/libuhd.so, may conflict with libboost_serialization.so.1.74.0
/usr/bin/ld: warning: libboost_thread.so.1.71.0, needed by /usr/lib/arm-linux-gnueabihf/libuhd.so, may conflict with libboost_thread.so.1.74.0
`

vladisslav2011, 2022-10-31:
You have 2 different boost versions installed: 1.71.0 and 1.74.0. Loading boost shared libraries of different versions into the same process usually results in bad things happening...
This is not Gqrx problem.
Remove either Boost 1.74.0 or Boost 1.71.0, run `ldconfig` and try again.

alexmealster19, 2022-10-31:
What command should I use to uninstall a specific version of boost? 
And will I need to recompile gnss-sdr or should it be able to run?

vladisslav2011, 2022-10-31:
Something like
`sudo apt remove libboost-thread1.74.0 libboost-thread1.74-dev libboost-filesystem1.74.0 libboost-filesystem1.74-dev libboost-serialization.1.74.0 libboost-serialization.1.74-dev`
may work.

You'll have to rebuild GNSS-SDR.

alexmealster19, 2022-11-01:
Thank you for the help so far! I'm still having issues cleaning up the boost library however. 

If I type 
`sudo apt remove libboost-thread1.74.0 libboost-thread1.74-dev libboost-filesystem1.74.0 libboost-filesystem1.74-dev libboost-serialization1.74.0 libboost-serialization1.74-dev`

I get 

`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  bladerf fonts-lyx gdal-data graphviz libairspy0 libairspyhf1 libbladerf2
  libboost-date-time1.74.0 libboost-program-options1.71.0 libboost-test1.71.0
  libcdt5 libcfitsio9 libcgraph6 libcharls2 libcppunit-1.15-0 libcppunit-dev
  libdap27 libdapclient6v5 libepsilon1 libfreesrp0 libfreexl1 libfyba0 libgdal28
  libgeos-3.9.0 libgeos-c1v5 libgeotiff5 libglfw3 libgsl25 libgslcblas0 libgsm1-dev
  libgts-0.7-5 libgts-bin libgvc6 libgvpr2 libhamlib4 libhdf4-0-alt libheif1
  libhidapi-libusb0 libjs-jquery-ui libkmlbase1 libkmldom1 libkmlengine1
  liblab-gamut1 liblbfgsb0 liblimesuite20.10-1 libmariadb3 libminizip1 libmirisdr0
  libnetcdf18 libodbc1 libogdi4.1 libopengl0 libosmosdr0 libpathplan4 libpq5
  libproj19 libqhull8.0 libqt5opengl5 libqwt-qt5-6 librtaudio6 librtlsdr0 librttopo1
  libsoapysdr0.7 libspatialite7 libtecla1 libthrift-0.13.0 libthrift-dev
  liburiparser1 libvolk2-bin libvolk2-dev libvolk2.4 libxerces-c3.2 limesuite-udev
  mariadb-common mysql-common odbcinst odbcinst1debian2 proj-bin proj-data
  pyqt5-dev-tools python-matplotlib-data python3-click-plugins python3-cycler
  python3-dateutil python3-decorator python3-gdal python3-kiwisolver
  python3-matplotlib python3-networkx python3-pydot python3-pygraphviz
  python3-pyparsing python3-pyqt5.qtopengl python3-pyqt5.qwt python3-pyqtgraph
  python3-scipy python3-sip python3-thrift python3-yaml python3-zmq
  qtbase5-dev-tools qtchooser rtl-sdr soapyosmo-common0.7 soapysdr0.7-module-airspy
  soapysdr0.7-module-audio soapysdr0.7-module-bladerf soapysdr0.7-module-hackrf
  soapysdr0.7-module-lms7 soapysdr0.7-module-redpitaya soapysdr0.7-module-remote
  soapysdr0.7-module-rtlsdr thrift-compiler uhd-host
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  gnuradio gnuradio-dev gr-fcdproplus gr-fosphor gr-iqbal gr-osmosdr
  libboost-date-time-dev libboost-date-time1.74-dev libboost-filesystem-dev
  libboost-filesystem1.74-dev libboost-filesystem1.74.0 libboost-serialization-dev
  libboost-serialization1.74-dev libboost-serialization1.74.0 libboost-thread-dev
  libboost-thread1.74-dev libboost-thread1.74.0 libgnuradio-analog3.8.2
  libgnuradio-audio3.8.2 libgnuradio-blocks3.8.2 libgnuradio-channels3.8.2
  libgnuradio-digital3.8.2 libgnuradio-dtv3.8.2 libgnuradio-fcdproplus3.8.0
  libgnuradio-fec3.8.2 libgnuradio-fft3.8.2 libgnuradio-filter3.8.2
  libgnuradio-fosphor3.8.0 libgnuradio-iqbalance3.8.0 libgnuradio-osmosdr0.2.0
  libgnuradio-pmt3.8.2 libgnuradio-qtgui3.8.2 libgnuradio-runtime3.8.2
  libgnuradio-trellis3.8.2 libgnuradio-uhd3.8.2 libgnuradio-video-sdl3.8.2
  libgnuradio-vocoder3.8.2 libgnuradio-wavelet3.8.2 libgnuradio-zeromq3.8.2
0 upgraded, 0 newly installed, 39 to remove and 45 not upgraded.
After this operation, 132 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 138679 files and directories currently installed.)
Removing gr-fosphor (3.8~2.2d4fe78-1+b5) ...
Removing gnuradio-dev:armhf (3.8.2.0-14) ...
Removing gnuradio (3.8.2.0-14) ...
Removing gr-fcdproplus (3.8~20190817-3+b4) ...
Removing gr-iqbal (0.38-4+b4) ...
Removing gr-osmosdr (0.2.2-1+b3) ...
Removing libboost-date-time-dev:armhf (1.74.0.3+b1) ...
Removing libboost-thread-dev:armhf (1.74.0.3+b1) ...
Removing libboost-thread1.74-dev:armhf (1.74.0-9) ...
Removing libboost-date-time1.74-dev:armhf (1.74.0-9) ...
Removing libboost-filesystem-dev:armhf (1.74.0.3+b1) ...
Removing libboost-filesystem1.74-dev:armhf (1.74.0-9) ...
Removing libgnuradio-wavelet3.8.2:armhf (3.8.2.0-14) ...
Removing libgnuradio-zeromq3.8.2:armhf (3.8.2.0-14) ...
Removing libboost-serialization-dev:armhf (1.74.0.3+b1) ...
Removing libboost-serialization1.74-dev:armhf (1.74.0-9) ...
Removing libboost-serialization1.74.0:armhf (1.74.0-9) ...
Removing libgnuradio-osmosdr0.2.0:armhf (0.2.2-1+b3) ...
Removing libgnuradio-uhd3.8.2:armhf (3.8.2.0-14) ...
[... 58 lines cut ...]
-- GNURADIO_FFT_FOUND = 
-- Required GNU Radio Component: FFT missing!
-- Checking for GNU Radio Module: FILTER
--  * INCLUDES=GNURADIO_FILTER_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_FILTER_LIBRARIES_gnuradio-filter-NOTFOUND
-- GNURADIO_FILTER_FOUND = 
-- Required GNU Radio Component: FILTER missing!
-- Checking for GNU Radio Module: ANALOG
--  * INCLUDES=GNURADIO_ANALOG_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_ANALOG_LIBRARIES_gnuradio-analog-NOTFOUND
-- GNURADIO_ANALOG_FOUND = 
-- Required GNU Radio Component: ANALOG missing!
-- Checking for GNU Radio Module: UHD
--  * INCLUDES=GNURADIO_UHD_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_UHD_LIBRARIES_gnuradio-uhd-NOTFOUND
-- GNURADIO_UHD_FOUND = 
-- Required GNU Radio Component: UHD missing!
-- Could NOT find GNURADIO (missing: GNURADIO_RUNTIME_FOUND) 
-- Could NOT find GPSTK (missing: GPSTK_LIBRARY GPSTK_INCLUDE_DIR) 
CMake Error at CMakeLists.txt:694 (message):
  GNU Radio v requires C++17.  Please update your environment.

-- Configuring incomplete, errors occurred!
See also "/home/bgrg-pi/Desktop/Mealey/gnss-sdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/bgrg-pi/Desktop/Mealey/gnss-sdr/build/CMakeFiles/CMakeError.log".`

vladisslav2011, 2022-11-01:
Try removing boost-1.71.0 libs and headers:
```
sudo apt build-dep gnss-sdr
sudo apt remove libboost-thread1.71.0 libboost-thread1.71-dev libboost-filesystem1.71.0 libboost-filesystem1.71-dev libboost-serialization.1.71.0 libboost-serialization.1.71-dev
```
And rebuild GNSS-SDR.

alexmealster19, 2022-11-03:
I am still running into issues. If I run the two commands you recommended, all the dependencies are built for gnss-sdr and then when I use apt to remove the older libboost it wants to remove other importat packages (gr- osmosdr gnu radio etc).

`sudo apt remove libboost-thread1.71.0 libboost-thread1.71-dev libboost-filesystem1.71.0 libboost-filesystem1.71-dev libboost-serialization1.71.0 libboost-serialization1.71-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'libboost-filesystem1.71-dev' is not installed, so not removed
Package 'libboost-serialization1.71-dev' is not installed, so not removed
Package 'libboost-thread1.71-dev' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libairspy0 libairspyhf1 libbladerf2 libboost-chrono1.71.0 libboost-date-time1.71.0 libcppunit-1.15-0 libcppunit-dev
  libfreesrp0 libgnuradio-analog3.8.2 libgnuradio-audio3.8.2 libgnuradio-blocks3.8.2 libgnuradio-channels3.8.2
  libgnuradio-digital3.8.2 libgnuradio-dtv3.8.2 libgnuradio-fcdproplus3.8.0 libgnuradio-fec3.8.2 libgnuradio-fft3.8.2
  libgnuradio-filter3.8.2 libgnuradio-iqbalance3.8.0 libgnuradio-pmt3.8.2 libgnuradio-qtgui3.8.2
  libgnuradio-runtime3.8.2 libgnuradio-trellis3.8.2 libgnuradio-video-sdl3.8.2 libgnuradio-vocoder3.8.2
  libgnuradio-wavelet3.8.2 libgnuradio-zeromq3.8.2 libgsl25 libgslcblas0 libgsm1-dev libhidapi-libusb0 liblbfgsb0
  libqt5opengl5 libqwt-qt5-6 librtlsdr0 libsoapysdr0.7 libthrift-0.13.0 libthrift-dev libvolk2-bin libvolk2-dev
  libvolk2.4 pyqt5-dev-tools python3-click-plugins python3-decorator python3-pyqtgraph python3-scipy python3-sip
  python3-thrift python3-yaml python3-zmq qtbase5-dev-tools qtchooser thrift-compiler
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  gnuradio gnuradio-dev gr-iio gr-osmosdr libboost-filesystem1.71.0 libboost-serialization1.71.0 libboost-thread1.71.0
  libgnuradio-iio1 libgnuradio-osmosdr0.2.0 libgnuradio-uhd3.8.2 libuhd-dev libuhd3.15.0
0 upgraded, 0 newly installed, 12 to remove and 1 not upgraded.
After this operation, 116 MB disk space will be freed.
Do you want to continue? [Y/n] 
`

What I did instead was I removed all boost1.71 libraries manually using     

`sudo dpkg -r --force-depends "libboost1.71-dev"` and so forth.

But then during sudo make I run into issues where certain libraries depend on libboost1.71. 

I'm confused as to why this is happening only for bullseye and not for buster. And why does this not seem like a common issue? I'm installing software packages as instructed using 

`sudo apt-get install build-essential cmake git pkg-config libboost-dev libboost-date-time-dev \
       libboost-system-dev libboost-filesystem-dev libboost-thread-dev libboost-chrono-dev \
       libboost-serialization-dev liblog4cpp5-dev libuhd-dev gnuradio-dev gr-osmosdr \
       libblas-dev liblapack-dev libarmadillo-dev libgflags-dev libgoogle-glog-dev \
       libgnutls-openssl-dev libpcap-dev libmatio-dev libpugixml-dev libgtest-dev \
       libprotobuf-dev protobuf-compiler python3-mako`

vladisslav2011, 2022-11-03:
It looks like GNU Radio depends on two different versions of boost. This should not happen. Packaging issue...
I'll install bullseye and test.

vladisslav2011, 2022-11-03:
I've instaled debian bullseye, installed gnss-sdr build dependencies `apt build-dep gnss-sdr`, built and tested GNSS-SDR 'next' branch. It works. I can't confirm this issue.

alexmealster19, 2022-11-04:
Okay well that is good news. Wonder if my issue stems from being on a raspberry pi. I'm going to reflash with bullseye and try again. Were your steps as follows below?

`sudo apt build-dep gnss-sdr`

`git clone https://github.com/gnss-sdr/gnss-sdr`

`git checkout next`

`cd gnss-sdr/build`

`cmake ..`

`sudo make`

`sudo make install`

`gnss-sdr --version`

jwmelto, 2022-11-04:
This just came across the gnu-radio mailing list. Related? 
https://lists.gnu.org/archive/html/discuss-gnuradio/2022-11/msg00014.html

vladisslav2011, 2022-11-04:
@alexmealster19 I'm usually running GNSS-SDR from it's build directory without installation: `src/main/gnss-sdr -c ...`.
You may install bullseye into a chroot using debootstrap without altering your main system.

@jwmelto No. It does not look related. This crash is a result of constructing `boost::filesystem::directory_iterator` using a generator from one version of libboost_filesystem and incrementing it using a method from another version of libbost_filesystem...

alexmealster19, 2022-11-07:
@vladisslav2011 can you elaborate on what you mean when you say you are running it from the build directory? Do you mean you're just running the executable in the build folder instead of doing the `sudo make install` step?

And I'll look into using a chroot... I'm assuming it will work using a raspberry pi as well?

vladisslav2011, 2022-11-07:
Yes. I'm skipping `sudo make install`.
Chroot should work on any linux distribution/platform. It's lightweight alternative to docker if you want to test different distribution.

alexmealster19, 2022-11-07:
@vladisslav2011 so I was able to use chroot to run bullseye separately on the raspberry pi and gnss-sdr worked perfectly. However, my final goal is to be able to run gnss-sdr is real time using a hackrf and I'm not sure how to communicate with usb devices while in the chroot. I will continue looking into the issue.

What is baffling me is the fact that I've successfully installed gnss-sdr dozens of times at this point. I've done it on both buster and now on bullseye. However, when I try the exact same steps on raspbian bullseye I get a segmentation fault. Is there anyone who has been able to successfully run gnss-sdr on raspbian bullseye?

vladisslav2011, 2022-11-07:
> how to communicate with usb devices while in the chroot.

Mmmm... Have you tried mounting a minimal set of linux 'system' directories:
```
mount --bind /dev yourchroot/dev
chroot yourchroot mount -t sysfs none /sys
chroot yourchroot mount -t proc none /proc
chroot yourchroot mount -t tmpfs none /dev/shm
chroot yourchroot mount -t devpts none /dev/pts
```

alexmealster19, 2022-11-14:
This worked! Not exactly how I want to be running gnss-sdr. Eventually I want to be able to send serial commands to the pi that utilize gnss-sdr and this might be difficult using chroot. But I guess I'll cross that bridge when I get there. 

I'm going to open a new issue regarding gnss-sdr on raspbian bullseye. I've now tried on multiple raspberry pis, sd cards and installation methods while running into a segmentation fault every time.

@vladisslav2011 thank you so much for helping me troubleshoot!

closed 2024-10-11 by carlesfernandez