gnss-sdr/gnss-sdr #682: TWO_BIT_PACKED_FILE_SIGNAL_SOURCE loses lock constantly

Horsekiro, 2022-10-31:
I am using TWO_BIT_PACKED_FILE_SIGNAL_SOURCE implementation, and here is my configuration:

GNSS-SDR.internal_fs_sps=2560000

SignalSource.implementation=Two_Bit_Packed_File_Signal_Source
SignalSource.filename=/home/young/work/out1.DAT  ;
SignalSource.item_type=byte
SignalSource.big_endian_bytes=false
SignalSource.sample_type=real
SignalSource.sampling_frequency=20480000;
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.enable_throttle_control=false

SignalConditioner.implementation=Signal_Conditioner

DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=float

InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.input_item_type=float
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.45
InputFilter.band2_begin=0.55
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=20480000
InputFilter.IF=5499998.47412109
InputFilter.decimation_factor=8

Resampler.implementation=Pass_Through
Resampler.item_type=gr_complex

Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.sampled_ms=1
Acquisition_1C.pfa=0.015
;Acquisition_1C.pfa=0.01
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=500
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=/home/young/acquisition

Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=45.0;
Tracking_1C.dll_bw_hz=2.0;
Tracking_1C.order=3;
Tracking_1C.dump=true
Tracking_1C.dump_filename=/home/young/track

TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

Observables.implementation=Hybrid_Observables
[... 386 lines cut ...]
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 23 (Block III)
Current receiver time: 5 min 4 s
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 05 (Block IIR-M)
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 05 (Block IIR-M)
Current receiver time: 5 min 5 s
Loss of lock in channel 6!
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 5 min 7 s
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 24 (Block IIF)
Current receiver time: 5 min 8 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 5 min 9 s
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 19 (Block IIR)
Current receiver time: 5 min 12 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Stopping GNSS-SDR, please wait!

Can someone help me analyze how to modify my configuration to get the correct result?

vladisslav2011, 2022-10-31:
Hello.
I looks, like your filter is too wide...
Try this 'automatic' InputFilter configuration:
```
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.input_item_type=float
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.filter_type=lowpass
InputFilter.sampling_frequency=20480000
InputFilter.IF=5499998.47412109
InputFilter.decimation_factor=8
```

Isn't doppler_step too large?
```
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=500
```
It may be better to set doppler_step to 100...200Hz and doppler_max to 5000Hz:
```
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=200
```

It may be better to enable extended correlations:
```
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.pll_bw_narrow_hz=5.0
Tracking_1C.dll_bw_narrow_hz=1.50
```

Horsekiro, 2022-11-01:
> Hello. It looks, like your filter is too wide... Try this 'automatic' InputFilter configuration:
> 
> ```
> InputFilter.implementation=Freq_Xlating_Fir_Filter
> InputFilter.input_item_type=float
> InputFilter.output_item_type=gr_complex
> InputFilter.taps_item_type=float
> InputFilter.filter_type=lowpass
> InputFilter.sampling_frequency=20480000
> InputFilter.IF=5499998.47412109
> InputFilter.decimation_factor=8
> ```
> 
> Isn't doppler_step too large?
> 
> ```
> Acquisition_1C.doppler_max=10000
> Acquisition_1C.doppler_step=500
> ```
> 
> It may be better to set doppler_step to 100...200Hz and doppler_max to 5000Hz:
> 
> ```
> Acquisition_1C.doppler_max=5000
> Acquisition_1C.doppler_step=200
> ```
> 
> It may be better to enable extended correlations:
> 
> ```
> Tracking_1C.extend_correlation_symbols=10
> Tracking_1C.pll_bw_narrow_hz=5.0
> Tracking_1C.dll_bw_narrow_hz=1.50
> ```

Thank you for your advice. I changed the configuration and here is my new configuration:

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2560000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Two_Bit_Packed_File_Signal_Source
SignalSource.filename=/home/young/work/out1.DAT  ;
SignalSource.item_type=byte
SignalSource.big_endian_bytes=false
SignalSource.sample_type=real
SignalSource.sampling_frequency=20480000;
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=float

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.input_item_type=float
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.filter_type=lowpass
InputFilter.sampling_frequency=20480000
InputFilter.IF=5499998.47412109
InputFilter.decimation_factor=8

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through
Resampler.item_type=gr_complex

[... 267 lines cut ...]
Current receiver time: 4 min 56 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 26 (Block IIF)
Number of samples between Acquisition and Tracking = 7892 ( 0.00308281 s) PULL-IN Doppler [Hz] = 400. 
PULL-IN Code Phase [samples] = 1797
Current receiver time: 4 min 57 s
Current receiver time: 4 min 58 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 19 (Block IIR)
Number of samples between Acquisition and Tracking = -4972 ( -0.00194219 s) PULL-IN Doppler [Hz] = -2400.
PULL-IN Code Phase [samples] = 2828
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Number of samples between Acquisition and Tracking = 15128 ( 0.00590937 s) PULL-IN Doppler [Hz] = 1800. 
PULL-IN Code Phase [samples] = 1462
Loss of lock in channel 5!
Current receiver time: 4 min 59 s
Loss of lock in channel 2!
Current receiver time: 5 min 0 s
Current receiver time: 5 min 12 s
Loss of lock in channel 3!
Current receiver time: 5 min 13 s
Current receiver time: 5 min 14 s
Loss of lock in channel 0!
Current receiver time: 5 min 15 s
Stopping GNSS-SDR, please wait!

The lock still loses. I let the code output the tracking variable's value to help you better analyze it. What's the matter with this situation?

vladisslav2011, 2022-11-01:
`InputFilter.IF` should be equal to real signal frequency - tuning center frequency. If you are tuning your frontend to 1569420000, then `InputFilter.IF` should be equal to 6000000. If you are tuning your frontend to 1581420000, then `InputFilter.IF` should be equal to -6000000.

Horsekiro, 2022-11-02:
> `InputFilter.IF` should be equal to real signal frequency - tuning center frequency. If you are tuning your frontend to 1569420000, then `InputFilter.IF` should be equal to 6000000. If you are tuning your frontend to 1581420000, then `InputFilter.IF` should be equal to -6000000.

Thank you for your explanation. I changed the InputFilter.IF  into 16369000.
The result looks like this:
Processing file /home/young/work/out1.DAT, which contains 6437732404 samples (1609433101 bytes)
GNSS signal recorded time to be processed: 314.242 [s]
Current receiver time: 1 s
Current receiver time: 9 s
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 31 (Block IIR-M)
Number of samples between Acquisition and Tracking = 27073 ( 0.0105754 s)PULL-IN Doppler [Hz] = -1000. PULL-IN Code Phase [samples] = 1872
Current receiver time: 10 s
Current receiver time: 12 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 20 (Block IIR)
Number of samples between Acquisition and Tracking = -952 ( -0.000371875 s)PULL-IN Doppler [Hz] = 2400. PULL-IN Code Phase [samples] = 3839
Current receiver time: 13 s
Current receiver time: 17 s
Loss of lock in channel 1!
Current receiver time: 18 s
Current receiver time: 23 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 11 (Block III)
Number of samples between Acquisition and Tracking = -952 ( -0.000371875 s)PULL-IN Doppler [Hz] = 2600. PULL-IN Code Phase [samples] = 4583
Current receiver time: 24 s
Current receiver time: 25 s
Loss of lock in channel 7!
Current receiver time: 26 s
Current receiver time: 28 s
Loss of lock in channel 5!
Current receiver time: 29 s
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 28 (Block IIR)
Number of samples between Acquisition and Tracking = -2560 ( -0.001 s)PULL-IN Doppler [Hz] = -1400. PULL-IN Code Phase [samples] = 4502
Current receiver time: 33 s
Current receiver time: 37 s
Loss of lock in channel 6!
Current receiver time: 38 s
Current receiver time: 42 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 24 (Block IIF)
Number of samples between Acquisition and Tracking = -2660 ( -0.00103906 s)PULL-IN Doppler [Hz] = -2800. PULL-IN Code Phase [samples] = 3793
Current receiver time: 43 s
Current receiver time: 47 s
Loss of lock in channel 1!
Current receiver time: 48 s
Current receiver time: 1 min 20 s
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 17 (Block IIR-M)
Number of samples between Acquisition and Tracking = 2465 ( 0.000962891 s)PULL-IN Doppler [Hz] = -400. PULL-IN Code Phase [samples] = 2424
Current receiver time: 1 min 21 s
Current receiver time: 1 min 23 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 15 (Block IIR-M)
Number of samples between Acquisition and Tracking = 3872 ( 0.0015125 s)PULL-IN Doppler [Hz] = 2600. PULL-IN Code Phase [samples] = 2030
Current receiver time: 1 min 24 s
Current receiver time: 1 min 36 s
Loss of lock in channel 3!
Current receiver time: 1 min 37 s
Current receiver time: 1 min 38 s
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 03 (Block IIF)
Number of samples between Acquisition and Tracking = 7375 ( 0.00288086 s)PULL-IN Doppler [Hz] = -2800. PULL-IN Code Phase [samples] = 244
Current receiver time: 1 min 39 s
Loss of lock in channel 5!
Current receiver time: 1 min 40 s
Current receiver time: 1 min 51 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 23 (Block III)
Number of samples between Acquisition and Tracking = -601 ( -0.000234766 s)PULL-IN Doppler [Hz] = -3600. PULL-IN Code Phase [samples] = 3396
Current receiver time: 1 min 52 s
Current receiver time: 1 min 54 s
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 10 (Block IIF)
Number of samples between Acquisition and Tracking = -3364 ( -0.00131406 s)PULL-IN Doppler [Hz] = -3000. PULL-IN Code Phase [samples] = 2917
Loss of lock in channel 1!
Current receiver time: 1 min 55 s
Current receiver time: 1 min 56 s
Loss of lock in channel 0!
[... 81 lines cut ...]
Loss of lock in channel 2!
Current receiver time: 4 min 20 s
Current receiver time: 4 min 22 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 17 (Block IIR-M)
Number of samples between Acquisition and Tracking = 7188 ( 0.00280781 s)PULL-IN Doppler [Hz] = -800. PULL-IN Code Phase [samples] = 519
Current receiver time: 4 min 23 s
Current receiver time: 4 min 38 s
Loss of lock in channel 2!
Current receiver time: 4 min 39 s
Current receiver time: 4 min 50 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 24 (Block IIF)
Number of samples between Acquisition and Tracking = 21006 ( 0.00820547 s)PULL-IN Doppler [Hz] = 3200. PULL-IN Code Phase [samples] = 316
Current receiver time: 4 min 51 s
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 08 (Block IIF)
Number of samples between Acquisition and Tracking = 5002 ( 0.00195391 s)PULL-IN Doppler [Hz] = 2200. PULL-IN Code Phase [samples] = 1740
Current receiver time: 4 min 52 s
Current receiver time: 5 min 6 s
Loss of lock in channel 5!
Current receiver time: 5 min 7 s
Loss of lock in channel 7!
Current receiver time: 5 min 8 s
Current receiver time: 5 min 15 s
Stopping GNSS-SDR, please wait!

Which value shall I modify？Does configuration file contain the param corresponding to RF-end sampling frequency？

Horsekiro, 2022-11-02:
Dear vladisslav2011
         I'm really appreciated for your advice and explanation. I'm a graduate student and my tutor asks me to use the gnss-sdr run the IF data we collected in the school. However, this is my first time using this software and the program is so complicated that I really don't know how to solve the problem as I've raised in the github. In this way, your answer is a great help to me. I've tried to modify the InputFilter.IF and other params'value according to your advice, but the lock still loses and received no nav message. In this way, I'm thinking maybe there are something wrong with IF data. In fact, the format of IF data we used  RF-end to collect were shown like this:

          Each sampling clock outputs 4-bit IF data, of which two are GPS data and two are Beidou data. Each byte stored in the IF data file contains two groups of sampling data. According to the time sequence, the two groups of sampling data are divided into Samp0 and Samp1. The .data file I opened in the UltraEdit looks like this:

           Each couple in the picture means one byte. For example, the first byte 58, after expanding in binary, the result is 0101 1000. Here is my thinking: According to the data format, if I want the IF data only contains the GPS data, all I have to do is collecting the  5th、4th and 1st、0 of each byte, and transform them into a new one. In the example of the first byte 58, that is 01 and 00. After change, they become 0100.The second byte 8A, expanded in binary:1000 1010, after change we get 0010.Combined with the first result 0100, that is 0100 0010,which is  42. Here is the new .dat file after the change in this way:

In the new .data file each byte contains 4 2-bit GPS data. So the new .dat file only contains half bytes of the old one, and that is exactly correct after I opened the file's property option to see the bytes they contained. So I set the new .dat file as the Signalsource.filename  and choose the Two_Bit_Packed_File_Signal_Source as SignalSource.implementation in the configuration.Do you think my thinkings right? Beacuse the lock keeps losing,I really don't know if the .data file or the configuration I made have the problem. If the way I deal with the .dat file is correct then I can make sure all I have to do is make the configuration correct. Can you help me analyze this question?I'll be really grateful for your advice. Best wishes for you.
                                                                                                                                                                                                                                                                        Sincerely
                                                                                                                                                                                                                                                                      Songyue Ma

抄送: "Never say ***@***.******@***.***&gt;;

Horsekiro, 2022-11-02:
> `InputFilter.IF` should be equal to real signal frequency - tuning center frequency. If you are tuning your frontend to 1569420000, then `InputFilter.IF` should be equal to 6000000. If you are tuning your frontend to 1581420000, then `InputFilter.IF` should be equal to -6000000.

Dear vladisslav2011
         I'm really appreciated for your advice and explanation. I'm a graduate student and my tutor asks me to use the gnss-sdr run the IF data we collected in the school. However, this is my first time using this software and the program is so complicated that I really don't know how to solve the problem as I've raised in the github. In this way, your answer is a great help to me. I've tried to modify the InputFilter.IF and other params'value according to your advice, but the lock still loses and received no nav message. In this way, I'm thinking maybe there are something wrong with IF data. In fact, the format of IF data we used  RF-end to collect were shown like this:[https://mail.qq.com/cgi-bin/viewfile?f=3ED2A22717A40C769BBD590D585146E4A69FC2C763FD833955A72403A7AC32C24D7DF7A9D423F575CCB86F5CAF12E16FE0B1E845BDCFB39DD873FC06D534F6B2F74C55696ED6EEC337334CD704ED3FB317C8B513CB3317FE4EC20C794A414E28&mailid=ZL0702-TRv4V5_Vu3sZKg5tEKaITcb&sid=eC5RcbmJT4-tc1c0&net=1370918411](url)
       Each sampling clock outputs 4-bit IF data, of which two are GPS data and two are Beidou data. Each byte stored in the IF data file contains two groups of sampling data. According to the time sequence, the two groups of sampling data are divided into Samp0 and Samp1. The .data file I opened in the UltraEdit looks like this:[https://mail.qq.com/cgi-bin/viewfile?f=3ED2A22717A40C769BBD590D585146E4A69FC2C763FD833955A72403A7AC32C24D7DF7A9D423F575CCB86F5CAF12E16FE0B1E845BDCFB39D265441301FDCAA28934B5BA26663A58C2675F51F368617384F509FCC2A61055600CF06FE1E7B8FE4&mailid=ZL0702-TRv4V5_Vu3sZKg5tEKaITcb&sid=eC5RcbmJT4-tc1c0&net=1370918411](url)
      Each couple in the picture means one byte. For example, the first byte 58, after expanding in binary, the result is 0101 1000. Here is my thinking: According to the data format, if I want the IF data only contains the GPS data, all I have to do is collecting the  5th、4th and 1st、0 of each byte, and transform them into a new one. In the example of the first byte 58, that is 01 and 00. After change, they become 0100.The second byte 8A, expanded in binary:1000 1010, after change we get 0010.Combined with the first result 0100, that is 0100 0010,which is  42. Here is the new .dat file after the change in this way:
[https://mail.qq.com/cgi-bin/viewfile?f=3ED2A22717A40C769BBD590D585146E4A69FC2C763FD833955A72403A7AC32C24D7DF7A9D423F575CCB86F5CAF12E16FE0B1E845BDCFB39D9B7F4AA5DEBBCD3044B908D510A4D825AEE9E99E3D5009B350F69A2DFF5EA2FD53C18C8AC2FE8CC4&mailid=ZL0702-TRv4V5_Vu3sZKg5tEKaITcb&sid=eC5RcbmJT4-tc1c0&net=1370918411](url)
     In the new .data file each byte contains 4 2-bit GPS data. So the new .dat file only contains half bytes of the old one, and that is exactly correct after I opened the file's property option to see the bytes they contained. So I set the new .dat file as the Signalsource.filename  and choose the Two_Bit_Packed_File_Signal_Source as SignalSource.implementation in the configuration.Do you think my thinkings right? Beacuse the lock keeps losing,I really don't know if the .data file or the configuration I made have the problem. If the way I deal with the .dat file is correct then I can make sure all I have to do is make the configuration correct. Can you help me analyze this question?I'll be really grateful for your advice. Best wishes for you.
Sincerely                                                                                                                                                                                                                                                           Songyue Ma

Horsekiro, 2022-11-02:
Dear vladisslav2011
         I'm really appreciated for your advice and explanation. I'm a graduate student and my tutor asks me to use the gnss-sdr run the IF data we collected in the school. However, this is my first time using this software and the program is so complicated that I really don't know how to solve the problem as I've raised in the github. In this way, your answer is a great help to me. I've tried to modify the InputFilter.IF and other params'value according to your advice, but the lock still loses and received no nav message. In this way, I'm thinking maybe there are something wrong with IF data. In fact, the format of IF data we used  RF-end to collect were shown like this:
[image]
       Each sampling clock outputs 4-bit IF data, of which two are GPS data and two are Beidou data. Each byte stored in the IF data file contains two groups of sampling data. According to the time sequence, the two groups of sampling data are divided into Samp0 and Samp1. The .data file I opened in the UltraEdit looks like this:
[image]
      Each couple in the picture means one byte. For example, the first byte 58, after expanding in binary, the result is 0101 1000. Here is my thinking: According to the data format, if I want the IF data only contains the GPS data, all I have to do is collecting the  5th、4th and 1st、0 of each byte, and transform them into a new one. In the example of the first byte 58, that is 01 and 00. After change, they become 0100.The second byte 8A, expanded in binary:1000 1010, after change we get 0010.Combined with the first result 0100, that is 0100 0010,which is  42. Here is the new .dat file after the change in this way:
[image]
     In the new .data file each byte contains 4 2-bit GPS data. So the new .dat file only contains half bytes of the old one, and that is exactly correct after I opened the file's property option to see the bytes they contained. So I set the new .dat file as the Signalsource.filename  and choose the Two_Bit_Packed_File_Signal_Source as SignalSource.implementation in the configuration.Do you think my thinkings right? Beacuse the lock keeps losing,I really don't know if the .data file or the configuration I made have the problem. If the way I deal with the .dat file is correct then I can make sure all I have to do is make the configuration correct. Can you help me analyze this question?I'll be really grateful for your advice. Best wishes for you.
Sincerely                                                                                                                                                                                                                                                           Songyue Ma

jwmelto, 2023-01-06:
What is the source of this data file? It seems to me that, at the IF level, you cannot know that one sample is GPS and the next is Beidou. It is just detected RF energy.

I see processing inconsistencies when using un-throttled file playback (`SignalSource.enable_throttle_control=false`) because GNU Radio will read the file as fast as possible. I prefer the greater determinism of using real-time output.

DISCLAIMER: I've never used the 2-bit packed format

carlesfernandez, 2024-10-11:
We close this issue, hopefully the problem is solved in the most recent version of the software. Please feel free to reopen it if the problem arises again.

closed 2024-10-11 by carlesfernandez