gnss-sdr/gnss-sdr #684: FPGA offloading

devmonkZA, 2022-11-08:
Hello , I have been working through the code , building source , on Laptop (x86) , Raspberry Pi 3 and 4 , and a cross compiled using Yocto for ADRV9361 Zynq board, also managed to build for FPGA but I cannot find the code for the UIO driver and HDL code , I see it 
"experimental", is the code not open like the rest of the project , if not then where could I find it and how much will it cost ? 

Awesome project , Thanks for sharing , learned so much about GPS, thanks to your hard work :)

ef-newspace, 2022-11-15:
Guess I'm on my own, came across an interesting project for anyone else going through this 

http://oscillator-imp.com/dokuwiki/doku.php?id=research
https://github.com/oscimp
https://github.com/oscimp/fpga_ip

gpswizard, 2023-06-02:
> Hello , I have been working through the code , building source , on Laptop (x86) , Raspberry Pi 3 and 4 , and a cross compiled using Yocto for ADRV9361 Zynq board, also managed to build for FPGA but [...]

hello for the raspberry pi3 and 4 did you cross compile with yocto?

carlesfernandez, 2024-10-11:
The FPGA IP cores, user manuals, and example project are commercially available at https://navposproducts.cttc.es/

closed 2024-10-11 by carlesfernandez

devmonkZA, 2024-10-14:
Hello Carles , This is great news thanks for the the update , I will be in contact shortly

devmonkZA, 2024-10-14:
> hello for the raspberry pi3 and 4 did you cross compile with yocto?

Hello , Sorry only saw you message now , Yes  , but you could probably use Buildroot , Bootlin has toolchains as well 
https://www.buildroot.org/docs.html
https://toolchains.bootlin.com/