gnss-sdr/gnss-sdr #685: L2C solution accuracy?

jmfriedt, 2022-11-09:
I am investigating L2C using gnss-sdr v.0.17 (binary distribution from Debian/sid), processing IQ streams collected by an Ettus Research B210 SDR connected to a Septentrio dual-frequency active antenna. Tuning on L2C according to the configuration file below, the PVT solution found is
```
Current receiver time: 11 min 36 s
New GPS CNAV message received in channel 1: ephemeris from satellite GPS PRN 29 (Block IIR-M)
Satellite GPS PRN 29 (Block IIR-M) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
New GPS CNAV message received in channel 3: ephemeris from satellite GPS PRN 11 (Block III)
Satellite GPS PRN 11 (Block III) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
Current receiver time: 11 min 37 s
New GPS CNAV message received in channel 0: ephemeris from satellite GPS PRN 32 (Block IIF)
Satellite GPS PRN 32 (Block IIF) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
First position fix at 2022-Nov-09 18:43:20.286289 UTC is Lat = 46.659 [deg], Long = 6.4587 [deg], Height= 5384.16 [m]
Position at 2022-Nov-09 18:43:20.586289 UTC using 4 observations is Lat = 46.659026319 [deg], Long = 6.458715050 [deg], Height = 5380.590 [m]
```
which is awkward since a solution is reached (good) despite the warnings on the unhealthy CNAV (same issue than L5 as long as L2C is not officially commissioned?)  but I am definitely not at a 5380 m altitude and the lat/lon is off by a few hundred kilometers. My location according to a dual frequency UBlox Zed F9P receiver is 47.25157523 °N, 5.99319628 E.

Here is the configuration file I used for real time processing of L2C:
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2500000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=UHD_Signal_Source
SignalSource.device_address=
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2500000
SignalSource.freq=1227600000
SignalSource.clock_source=external
SignalSource.gain=40
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Pass_Through

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=gr_complex

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through
Resampler.dump=false
Resampler.dump_filename=../data/resampler.dat
Resampler.item_type=gr_complex
Resampler.sample_freq_in=2500000
Resampler.sample_freq_out=2500000

;######### CHANNELS GLOBAL CONFIG ############
Channels_2S.count=4
Channels.in_acquisition=1

Channel.signal=2S

Channel0.signal=2S
Channel1.signal=2S
Channel2.signal=2S
Channel3.signal=2S

;######### ACQUISITION GLOBAL CONFIG ############

;# GPS L2C M
Acquisition_2S.implementation=GPS_L2_M_PCPS_Acquisition
Acquisition_2S.item_type=gr_complex
Acquisition_2S.pfa=0.015
;Acquisition_2S.pfa=0.001
Acquisition_2S.doppler_max=3500
Acquisition_2S.doppler_min=-3500
Acquisition_2S.doppler_step=500
Acquisition_2S.max_dwells=1
Acquisition_2S.dump=false
Acquisition_2S.dump_filename=./acq_dump.dat

Tracking_2S.implementation=GPS_L2_M_DLL_PLL_Tracking
Tracking_2S.item_type=gr_complex
Tracking_2S.pll_bw_hz=2.0;
Tracking_2S.dll_bw_hz=0.25;
Tracking_2S.order=2;
Tracking_2S.early_late_space_chips=0.5;
Tracking_2S.dump=false
Tracking_2S.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_2S.implementation=GPS_L2C_Telemetry_Decoder
TelemetryDecoder_2S.dump=false

;######### OBSERVABLES CONFIG ############.
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=PPP_Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=false
PVT.dump_filename=./PVT
```
obviously strongly inspired from https://github.com/gnss-sdr/gnss-sdr/blob/main/conf/gnss-sdr_GPS_L2C_USRP_X300_realtime.conf with a few adjustments to allow for real time processing on the Atom processor (lower sampling rate, lower Doppler range and larger Doppler step).
Has L2C been validated on genuine signals?

Thanks, JM

vladisslav2011, 2022-11-09:
Hello.
I think, you have received corrupt ephemeris.
4 satellites at 4 channels is a bare minimum. It is better to have 6...10 channels.
500 Hz doppler step is too large for L2. 50 Hz, or even 20 Hz should be much better.

chalupabatmann, 2023-04-13:
@jmfriedt , I know that it has been awhile, but where you able to get this resolved? I am currently working in the next branch and I cannot seem to be able to get anything (e.g. NAV messages, fixes, etc) out of the GPS L2C receiver, either from file based playback or streaming. I am pretty familiar with using UHD for Ettus products and have tried multiple approaches to the L2C with no luck. 
I have had much success working with most of the other receivers so if you have any advise or config files that are currently working for you please let me know. 

Thanks!.

chalupabatmann, 2023-04-14:
> Hey @jmfriedt, as of hash 306c566d4839913f2e1365d7629ae9f1c8cb4594, and a new antenna placement, I was able to verify L2C running live and receive continuous and accurate fixes on my location.

jmfriedt, 2023-07-05:
I just ran a 30 min measurement and processing the L2C band using the configuration below. I have the awkward result the initial fix is mostly erroneous
```
Current receiver time: 3 min 9 s
Current receiver time: 3 min 10 s
Current receiver time: 3 min 11 s
Current receiver time: 3 min 12 s
Current receiver time: 3 min 13 s
New GPS CNAV message received in channel 4: ephemeris from satellite GPS PRN 09 (Block IIF)
Satellite GPS PRN 09 (Block IIF) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
New GPS CNAV message received in channel 12: ephemeris from satellite GPS PRN 11 (Block III)
Satellite GPS PRN 11 (Block III) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
First position fix at 2023-Jul-05 18:52:07.760000 UTC is Lat = 47.043 [deg], Long = 5.75286 [deg], Height= 30074.2 [m]
New GPS CNAV message received in channel 10: ephemeris from satellite GPS PRN 30 (Block IIF)
The RINEX Navigation file header has been updated with UTC and IONO info.
Position at 2023-Jul-05 18:52:08.000000 UTC using 4 observations is Lat = 47.043020965 [deg], Long = 5.752901012 [deg], Height = 30066.770 [m]
Velocity: East: 4.402 [m/s], North: -2.183 [m/s], Up = -4.500 [m/s]
Satellite GPS PRN 30 (Block IIF) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
Current receiver time: 3 min 14 s
Position at 2023-Jul-05 18:52:08.500000 UTC using 4 observations is Lat = 47.042950654 [deg], Long = 5.752941883 [deg], Height = 30071.408 [m]
```
but after a few more seconds I have the good surprise of being given the correct location
```
Current receiver time: 4 min 1 s
New GPS CNAV message received in channel 10: ephemeris from satellite GPS PRN 30 (Block IIF)
Satellite GPS PRN 30 (Block IIF) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
Position at 2023-Jul-05 18:52:55.500000 UTC using 4 observations is Lat = 47.042151781 [deg], Long = 5.755489756 [deg], Height = 29868.937 [m]
Velocity: East: 4.386 [m/s], North: -2.293 [m/s], Up = -4.150 [m/s]
New GPS CNAV message received in channel 5: ephemeris from satellite GPS PRN 04 (Block III)
Satellite GPS PRN 04 (Block III) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
Position at 2023-Jul-05 18:52:56.000000 UTC using 4 observations is Lat = 47.042143098 [deg], Long = 5.755530170 [deg], Height = 29856.520 [m]
Velocity: East: 4.390 [m/s], North: -2.304 [m/s], Up = -4.107 [m/s]
New GPS CNAV message received in channel 7: ephemeris from satellite GPS PRN 06 (Block IIF)
Satellite GPS PRN 06 (Block IIF) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
New GPS CNAV message received in channel 4: ephemeris from satellite GPS PRN 09 (Block IIF)
Satellite GPS PRN 09 (Block IIF) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
Current receiver time: 4 min 2 s
New GPS CNAV message received in channel 12: ephemeris from satellite GPS PRN 11 (Block III)
Satellite GPS PRN 11 (Block III) does not report a healthy status in the CNAV message, use PVT solutions at your own risk.
Position at 2023-Jul-05 18:52:56.500000 UTC using 4 observations is Lat = 47.251670509 [deg], Long = 5.993532899 [deg], Height = 338.389 [m]
Velocity: East: 0.010 [m/s], North: -0.032 [m/s], Up = 0.043 [m/s]
Position at 2023-Jul-05 18:52:57.000000 UTC using 4 observations is Lat = 47.251634003 [deg], Long = 5.993509238 [deg], Height = 339.577 [m]
Velocity: East: -0.036 [m/s], North: 0.015 [m/s], Up = -0.029 [m/s]
```
so not sure how to dismiss erroneous solutions (despite the many warnings on their suitability).

Here is the config file:
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=5000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/data/4.bin
SignalSource.item_type=ishort
SignalSource.sampling_frequency=5000000
SignalSource.freq=1227600000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=../data/signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex

InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
[... 35 lines cut ...]
Tracking_2S.order=2;
Tracking_2S.early_late_space_chips=0.5;
Tracking_2S.dump=false
Tracking_2S.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_2S.implementation=GPS_L2C_Telemetry_Decoder
TelemetryDecoder_2S.dump=false

;######### OBSERVABLES CONFIG ############.
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
;PVT.positioning_mode=PPP_Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
;PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
;PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.rinex_output_enabled=true
PVT.rinex_name=L2output.rinex
PVT.rinex_version=2
```

jwmelto, 2023-07-06:
I wonder if we should add an option to exclude unhealthy PRNs from position calculation?

Seems like your problem is a 4 satellite solution with 2 unhealthy satellites.

jmfriedt, 2023-07-06:
I wonder if the unhealthy status is meaningful: indeed https://www.gps.gov/systems/gps/modernization/civilsignals/ seems to have changed the status of L2C from the unhealthy status of L5 to healthy now that the status has become "Pre-operational signal with message set "healthy""
Thanks.

carlesfernandez, 2024-11-15:
We added a `PVT.use_unhealthy_sats`, set to `false` by default, that allows to control the usage of satellites marked as unhealthy.

closed 2024-11-15 by carlesfernandez