gnss-sdr/gnss-sdr #686: GNSS-SDR not running on Raspbian Bullseye

alexmealster19, 2022-11-14:
I've been working with GNSS-SDR for almost a year now and recently I've been installing it onto a raspberry pi. I was able to get it fully functional on a raspberry pi 4, running raspbian buster (the raspberry pi version of debian buster). However, this all falls apart once raspbian bullseye (the newest version of raspberry pi os) is installed. Gnss-sdr still installs without much fuss but it will not run, and returns a segmentation fault. 

I have tried running gnss-sdr on multiple pis and on multiple sd cards. I have tried installing from apt-get, from snap and building from source. I have successfully installed gnss-sdr on many other machines so I'm confident my methodology of building from source is sound. 

I've been troubleshooting for a few weeks now and I believe the error is due to a dependency on different versions of boost. I'm not sure why this issue only comes up in raspbian bullseye and not debian bullseye. 

For the time being, I have a work around and I might just roll my pi back to the latest os. So I'm making this issue just to let people know that raspbian bullseye and gnss-sdr are incompatible. Or, if anyone has been able to get the two to work together if you could please share your details so that I may try to replicate them.

jwmelto, 2023-01-06:
Is this a duplicate of #681 ?

Fangrf, 2024-01-18:
@ alexmealster19 Have you solved the problem?I used ADI Kuiper Linux.It is accully Raspbian Bullseye.So I have met the same problem.

Fangrf, 2024-01-18:
@alexmealster19 Have you solved the problem?I used ADI Kuiper Linux.It is accully Raspbian Bullseye. I have met the same problem.

closed 2024-10-11 by carlesfernandez