gnss-sdr/gnss-sdr #700: Sampling frequency for GPS L1

fener1efes2, 2023-01-10:
Why do not we set sampling frequency as 2 times of 1.023 MHz or 4 times of 1.023 MHz?

Does not sampling frequency of 4 MHz deteriorate the signal?

vladisslav2011, 2023-01-10:
Hello.
Setting sampling frequency to 2M, or, better 2048k improves acquisition performance as LibFFTW builds much better plan at even widths, than at odd widths and it shows best performance at 2^n widths.
4Mhz is more, than GPS L1C/A chip rate (1.023M), so it's fine. 
GNSS-SDR will work even at 800ksps sample rate with slightly lower C/N0, than at optimal sample rate.

fener1efes2, 2023-01-10:
So, sampling frequency does not have to be integer multiplication of 1.023M since LibFFTW library "corrects" the sampling frequency in that way.

If I write sampling freq as 4 MHz into a config file, then LibFFTW changes sampling frequency to 4.092 MHz which is 4 times 1.023 MHz? Am I correct?

vladisslav2011, 2023-01-11:
No.
The signal replica is interpolated to the GNSS-SDR internal sampling frequency. LibFFTW is used to optimize acquisition - single FFT/iFFT instead of multiple convolutions.
https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/acquisition/gnuradio_blocks/pcps_acquisition.cc#L601-L865