gnss-sdr/gnss-sdr #710: Support for non integer sample rate

minecraft2048, 2023-03-01:
My frontend samples the signal at 16.368 Msps and decimated by factor of 7 internally, which results in output sample rate of 2.338285714.. Msps. The problem is that GNSS-SDR only support integer sample rate on `GNSS-SDR.internal_fs_sps`. 

If I try to use `Mmse_Resampler`, the input sample rate also only support integer sample rate, so I can't use that to resample my signal to integer sample rate. 

Currently I'm using a external gnuradio flowgraph with rational resampler block to resample it to 16.368/5 = 3273600 sps, but its clunky because I need to make intermediate file.

I have several suggestions to solve this:

1.  Natively support non integer sample rate in GNSS-SDR, this is the best because we skip resampling, which make processing faster. I don't know whether its possible
2. Support non integer sample rate parameter on `Mmse_Resampler`, which the underlying gnuradio fractional resampler support anyway: https://wiki.gnuradio.org/index.php/Fractional_Resampler because it accepts resampling ratio

vladisslav2011, 2023-03-01:
Hello.
Have you tried setting
```
Resampler.sample_freq_in=5
Resampler.sample_freq_out=7
GNSS-SDR.internal_fs_sps=3273600
```
?
It should work in your case.

jwmelto, 2023-03-01:
First, I'd listen to @vladisslav2011 because he's pretty smart.

I would also suggest that your external GNU Radio flowgraph can end in a `zeromq_pub_sink` and your GNSS-SDR flowgraph can use a [ZMQ Signal Source](https://gnss-sdr.org/docs/sp-blocks/signal-source/#reading-data-from-a-zeromq-stream).

GNSS-SDR does make some prescriptive design decisions; your solution of solving it externally may, in many cases, be the most reasonable.

minecraft2048, 2023-03-02:
> Hello. Have you tried setting
> 
> ```
> Resampler.sample_freq_in=5
> Resampler.sample_freq_out=7
> GNSS-SDR.internal_fs_sps=3273600
> ```
> 
> ? It should work in your case.

@vladisslav2011  It doesn't work, if I used these parameters:
```
Resampler.implementation=Mmse_Resampler
Resampler.sample_freq_in=5
Resampler.sample_freq_out=7
Resampler.item_type=gr_complex
GNSS-SDR.internal_fs_sps=3273600
```

it doesn't want to run 
```
^C⏎                                                                                                                                                                                                                byakuya@ifrit-hpc ~/w/g/kea_test [SIGINT]> gnss-sdr --config_file=kea2.conf
Initializing GNSS-SDR v0.0.17.git-next-0d60e4639 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
RF Channels: 1
CONFIGURATION WARNING: Parameters GNSS-SDR.internal_fs_sps and Resampler.sample_freq_out are not set to the same value!
Configuration error in Resampler block, implementation Mmse_Resampler. The error was:
firdes check failed: 0 < fa <= sampling_freq / 2
GNSS-SDR program ended.

```

vladisslav2011, 2023-03-02:
@minecraft2048 
OK. Mmse_resampler does only decimation, but you want interpolation, so this config will work

```
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=5
Resampler.sample_freq_out=7
Resampler.item_type=gr_complex
GNSS-SDR.internal_fs_sps=3273600
```
Test result:
```
gnss-sdr -c gps+gal-file2.conf
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.17.git-hide_pvt-5cfb37ac8 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
RF Channels: 1
CONFIGURATION WARNING: Parameters GNSS-SDR.internal_fs_sps and Resampler.sample_freq_out are not set to the same value!
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Processing file /media/vlad/TOSHIBA/iq/gqrx_20230302_090911_1575420000_2338285_8.raw, which contains 1130364928 samples (1130364928 bytes)
GNSS signal recorded time to be processed: 141.246 [s]
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 14 (Block III)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 17 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 19 (Block IIR)
Current receiver time: 2 s
Tracking of Galileo E1C signal started on channel 23 for satellite Galileo PRN E24 (Block FOC-FM5)
Tracking of Galileo E1C signal started on channel 12 for satellite Galileo PRN E25 (Block FOC-FM16)
Current receiver time: 3 s
Current receiver time: 4 s
Tracking of Galileo E1C signal started on channel 13 for satellite Galileo PRN E10 (Block FOC-FM24)
Tracking of Galileo E1C signal started on channel 15 for satellite Galileo PRN E12 (Block IOV-FM2)
Current receiver time: 5 s
Current receiver time: 6 s
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Current receiver time: 12 s
Current receiver time: 13 s
GPS L1 C/A tracking bit synchronization locked in channel 5 for satellite GPS PRN 19 (Block IIR)
Enabled 10 ms extended correlator in channel 5 for satellite GPS PRN 19 (Block IIR)
GPS L1 C/A tracking bit synchronization locked in channel 4 for satellite GPS PRN 17 (Block IIR-M)
Enabled 10 ms extended correlator in channel 4 for satellite GPS PRN 17 (Block IIR-M)
GPS L1 C/A tracking bit synchronization locked in channel 1 for satellite GPS PRN 14 (Block III)
Enabled 10 ms extended correlator in channel 1 for satellite GPS PRN 14 (Block III)
Current receiver time: 14 s
Current receiver time: 15 s
Galileo E1C secondary code locked in channel 15 for satellite Galileo PRN E12 (Block IOV-FM2)
Enabled 16 ms extended correlator in channel 15 for satellite Galileo PRN E12 (Block IOV-FM2)
Current receiver time: 16 s
Current receiver time: 17 s
Current receiver time: 18 s
Current receiver time: 19 s
@ 328  =>  New GPS NAV message received in channel 4: subframe 3 from satellite GPS PRN 17 (Block IIR-M)
@ 328  =>  New GPS NAV message received in channel 5: subframe 3 from satellite GPS PRN 19 (Block IIR)
@ 324  =>  New GPS NAV message received in channel 1: subframe 3 from satellite GPS PRN 14 (Block III)
Current receiver time: 20 s
Tracking of Galileo E1C signal started on channel 17 for satellite Galileo PRN E11 (Block IOV-PFM)
```

closed 2024-10-25 by carlesfernandez