gnss-sdr/gnss-sdr #712: Galileo E1 dll_pll_veml_tracking error report

aardric, 2023-03-31:
W0331 13:30:16.832731 10684 dll_pll_veml_tracking.cc:2057] Bad any_cast: pmt_any_ref: wrong_type {0 0.74409}

gnss-sdr version 0.0.17.git-next-79ebc32d8

I received this after an update  3a235e485..79ebc32d8.
Full disclosure: I did do an o/s system re-install before this upgrade.

jwmelto, 2023-04-01:
Does your source have any tags? This is, I think caused by a tag in the stream that wasn't expected. The tag handling is very brittle.

aardric, 2023-04-01:
I am using the SignalSource.implementation=Limesdr_Signal_Source.  This does not use tags that I know of.  I replaced  the signal source of a working configuration using a file source (2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat) keeping everything else the same except for the adapter (Ishort vrs gr_complex).  The error in question no longer appears with the limeSDR. However, the "q+enter" sequence now quits whereas it had no effect previously but the line SignalSource.implementation=Limesdr_Signal_Source is now ineffective (executes until "q+enter"). I will look at this with a clear head over this weekend and try some additional experiments. I will hopefully be able to provide more useful information.

aardric, 2023-04-02:
Apparently, the error appears with higher sample rates. I see that gr-limesdr  supports tags and I suspect that at high sample rates (above 8 Msps in my case) something unexpected appears in the stream.

closed 2023-10-12 by aardric