gnss-sdr/gnss-sdr #713: Firehose SDR. Anyone tried it?

AgileEngineeringLLC, 2023-05-11:
Has anyone worked with the Firehose 3-channel SDR? It sells for $800, and has an ethernet interface. Much much cheaper than something like a USRP N210. If you've tried it, what did you like or dislike about it for gnss processing? Thanks!!

wmszeliga, 2024-06-06:
I just got a GNSS Firehose. I've been successful in collecting data and acquiring and tracking signals from file (though not using GNSS-SDR yet). Communication is over a local Ethernet connection, and it really is a firehose. The data rate is close to 70 MHz and positioning directly from file captures is likely not feasible without immediate signal decimation just due to the shear size of the file captures. I've written a libpcap-based listener that outputs to a FIFO pipe, but so far I cannot get GNSS-SDR to process from the pipe due to back-pressure. I'd be happy to help work on a SignalSource for the GNSS Firehose, or at least a recipe to get it working with an existing source, if anyone is interested in partnering with me on this (I'm new to GNSS-SDR).

AgileEngineeringLLC, 2024-06-06:
I use a modified version of gnss-sdrlib (C code) with a bladerf, and it runs great in real-time. i am now hoping to get a Firehose API up and running, and interfaced with gnss-sdrlib. gnss-sdrlib is much much simpler (easier to follow and edit) than gnss-sdr, but both are terrific programs. i wonder if your libpcap-listener could be interfaced with gnss-sdrlib? did you write it in C or C++?

wmszeliga, 2024-06-06:
Ah, I see the GNSS-SDRLIB repo now and it seems as if you have posted some instructions there as well. My libpcap listener is in C.

AgileEngineeringLLC, 2024-06-06:
Great, since your listener is in C, it seems like it would be fairly easy to integrate it in to gnss-sdrlib. If you're interested in partnering on giving it a try, reach out to me at don.kelly at mac dot com. i can help you get gnss-sdrlib up and running, and clue you in on how it works and a few of the challenges i've seen with it. but it's a very cool program, as is gnss-sdr!

carlesfernandez, 2024-09-20:
At this time we do not support the GNSS Firehose. We have it in the roadmap.

closed 2024-09-20 by carlesfernandez

AgileEngineeringLLC, 2024-09-20:
thx CarlosDonDon KellyLinkedIn: www.linkedin.com/in/kellydakCell: 281-221-28534403 Orange Leaf CourtHouston, TX 77059

AgileEngineeringLLC, 2025-04-08:
Hi Carlos,

I hope 2025 is going well for you! Any updates on interfacing with the GNSS Firehose RF front end?

Thanks,

Don

Don Kelly

LinkedIn: www.linkedin.com/in/kellydak
Cell:  281-221-2853
5 Riverton Ct, Greer SC 29650