gnss-sdr/gnss-sdr #718: How can I use HackRF with non-zero IF frequency signal ?

gnsssdr, 2023-06-29:
I'm new to the GNSS world. I have successfully run gnss-sdr with HackRF and obtained the correct position. Additionally, I have been able to record data using GNU Radio Companion and HackRF, and then process it offline with gnss-sdr. However, when I attempt to use a non-zero IF frequency in either method, it does not work at all. What do I need to do for this purpose?

vladisslav2011, 2023-06-29:
Hello.
You may use `freq_xlating_fir_filter` `IF` parameter to compensate IF offset.
https://gnss-sdr.org/docs/sp-blocks/input-filter/#implementation-freq_xlating_fir_filter

gnsssdr, 2023-06-30:
Hello vladislav 2011 
Thanks for reply. 
Does HackRF capable of sampling signal in non-zero IF at all? Based on this block diagram it has not band pass filter at mixer output, it has only two tunable LPF.
[image]

vladisslav2011, 2023-06-30:
HackRF has non-zero first IF (2300...2700MHz), but band-pass filter between RRFC5072 and MAX283 is not shown on this diagram.

[image]

HackRF does not have complex band-pass filters after second mixer inside of MAX2837, so the
 ADC will always sample at zero, but it is possible to tune several MHz above or below frequency of interest to get rid of DC spike / cellular base stations products.