gnss-sdr/gnss-sdr #719: New to .dat files any idea how to extract data from tracking_ch_xx.dat files?

Imtiaz08, 2023-07-06:
Hi there, I wanted to study the .dat files using python and MATLAB and I have read them but they are giving me data in binary format. Is there any way how to recover information from these files please?

ShahZahidKhan, 2023-07-25:
Have you extracted the IQ data? Could you explain what you mean by recovering the information? Would you like to plot the IQ data?

Imtiaz08, 2023-07-25:
I have extracted the IQ data successfully and even got results in MATLAB using .mat files but now I want to learn data extraction from .dat channel files not signal files

fgallardo, 2023-12-29:
The best you can do is to check the SC of the GNSS-SDR where those files are written and then generate in Python the equivalent code to read those. That is what I did for TLM dat files.

In which stage (and with what constellation) are you dumping those files: ACQ, TRK, TLM? So 

I may be able to point you to the relevant part, then it is a matter to look into how to read those in Python (opening the binary file and using struct to decode the parts as they were written).

closed 2024-09-13 by carlesfernandez