gnss-sdr/gnss-sdr #720: ZCU102+FMCOMMS5 software only

DiogoBaptistaBosch, 2023-07-11:
Hi,

I am trying to run the GNSS SDR software only on a ZCU102+FMCOMMS5 runinning geniux. I made sure that the front-end is working (used ADI Reference design to generate the boot files and with IIO-oscilloscope made sure it was receiving data). Since the GNSS SDR only has support for FMCOMMS2/3/4, I tried to change the devices names under /src/algorithms/signal_source/libs/ad9361_manager.h, so that one of the FMCOMMS5 transceivers would be used as if is was the FMCOMMS2. But, when running the the library using the FMCOMMS2 scripts it does not detect the front-end. Does anyone as some insights on this? Do I have to modify even more stuff? Or just tell me if what I'm trying to do is not possible?

Best regards,
Diogo Baptista

carlesfernandez, 2024-09-13:
In software-only mode, the FMCOMMS5 board is currently not supported. However, you can try to modify the FMCOMMS2 signal source to access the FMCOMMS5 signals.

closed 2024-09-13 by carlesfernandez

felixei, 2024-10-30:
Has there been an update on this matter? We are working on a sort-of FMCOMMS5 architecture. Before I dig into modifying the signals myself, I would be great to hear if there has been any updates by @DiogoBaptistaBosch