gnss-sdr/gnss-sdr #722: UHD clock source but not time?

jmfriedt, 2023-08-25:
https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/signal_source/adapters/uhd_signal_source.cc#L214
sets the UHD frequency source to internal, external or MIMO but does not act on the 1-PPS source. Is this intended or is the
``uhd_source_->set_time_source(clock_source_);`` missing?
Thanks

carlesfernandez, 2024-09-13:
Signal sample time tags are not used in the whole process. Consequently, we left the time source not configured.

closed 2024-09-13 by carlesfernandez

jmfriedt, 2024-09-13:
Isn't the time synchronization of UHD used to align the samples with 1-PPS reference input? This is the reason I am considering time-alignment in addition to frequency synchronization: for using GNSS for time and frequency transfer, it might be easier to collect samples aligned on reference 1-PPS I believe.