gnss-sdr/gnss-sdr #723: How to processing data from TEXBAT?

cai321, 2023-08-26:
Hello
I want to get one. The conf file to handle from TEXBAT scene. Bin data file

lmaynard6, 2024-01-10:
Yes, it does work with TEXBAT, just tested it. You have to adjust the sample rate to match that of TEXBAT. Using the base configuration from https://gnss-sdr.org/my-first-fix/ you have to adjust two items to 25M/S

SignalSource.sampling_frequency=25000000 ;# Match that of TEXBAT
Resampler.sample_freq_in=25000000 ;# Resample rate that resamples from 25MS to 2M/S

Let me know if you have other questions, I am working on OAKBAT now.

closed 2024-09-13 by carlesfernandez

teansyfeal, 2024-09-21:
Hey guys! Is there any guide or info on how do I process textbat ds7.bin data via python? Im new to gnss-sdr and trying to correctly transform 40+gb binary file via python..

lmaynard6, 2024-09-22:
Sure, you can load the data via Numpy fromfile, below is the code I used.

Once loaded you can conduct any tranforms you like, but simulating the GPS processing chain can be challenging.  In addition you will need to scale the results appropriately.  You may have better luck doing exploration via GNURadio. You can send me a message if you need some additional help.

Data is loaded from binary file. Note, these are large datasets and require a significant amount of system memory. On a system with low memory numpy memmap may be used to index files on disk and read portions.
The function to convert from int16 to complex64 is faster than dtype conversion and uses less memory.

def process_bin(directory, file):
    gps_raw = np.fromfile(directory + file, dtype=np.int16)
    print("Loaded signal with shape " + str(gps_raw.shape))
    gps_complex = gps_raw.astype(np.float32).view(np.complex64)
    np.save(directory + file[0:-4] + "_complex.npy", gps_complex)
    print("Processing for file " + file + " complete.")