gnss-sdr/gnss-sdr #727: Spurious "Tracking" messages appear even if no antenna and no signal present and gain set to 0

RobertSwirsky, 2023-09-06:
I wasn't getting NAV messages but I was getting Tracking messages. I assumed either my signal wasn't strong enough or, maybe, my B200's internal oscillator wasn't stable enought.

After trying everything, I set the gain to zero in the config file and disconnected the GPS antenna. I still get the "Tracking" messages!

```
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Current receiver time: 1 s
Current receiver time: 2 s
Current receiver time: 3 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Current receiver time: 4 s
Current receiver time: 5 s
Current receiver time: 6 s
Current receiver time: 7 s
Current receiver time: 8 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 07 (Block IIR-M)
Current receiver time: 9 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 09 (Block IIF)
Current receiver time: 10 s
```
I didn't expect this! Is it just matching noise, or is there something wrong?

JohnNatHaysys, 2023-09-19:
We have a system which generates a simulated satellite signal. Even when switched off we can see tracking messages. I assumed this was seeing a real satellite giving a strong signal. It usually lost lock, and never resulted in a Nav message. However, more recent testing suggest the acquisition pfa value may result in a 'lock' on noise. Any views on a suitable pfa value?

RobertSwirsky, 2023-09-19:
It's definately noise! I can turn the gain to zero and put a 50 ohm resistor on the input and I still get tracking messages. (I'll never see a NAV though.)

I'm having trouble using GNSS-SDR on real signals in general. I need a _very_ strong signal to get it to work.  I can get it to work from a simulator, but I don't have a solid enough signal from real satellites (in a place where a uBlox module will have no problem).

I tried from a USRP 200, 205, and an x310 and none of them was able to hear. Yes, I'm using a gain antenna, and I have a bias-T injector to supply power (and I verified that the antenna was getting power).

JohnNatHaysys, 2023-09-20:
Well, I can confirm that a RTL-SDR V3 dongle connected to a good external antenna (Tallysman triband) works perfectly well. Replacing the antenna with a 50R and a DC block ( to prevent the bias T driving 50R) resulted in an occasional tracking message - perhaps one or two a minute on average. The loss of lock occurs 15s later - which is a function of some timeout counters, even if there is a brief DLL match.

RobertSwirsky, 2023-09-20:
Thanks! I tried an RTS-SDR using the GNSS-SDRLIB software and didn't get a signal either but I was using one of those little square black cheap antennas. (I verified it was working by putting the simulator into a dummy load near the antenna).

Maybe all the antennas I have are junk! I have some old "marine" style antennas I thought were good, but maybe not (though they work with uBlox). I'll try getting a the same "Tallysman triband" you have and put it on a pole on the roof.

JohnNatHaysys, 2023-10-18:
Our gnss-sdr based system appears to detect non existent signals! Locking on noise I can understand but we also see GPS NAV messages with a constant pseudorange from channels that are not supplied with a signal. It seems to be a function of the configuration. Anyone else seen this?

jmfriedt, 2023-11-29:
Tracking is meaningless, just telling you that a given SV was assigned to a given tracking channel and searching for the signal. It has no relation to the actual existence of a signal, it is the internal state of gnss-sdr.
Only when seeing
``GPS L1 C/A tracking bit synchronization locked in channel ...``
can you consider that at actual was seen, strong enough if you reach
``New GPS NAV message received in channel``.