gnss-sdr/gnss-sdr #734: Galileo E1 tracking carr_error_filt_hz = carrier_doppler_hz?

aardric, 2023-09-30:
Within a dump file created by the tracking configuration for Tracking_1B.implementation=Galileo_E1_DLL_PLL_VEML_Tracking
is  tracking carr_error_filt_hz = carrier_doppler_hz to be expected (y/n/depends)?

gnss-sdr version 0.0.18.git-next-fd57c34d5

aardric, 2023-10-12:
This issue was closed prematurely and then reopened.

fgallardo, 2023-12-29:
Hi,

I think one is the error signal from phase discriminator of the PLL and the other, the input to the NCO (i.e. the smoothed error signal that comes out of the filter)

Error signal from the phase discriminator of the PLL -> carr_error_hz 
Output of the Loop filter -> carr_error_filt_hz

[image]

Both blocks are on the right part of this diagram:
https://raw.githubusercontent.com/gnss-sdr/gnss-sdr/next/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/docs/images/VOLK_GNSSSDR_Usage_Example.png

BTW, I guess you already checked, but you can find how these are computed in the LoC 538 and 540 below.
https://github.com/gnss-sdr/gnss-sdr/blob/2c69d2f2135a3bfe7aa841522aaa161d8905f9ac/src/algorithms/tracking/gnuradio_blocks/glonass_l1_ca_dll_pll_tracking_cc.cc#L538

Indeed you can see how it uses the carr_error_filt_hz signal to make the new estimate of the carrier frequency in LoC 542.

aardric, 2023-12-31:
The mat file contains three quantities:
(1) carr_error_hz: Raw carrier error (unfiltered) at the PLL output, in Hz.
(2) carrier_error_filt_hz: Carrier error at the output of the PLL filter, in Hz.
(3) carrier_doppler_hz: Doppler shift, in Hz

I understand the relationship between (1) and (2) as referenced 
above but not the relationship between (2) and (3).  I plan to look into this further after January 10, 2024.  I will update  this comment with my findings.

fgallardo, 2024-01-02:
Ups, so sorry, I think I did not read your question carefully.

If I understand your question, it is about the relationship between the output of the filter of the PLL loop and the new Doppler estimate, no?

Then, I would say, and I'm a hoping that I am getting your point right:

The value of the estimates of the Doppler at tracking stage starts with the estimate from the ACQ block, LoC 651, as it should be:

https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#L651

And then. it gets updated in run_dll_pll() to the value of the filtered frequency error, in LoC 1107
https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/tracking/gnuradio_blocks/dll_pll_veml_tracking.cc#1107

At that moment in time, it should be the same. I assume. 

Does it help somehow? Are you seeing any type of strange behaviour?

Thanks!
Fran

aardric, 2024-01-12:
Given this advice, the quantities in question behave exactly as 
expected. However, I now raise the question as to how they can ever be 
different if  'd_carrier_doppler_hz = d_carr_error_filt_hz' (line # 
1107) is executed when tracking begins (triggering  the start of a dump 
file). I will examine the code more carefully but you have provided a 
good starting point. Thanks for this.

Rick