gnss-sdr/gnss-sdr #744: Observables doesnt give solution, but messages income

fl1ckje, 2023-11-15:
Using hackrf one + gps antenna.
GNSS-SDR receives nav messages, but doesnt provide solution
[image]

[image]
[image]
[image]
`[GNSS-SDR]
;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=8000000
GNSS-SDR.telecommand_enabled=true
GNSS-SDR.telecommand_tcp_port=3333

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=8000000
SignalSource.freq=1602000000
SignalSource.gain=60
SignalSource.rf_gain=50
SignalSource.if_gain=40
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=./glonass/dump_ss.dat
SignalSource.enable_throttle_control=false
SignalSource.osmosdr_args=hackrf,bias=1

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1G.count=10
Channel.signal=1G
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1G.implementation=GLONASS_L1_CA_PCPS_Acquisition
Acquisition_1G.item_type=gr_complex
Acquisition_1G.pfa=0.01
Acquisition_1G.doppler_max=8000
Acquisition_1G.doppler_step=250
Acquisition_1G.coherent_integration_time_ms=2
Acquisition_1G.repeat_satellite=false
Acquisition_1G.dump=false
Acquisition_1G.dump_filename=./acq_dump.dat

;######### TRACKING GLONASS CONFIG ############
Tracking_1G.implementation=GLONASS_L1_CA_DLL_PLL_C_Aid_Tracking
Tracking_1G.item_type=gr_complex
Tracking_1G.dump=false
Tracking_1G.dump_filename=./tracking_ch_
Tracking_1G.pll_bw_hz=40.0;
Tracking_1G.dll_bw_hz=1.5;
Tracking_1G.early_late_space_chips=0.5;

;######### TELEMETRY DECODER GLONASS CONFIG ############
TelemetryDecoder_1G.implementation=GLONASS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1G.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat
Observables.enable_carrier_smoothing=false
Observables.smoothing_factor=200

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.enable_rx_clock_correction=false

PVT.output_rate_ms=100
PVT.output_enabled=true
PVT.rtcm_output_file_enabled=false
PVT.gpx_output_enabled=true
PVT.geojson_output_enabled=true
PVT.kml_output_enabled=true
PVT.xml_output_enabled=true
PVT.rinex_output_enabled=true
PVT.nmea_output_file_enabled=true

PVT.output_path=./glonass/others
PVT.gpx_output_path=./glonass/gpx
PVT.kml_output_path=./glonass
PVT.xml_output_path=./glonass
PVT.rinex_output_path=./glonass/rinex

PVT.rinexobs_rate_ms=100
PVT.display_rate_ms=500
PVT.elevation_mask=0
PVT.dump_filename=./glonass/PVT
PVT.nmea_dump_filename=pvt.nmea;
PVT.flag_nmea_tty_port=true;
PVT.nmea_dump_devname=/dev/pts/4
PVT.dump=true
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.enable_monitor=false
;PVT.monitor_client_addresses=127.0.0.1
;PVT.monitor_udp_port=1111

Monitor.enable_monitor=false
;Monitor.decimation_factor=1
;Monitor.client_addresses=127.0.0.1
;Monitor.udp_port=1112

`

vladisslav2011, 2023-11-15:
Hello.
Your config does not match your log.
You have correct Glonass config (too much gain and HackRF internal LNA is enabled, but it may still work with long cable and low gain antenna).
Your log is showing 11 GPS satellites tracking and telemetry is decoded. So your antenna is good. You should get a PVT.
It is possible that bad reflected signals from 1..2 satellites are breaking the solution. Try increasing `PVT.elevation_mask` or enable `PVT.raim_fde` to exclude bad reflected signals.

fl1ckje, 2023-11-15:
> Hello.
> Your config does not match your log.
> You have correct Glonass config (too much gain and HackRF internal LNA is enabled, but it may still work with long cable and low gain antenna).
> Your log [...]

Hello. Yeah, I noticed, that included globass cfg instead of gps, but the problem is the same. I will try to play around PVT ASAP. Thanks)

fl1ckje, 2023-11-20:
> Hello. Your config does not match your log. You have correct Glonass config (too much gain and HackRF internal LNA is enabled, but it may still work with long cable and low gain antenna). Your log [...]

Hello. Used your advice about parameters. HackRF is working with external clock 0,5 ppm (10 MHz). Still got no solution. External clock signal source is NI-5610. What's the problem? The same issue happens when using internal HackRF TCXO (0,1 ppm).
[image]

vladisslav2011, 2023-11-20:
I can't confirm. I'm getting a PVT after ~1 minute of tracking 4 or 5 satellites with current 'next' branch.
You may try to clean your build directory, reconfigure `cmake -DENABLE_OSMOSDR=ON ..` and rebuild `make -j9`.

fl1ckje, 2023-11-20:
> I can't confirm. I'm getting a PVT after ~1 minute of tracking 4 or 5 satellites with current 'next' branch.
> You may try to clean your build directory, reconfigure `cmake -DENABLE_OSMOSDR=ON ..` and rebuild `make -j9`.

I've built project with enabling osmosdr, gnss-sdr recognizes hackrf. Well, I'll do it again.

fl1ckje, 2023-11-20:
> I can't confirm. I'm getting a PVT after ~1 minute of tracking 4 or 5 satellites with current 'next' branch.
> You may try to clean your build directory, reconfigure `cmake -DENABLE_OSMOSDR=ON ..` and rebuild `make -j9`.

Could you tell please, what ppm has got your TCXO?

vladisslav2011, 2023-11-20:
I'm using generic chinese "0.1ppm" TCXO module.
HackRF works well even without a TCXO, but PLL/DLL loops BW should be increased to maintain lock and doppler search range should be increased to ~20000...30000 to compensate the crystal inaccuracy.
GNSS-SDR outputs only used observables, not all received to a NMEA file, so if you are getting some observables in a NMEA file, you should get a PVT. This may be considered a bug...

fl1ckje, 2023-11-20:
> I'm using generic chinese "0.1ppm" TCXO module.
> HackRF works well even without a TCXO, but PLL/DLL loops BW should be increased to maintain lock and doppler search range should be increased to [...]

Good. I saw your video about multichannel receiver (https://youtu.be/ZQs2sFchJ6w?si=GdMLKHHMkq4llGWY). Can I get similar solution with PPP static using, for example, glonass + beidou, glonass + gps?

vladisslav2011, 2023-11-20:
GPS/Galileo+BeiDou fit into HackRF bandwidth. Glonass+GPS/Galileo doesn't fit. So you'll need some better frontend. BladeRF or USRP will do Glonass+BeiDou+GPS/Galileo. https://www.youtube.com/watch?v=HnZkKj9a-QM

fl1ckje, 2023-11-23:
> GPS/Galileo+BeiDou fit into HackRF bandwidth. Glonass+GPS/Galileo doesn't fit. So you'll need some better frontend. BladeRF or USRP will do Glonass+BeiDou+GPS/Galileo. https://www.youtube.com/watch?v=HnZkKj9a-QM

Hello. I got one more question. I'm running gnss-sdr 0.0.18 (clean build) with hackrf and gnss signal simulator (сн-3803м). My conf file: 
```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=8000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=8000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.rf_gain=40
SignalSource.if_gain=30
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=./gps_l1_signal_source.dat
SignalSource.enable_throttle_control=false
SignalSource.osmosdr_args=hackrf,bias=1

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=13
Channel.signal=1C
Channels.in_acquisition=1

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.pfa=0.01
Acquisition_1C.coherent_integration_time_ms=2
Acquisition_1C.doppler_max=5000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=1
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./gps_l1_acquisition.dat

;######### TRACKING GPS CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.dump=true
Tracking_1C.dump_filename=./gps_l1_tracking.dat
Tracking_1C.pll_bw_hz=35.0
Tracking_1C.dll_bw_hz=1.5
Tracking_1C.pll_bw_narrow_hz=2.5
Tracking_1C.dll_bw_narrow_hz=0.5
Tracking_1C.extend_correlation_symbols=10
Tracking_1C.dll_filter_order=2;
Tracking_1C.pll_filter_order=3;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.early_late_space_narrow_chips=0.25

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=true

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true
Observables.dump_filename=./gps_l1_observables.dat
Observables.enable_carrier_smoothing=false
;Observables.smoothing_factor=200

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Static
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.enable_rx_clock_correction=false

PVT.output_rate_ms=100
PVT.output_enabled=true
PVT.rtcm_output_file_enabled=false
PVT.gpx_output_enabled=true
PVT.geojson_output_enabled=true
PVT.kml_output_enabled=true
PVT.xml_output_enabled=true
PVT.rinex_output_enabled=true
PVT.nmea_output_file_enabled=true

PVT.output_path=./others
PVT.gpx_output_path=.gpx
PVT.kml_output_path=./kml
PVT.xml_output_path=./xml
PVT.rinex_output_path=./rinex

PVT.rinexobs_rate_ms=100
PVT.display_rate_ms=500
PVT.elevation_mask=5
PVT.raim_fde=1
PVT.dump_filename=./pvt
PVT.nmea_dump_filename=pvt.nmea;
PVT.flag_nmea_tty_port=true;
PVT.nmea_dump_devname=/dev/pts/4
PVT.dump=true
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.enable_monitor=false

Monitor.enable_monitor=true
Monitor.decimator_factor=50
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=5005
```
Why do I get maximal only 9 observables instead of 11?
For example, 30 min trimble screen doesn't show PRN 05, but it exists it obs. The same situation with PRN 29, 30, 16 and 07. Then ok, got 9 obs, but why not 11? Does it strongly depend on signal power in current case? GPS antenna is in front of signal simulator's antenna.

[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]

vladisslav2011, 2023-11-24:
Really strange.
I think, the signal level is too high.
I'm using a cheap magnetic antenna and I have set gains to 0/32/30.
```
SignalSource.gain=0
SignalSource.rf_gain=32
SignalSource.if_gain=30
```
The antenna is near the window (half of sky is blocked by the building) and I'm getting 17..20 observables (BeiDou+GPS+Galileo).
In your case, gains should be a lot less. Try 0/32/10.
`Acquisition_1C.coherent_integration_time_ms=2` is not really required as you have very strong signal.
`Acquisition_1C.max_dwells=1` may be increased to 4, but it should not change anything as the signal is strong.

fl1ckje, 2023-11-24:
> Really strange.
> I think, the signal level is too high.
> I'm using a cheap magnetic antenna and I have set gains to 0/32/30.
> ```
> SignalSource.gain=0
> SignalSource.rf_gain=32
> SignalSource.if_gain=30
> ``` [...]

Thanks a lot. We made everything fine right now. What's about galileo? Have you got any recommendations about conf file values according to this system? I tried some default setups, got 4-5 observables and 1-2 mins after encountered loss of lock, then lost pvt at all.

vladisslav2011, 2023-11-24:
This config works well for me:
```
;######### ACQUISITION GALILEO CONFIG ############
Acquisition_1B.implementation=Galileo_E1_PCPS_Ambiguous_Acquisition
Acquisition_1B.coherent_integration_time_ms=2
Acquisition_1B.pfa=0.025
Acquisition_1B.doppler_max=5000
Acquisition_1B.doppler_step=200
Acquisition_1B.max_dwells=4
Acquisition_1B.cboc=true
Acquisition_1B.dump=false

;######### TRACKING GALILEO CONFIG ############
Tracking_1B.implementation=Galileo_E1_DLL_PLL_VEML_Tracking
Tracking_1B.extend_correlation_symbols=4
Tracking_1B.item_type=gr_complex
Tracking_1B.pll_bw_hz=30.0
Tracking_1B.dll_bw_hz=2.0
Tracking_1B.pll_bw_narrow_hz=20.0
Tracking_1B.dll_bw_narrow_hz=0.50
Tracking_1B.track_pilot=true
Tracking_1B.enable_fll_pull_in=true;
Tracking_1B.enable_fll_steady_state=false
Tracking_1B.fll_bw_hz=20

;######### TELEMETRY DECODER GALILEO E1B CONFIG ############
TelemetryDecoder_1B.implementation=Galileo_E1B_Telemetry_Decoder
TelemetryDecoder_1B.dump=false
```