gnss-sdr/gnss-sdr #748: E5a Adquisition block is not dumping raw files

fgallardo, 2023-12-29:
Hi,

Happy holidays, all!

I am unsure whether I am doing something wrong here. In short: I am not able to dump E5 ACQ raw files.

What I know:
1. I get a PVT, so the files I am using and the overall thing works.
2. I get dump files from the E5 tracking stage.
3. I checked the SC of the implementation of Galileo_E5a_Noncoherent_IQ_Acquisition_CAF and I would say the dumping part is there.  See https://github.com/gnss-sdr/gnss-sdr/blob/main/src/algorithms/acquisition/gnuradio_blocks/galileo_e5a_noncoherent_iq_acquisition_caf_cc.cc             (LoC 571)

Hence, it has to be something I am doing wrong in the config, the relevant part is:

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_5X.implementation=Galileo_E5a_Noncoherent_IQ_Acquisition_CAF
Acquisition_5X.item_type=gr_complex
Acquisition_5X.coherent_integration_time_ms=1
Acquisition_5X.pfa=0.0003
Acquisition_5X.doppler_max=10000
Acquisition_5X.doppler_step=50
Acquisition_5X.bit_transition_flag=false
Acquisition_5X.max_dwells=1
Acquisition_5X.CAF_window_hz=0
Acquisition_5X.Zero_padding=0
Acquisition_5X.dump=True
Acquisition_5X.dump_filename=./acq_dump.dat

Any clue?

Thanks!
Fran

jwmelto, 2023-12-29:
Just guessing, but two thoughts off the top of my head:

1. Try `true` instead of `True` (I don’t remember the Boolean parsing logic)
2. You didn’t provide any output. But is the current directory (the one from which you started gnss-sdr) writable? I find it safer to use explicit paths for files, or create a script that ensures your are in the directory you mean to be and all command line and environmental settings are consistent.

fgallardo, 2023-12-31:
Hi there, again.

First of all, thanks a lot for the support!

I tried with both True and true, and with a folder under control (I realized the original folder was not correct).

It still does not work (I am running it in a container, but I am sure now the folder should work, as it is the same I am using to place the tracking files, which work). 

Anyway, it is not a big deal, as I eventually sorted what I needed to do out, using tracking files.

Hence, I am closing this one with this comment anyway.

Thanks!
Fran

closed 2023-12-31 by fgallardo