gnss-sdr/gnss-sdr #749: Segmentation fault using plutoSDR on Ubuntu LTS 22.04

sdy623, 2023-12-31:
I am got trouble with my plutoSDR device using this tool. It always shows Segmentation fault when I try to run this program to get data form my device.

Here are some information about my issues

## Scenario
``` [bash]
gnss-sdr --config_file=../gnss-sdr.conf 
```
```
Initializing GNSS-SDR v0.0.18.git-next-6b9ad0332 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
device address: 192.168.137.101
frequency : 1575420000 Hz
sample rate: 3000000 Sps
gain mode: manual
item type: gr_complex
RF Channels: 1
[1]    109598 segmentation fault (core dumped)  gnss-sdr --config_file=../gnss-sdr.conf
```
## Config file

Here is my config file [My config](https://pastebin.com/y0LERX6g)

## Callstack from core dump file

And I used gdb to show the callstack at the error occurred. 
```
(gdb) bt
#0  0x000055870b52017c in std::__shared_ptr<gr::io_signature, (__gnu_cxx::_Lock_policy)2>::__shared_ptr (this=0x7fff1f8a2200)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1152
#1  0x000055870b5201cb in std::shared_ptr<gr::io_signature>::shared_ptr (
    this=0x7fff1f8a2200) at /usr/include/c++/11/bits/shared_ptr.h:150
#2  0x000055870b5201fc in gr::basic_block::input_signature (this=0x0)
    at /usr/include/gnuradio/basic_block.h:162
#3  0x000055870b68c483 in SignalConditioner::connect (this=0x55870ed720d0, 
    top_block=std::shared_ptr<gr::top_block> (use count 2, weak count 1) = {...})
    at /home/***/gnss-sdr/src/algorithms/conditioner/adapters/signal_conditioner.cc:68
#4  0x000055870b500989 in GNSSFlowgraph::connect_signal_conditioners (
    this=0x55870ed60f30)
    at /home/***/gnss-sdr/src/core/receiver/gnss_flowgraph.cc:664
#5  0x000055870b4ffcda in GNSSFlowgraph::connect_desktop_flowgraph (
    this=0x55870ed60f30)
    at /home/***/gnss-sdr/src/core/receiver/gnss_flowgraph.cc:423
#6  0x000055870b4ff65c in GNSSFlowgraph::connect (this=0x55870ed60f30)
    at /home/***/gnss-sdr/src/core/receiver/gnss_flowgraph.cc:369
#7  0x000055870b4c3a4d in ControlThread::run (this=0x55870ed5cc20)
    at /home/***/gnss-sdr/src/core/receiver/control_thread.cc:380
#8  0x000055870b4b6b5b in main (argc=1, argv=0x7fff1f8a2810)
    at /home/***/gnss-sdr/src/main/main.cc:149

```

## System Environment info:

System Version: Ubuntu LTS 22.04 
Kernel Version: 6.2.0-39-generic
Shell: zsh 5.8.1
DE: GNOME 42.9
GNSS-SDR Version: v0.0.18.git-next-6b9ad0332
Gnuradio Version: 3.10.1.1

I would really appreciate any insight possible as to how I can navigate this issue.

jwmelto, 2023-12-31:
You can see in `#2` of your trace that `this==0`, which is pretty much guaranteed to dump core. It's in the SignalConditioner `connect()` call, on the input side. 

I don't know anything about using your input device but when you have an IP address you generally need a port as well. I wonder if the source isn't being properly constructed and just isn't providing an error message?

sdy623, 2024-01-01:
I used the ADALM-Pluto device, its connection protocol based on libido, which just need the up address can work well.

For my configuration file, I copied the signal source config from the [documentation](https://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-plutosdr_signal_source).

Rupak910, 2024-01-31:
I am getting the same error. If you got any solution to your problem, kindly share. 

Thank You

wkazubski, 2024-07-07:
The GNU Radio package used by gnss-sdr should be built with PlutoSDR support enabled too.

carlesfernandez, 2024-10-25:
Could you check if this error has gone with the most recent GNSS-SDR version?

PaulPRob, 2024-10-28:
I can confirm that this error still seems to be in the V 0.19. I too get the seg fault on trying to use the PlutoSDR signal source. My error is identical gives the following. Has been built with Debug mode

```
ppr@paul-B450M:~/Documents/gnss-sdr$ gnss-sdr --config_file wamb_GPS_GAL_L2.conf 
Initializing GNSS-SDR v0.0.19.git-main-87fcfd237 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
device address: ip:192.168.2.1
frequency : 1575420000 Hz
sample rate: 3072000 Sps
gain mode: slow_attack
item type: gr_complex
RF Channels: 1
Segmentation fault (core dumped)
```
My system is Ubunto 22.04. I initially used the packaged Gnuradio 3.10.1.1 that came with the distribution and built gnss-sdr. I then built  Gnuradio 3.10.12 from source and a clean rebuild of gnss-sdr and still same seg fault.

The PlutoSDR does work in both versions of Gnuradio. I can make a gnuradio-companion project and capture and display realtime data from the PlutoSDR no problem. It's just when used as a signal source in gnss-sdr there is an issue. 

With the gnuradio 3.10.1.1 & gnss-sdr V0.19 combination the relevant output of gdb is the following
```
[New Thread 0x7ffee1600640 (LWP 56714)]
[New Thread 0x7ffee0c00640 (LWP 56715)]
[Thread 0x7ffee0c00640 (LWP 56715) exited]

Thread 44 "fmcomms2_source" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff67400640 (LWP 56645)]
0x00007ffff6538a68 in ?? () from /lib/x86_64-linux-gnu/libvolk.so.2.5
(gdb) bt
#0  0x00007ffff6538a68 in ?? () from /lib/x86_64-linux-gnu/libvolk.so.2.5
#1  0x00007ffff7cdd73f in ?? () from /lib/x86_64-linux-gnu/libgnuradio-iio.so.3.10.1
#2  0x00007ffff702e1cb in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.1
#3  0x00007ffff6ff8b9b in gr::block_executor::run_one_iteration() ()
   from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.1
#4  0x00007ffff70407eb in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.1
#5  0x00007ffff702ed65 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.1
#6  0x00007ffff7f970cb in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.74.0
#7  0x00007ffff5494ac3 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:442
#8  0x00007ffff5526850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```
With the gnuradio 3.10.12 & gnss-sdr V0.19 combination the relevant output of gdb is
```
[New Thread 0x7ffedf400640 (LWP 113428)]
[New Thread 0x7ffedea00640 (LWP 113429)]
[Thread 0x7ffedea00640 (LWP 113429) exited]

Thread 33 "fmcomms2_source" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff6e000640 (LWP 113358)]
0x00007ffff64832e0 in volk_16i_s32f_convert_32f_u_avx2 () from /usr/local/lib/libvolk.so.3.1
(gdb) bt
#0  0x00007ffff64832e0 in volk_16i_s32f_convert_32f_u_avx2 () from /usr/local/lib/libvolk.so.3.1
#1  0x00007ffff7cb7e07 in gr::iio::fmcomms2_source_impl<std::complex<float> >::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /usr/local/lib/libgnuradio-iio.so.3.11.0git
#2  0x00007ffff705e3eb in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /usr/local/lib/libgnuradio-runtime.so.3.11.0git
#3  0x00007ffff70019bd in gr::block_executor::run_one_iteration() ()
   from /usr/local/lib/libgnuradio-runtime.so.3.11.0git
#4  0x00007ffff7072fa5 in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) () from /usr/local/lib/libgnuradio-runtime.so.3.11.0git
#5  0x00007ffff705c21c in gr::tpb_container::operator()() ()
   from /usr/local/lib/libgnuradio-runtime.so.3.11.0git
#6  0x00007ffff705ccf4 in std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) () from /usr/local/lib/libgnuradio-runtime.so.3.11.0git
#7  0x00007ffff7f930cb in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.74.0
#8  0x00007ffff5294ac3 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:442
#9  0x00007ffff5326850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) 
```

The odd thing in both is the reference to fmcomms2_source. I do not use this source and I didn't build gnss-sdr with this option enabled? Is there possibly some mapping problem where PlutoSDR hardware gets confused as a fmcomms2_source hardware. (I did try building with the fmcomms ON and the AD9361 ON but that made no difference)

I'm confident the config files are correct. I used exactly the same ones previously on my older computer with the same PlutoSDR and earlier versions of gnuradio and gnss-sdr, but that computer died and I'm trying to get going again on a new system.

sdy623, 2024-10-28:
> I can confirm that this error still seems to be in the V 0.19. I too get the seg fault on trying to use the PlutoSDR signal source. My error is identical gives the following. Has been built with [...]

This is the code of ffcomms2 
https://github.com/analogdevicesinc/linux/blob/main/arch/arm/boot/dts/adi-fmcomms2.dtsi