gnss-sdr/gnss-sdr #750: Fifo with rtl

Koreko209, 2024-01-02:
I want to receive gps dat by rtl sdr using python and send the sample to fifo unix file and get the location by gnss sdr i use the configuration that is fix to me by rtl direct to gnss sdr but change the signal soursce to get dat from the fifo file with data type gr complex i get tracking but it loss frequrently and dont get the location.
Can any one help me
Thanks

wkazubski, 2024-01-14:
Simple way to create python script for saving data from rtl-sdr to file is to use GNU Radio. Just create a flowgraph with two blocks, the rtl-sdr source and the file sink, set some options as RF frequency and sample rate and run it. GNU Radio compiles the flowgraph to a pyton script that can be used as a standalone program.

carlesfernandez, 2024-08-09:
Please feel free to open a new issue with more details about what's your intent and how are you trying to implement it. Thanks.