gnss-sdr/gnss-sdr #752: Impossible to build on MAC Sonoma 14.1.1

SoniaKaudjhis, 2024-01-08:
Hi I try to build gnss-sdr on Mab Sonoma 14.1.1 I have the following issue when it try to install libraries : 
`[ 55%] No install step for 'protobuf-25.0'
[ 55%] Completed 'protobuf-25.0'
[ 55%] Built target protobuf-25.0`
Can you help me with that issue ? 
I notice also there is a missing libraries in opt/local/include/gnuradio/types
[image]
Thanks for your help.

jwmelto, 2024-01-08:
Can you post the entire output of your build (not a screen shot)?

SoniaKaudjhis, 2024-01-08:
Here is my build output : 
[build_ouput.txt](https://github.com/gnss-sdr/gnss-sdr/files/13855444/build_ouput.txt)

jwmelto, 2024-01-08:
What version of GNU Radio do you have and how was it installed? The commit that replaced `boost::shared_ptr` with `std::shared_ptr` was 4 years ago.

What version of GNSS-SDR do you have? I can't find the commit hash (`v0.0.18.git-next-8e1a7ce78`) you built in the next branch.

You also have a very new version of Boost (where, apparently, they removed the file `boost/shared_ptr.hpp`).

In my (limited) experience, if you are going to build from source, you commit to building all the dependencies from source as well. This appears to be a case where your Boost is too new for your version of GNU Radio.

jwmelto, 2024-01-08:
No, your GNU Radio is too old. https://formulae.brew.sh/formula/gnuradio says that it installs GNU Radio 3.10 (which is substantially different from 3.8).

If you want to use an older GNU Radio, you will need an older Boost.

SoniaKaudjhis, 2024-01-08:
@jwmelto which version on GNU Radio which has acceptable for make it work ?

SoniaKaudjhis, 2024-01-15:
@jwmelto I can build without issue. Thanks for your help

SoniaKaudjhis, 2024-01-15:
This issue is resolved

closed 2024-01-15 by SoniaKaudjhis