gnss-sdr/gnss-sdr #754: 'msg_connect' error during make

brendan-mills, 2024-01-16:
I am trying to compile and run gnss-sdr on my MacbookPro (M2) running MacOS Sonoma 14.1.2. I have been using homebrew as a package manager and I was able to run the cmake step with some minimal compatibility fixes. 

I tried running 'make' as a part of the installation process and I ran into an error during the build for `src/algorithms/channel/adapters/CMakeFiles/channel_adapters.dir/channel.cc` 

The error reads:
`.../gnss-sdr/src/algorithms/channel/adapters/channel.cc:140:24: error: no matching member function for call to 'msg_connect' `
            `top_block->msg_connect(acq_->get_right_block(), pmt::mp("events"), channel_msg_rx_, pmt::mp("events"));`

`/opt/homebrew/include/gnuradio/hier_block2.h:103:10: note: candidate function not viable: no known conversion from 'channel_msg_receiver_cc_sptr' (aka 'shared_ptr<channel_msg_receiver_cc>') to 'basic_block_sptr' (aka 'shared_ptr<basic_block>') for 3rd argument`
    `void msg_connect(basic_block_sptr src,`

`/opt/homebrew/include/gnuradio/hier_block2.h:107:10: note: candidate function not viable: no known conversion from 'pmt_t' (aka 'shared_ptr<pmt_base>') to 'std::string' (aka 'basic_string<char>') for 2nd argument`
    `void msg_connect(basic_block_sptr src,`

There were 4 of these errors present in the file which halted the compiling process. 

I have made sure my gnuradio package is up to date (3.10.7.0). And I have recently run `brew upgrade`
I have been able to manage libraties with relative success but editing source code on the fly is a little too far out of my comfort zone. I don't have any experience developing in C++ so if I am missing something obvious I would be grateful if you point me in the direction of some relevant resources. 

Any help would be appreciated

carlesfernandez, 2024-08-30:
This OS is continually tested in our CI. What GNSS-SDR version/commit are you working with?

LRParser, 2024-09-01:
I get the same error on OS X 14.4.1 using a M2 processing on both the main and the next branches as of today, when running CMake with `-DENABLE_OSMOSDR=ON`

```
-- GNSS-SDR v0.0.19.git-next-1088387f7 is ready to be built.
```
And
```
-- GNSS-SDR v0.0.19.git-main-87fcfd237 is ready to be built.
```