gnss-sdr/gnss-sdr #755: build of v0.0.19 fails with gnuradio v3.10.9.2

aardric, 2024-01-25:
Build configuration appears successful but build fails; any suggestions on where to go with this?

gnss-sdr: version v0.0.19
commit 9fd1c4ea0ff3bfae3498cf18fb267cec87a316ea

gnuradio:  v3.10.9.2
commit 2fa48b16d924a209633ecfa073e01863f87b646c

Below is a console excerpt following   make -j4':

`/usr/include/c++/9/bits/shared_ptr.h:349:2: note:   template argument deduction/substitution failed:
/ardisk/aarx/Aardvark/projects/AAVKX/sdr/gnss-sdr/src/algorithms/resampler/adapters/direct_resampler_conditioner.cc:78:96: note:   ‘direct_resampler_conditioner_cb_sptr’ {aka ‘boost::shared_ptr<direct_resampler_conditioner_cb>’} is not derived from ‘std::unique_ptr<_Tp, _Dp>’
   78 |             resampler_ = direct_resampler_make_conditioner_cb(sample_freq_in_, sample_freq_out_);
      |                                                                                                ^
[ 52%] Building CXX object src/algorithms/conditioner/adapters/CMakeFiles/conditioner_adapters.dir/signal_conditioner.cc.o
make[2]: *** [src/algorithms/resampler/adapters/CMakeFiles/resampler_adapters.dir/build.make:76: src/algorithms/resampler/adapters/CMakeFiles/resampler_adapters.dir/direct_resampler_conditioner.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1888: src/algorithms/resampler/adapters/CMakeFiles/resampler_adapters.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....`

jwmelto, 2024-01-26:
There is something wrong with your environment. The full output from CMake is needed to be sure, but [this code](https://github.com/gnss-sdr/gnss-sdr/blob/c94b2ddfe8668716cba191aa4ddae90e1d2cda32/cmake/Modules/FindGNURADIO.cmake#L278-L287) shows how `GNURADIO_USES_STD_POINTERS` is defined. Based on a define (which, for the life of me, I can't ferret out how it is set here) the definition of `gnss_shared_ptr` is either `std::shared_ptr` or `boost::shared_ptr`. In your compile, it is the latter. That is the source of your error.

The first simple step is to completely delete the contents of the `build` directory and build from scratch.

Look at `build/src/algorithms/resampler/adapters/CMakeFiles/resampler_adapters.dir/flags.make`. Mine includes this:
```
# compile CXX with /usr/bin/c++
CXX_DEFINES = -DBOOST_ALL_NO_LIB -DFMT_LOCALE -DFMT_SHARED -DGNURADIO_USES_STD_POINTERS=1 -DGR_GREATER_38=1 -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_LARGE_FILES
```
That `-DGNURADIO_USES_STD_POINTERS=1` is what makes it compile

aardric, 2024-01-26:
I agree that something is wrong with my environment. Please do not put 
any more effort into diagnosing this. I will investigate my gnuradio 
installation and then follow your advice. Thank you very much for this.

aardric, 2024-01-29:
After fixing some (I thought) unrelated issues in gnuradio and rebuilding both gnuradio and gnss-sdr, both appear to be working.  All good for now.

closed 2024-01-30 by aardric