gnss-sdr/gnss-sdr #760: Segmentation Fault using PlutoSDR on Ubuntu 20.04.6

acochrane, 2024-02-21:
This appears to be related to issue #749 as a segfault for use of plutosdr, but based on the backtrace it seems like a different issue.

Using the supplied conf/RealTime_input/gnss-sdr_GPS_L1_plutosdr_realtime.conf after setting the correct address, I receive a segfault.

> $ gnss-sdr --config_file=gnss-sdr_GPS_L1_plutosdr_realtime_new.conf
Initializing GNSS-SDR v0.0.19.git-next-fbca34a4b ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
device address: 192.168.55.1
frequency : 1575420000 Hz
sample rate: 4000000 Sps
gain mode: slow_attack
item type: gr_complex
RF Channels: 1
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
Segmentation fault (core dumped)

Collecting backtrace with gdb
> [Switching to Thread 0x7fff9f7fe700 (LWP 1238238)]
0x00007ffff7000526 in _mm_loadu_si128 (__P=0x31)
    at /usr/lib/gcc/x86_64-linux-gnu/9/include/emmintrin.h:703
703	  return *__P;
(gdb) bt
#0  0x00007ffff7000526 in _mm_loadu_si128 (__P=0x31)
    at /usr/lib/gcc/x86_64-linux-gnu/9/include/emmintrin.h:703
#1  volk_16i_s32f_convert_32f_u_avx2 (outputVector=0x555556f7ba80, inputVector=0x31, 
    scalar=2048, num_points=4096)
    at /home/acochrane/src/volk/kernels/volk/volk_16i_s32f_convert_32f.h:68
#2  0x00007ffff6e11ba3 in __volk_16i_s32f_convert_32f_d (outputVector=0x555556f7ba80, 
    inputVector=0x31, scalar=2048, num_points=4096)
    at /home/acochrane/src/volk/build/lib/volk.c:630
#3  0x00007ffff7cbdef7 in gr::iio::fmcomms2_source_impl<std::complex<float> >::work (
    this=0x555556f576b0, noutput_items=4096, 
    input_items=std::vector of length 0, capacity 0, 
    output_items=std::vector of length 1, capacity 1 = {...})
    at /usr/include/c++/9/bits/stl_vector.h:1040
#4  0x00007ffff7a7579b in gr::sync_block::general_work (this=0x555556f57898, 
    noutput_items=<optimized out>, ninput_items=..., input_items=..., output_items=...)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/sync_block.cc:49
#5  0x00007ffff7a1f771 in gr::block_executor::run_one_iteration (
    this=this@entry=0x7fff9f7df010)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/block_executor.cc:622
#6  0x00007ffff7a8825b in gr::tpb_thread_body::tpb_thread_body (this=<optimized out>, 
    block=..., start_sync=..., max_noutput_items=<optimized out>)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:94
#7  0x00007ffff7a73a0c in gr::tpb_container::operator() (this=this@entry=0x55555bc708c0)
    at /usr/include/c++/9/ext/atomicity.h:96
#8  0x00007ffff7a74977 in gr::thread::thread_body_wrapper<gr::tpb_container>::operator() (
    this=0x55555bc708c0)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/../include/gnuradio/thread/thread_body_wrapper.h:43
#9  std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) (__functor=...) at /usr/include/c++/9/bits/std_function.h:300
#10 0x00007ffff7f4843b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#11 0x00007ffff4964609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#12 0x00007ffff4889353 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

I could be reading this wrong, but it looks like the fmcomms2_source_impl is sending a bad address for the input vector to volk16_16i_s32f_d.

Any hints?

acochrane, 2024-02-21:
After recompiling libiio and gnuradio in Debug build types, I get better info from the backtrace

> 0x00007ffff7c746d7 in iio_channel_is_enabled (chn=0x555556f489f0)
    at /home/acochrane/src/libiio/channel.c:446
446		return chn->index >= 0 && chn->dev->mask &&
(gdb) bt
#0  0x00007ffff7c746d7 in iio_channel_is_enabled (chn=0x555556f489f0)
    at /home/acochrane/src/libiio/channel.c:446
#1  0x00007ffff7c79f20 in iio_buffer_first (buffer=0x7fff98001040, chn=0x555556f489f0)
    at /home/acochrane/src/libiio/buffer.c:258
#2  0x00007ffff7cdae09 in gr::iio::device_source_impl::channel_read (
    this=this@entry=0x555556f14198, chn=0x555556f489f0, dst=0x21, len=len@entry=8192)
    at /home/acochrane/src/gnuradio/gr-iio/lib/device_source_impl.cc:260
#3  0x00007ffff7cdb61e in gr::iio::device_source_impl::work (
    this=this@entry=0x555556f14198, noutput_items=noutput_items@entry=4096, input_items=
    std::vector of length 0, capacity 0, 
    output_items=std::vector of length 2, capacity 2 = {...})
    at /usr/include/c++/9/bits/stl_vector.h:1040
#4  0x00007ffff7ce7e7d in gr::iio::fmcomms2_source_impl<std::complex<float> >::work (
    this=0x555556f14190, noutput_items=4096, 
    input_items=std::vector of length 0, capacity 0, 
    output_items=std::vector of length 1, capacity 1 = {...})
    at /home/acochrane/src/gnuradio/gr-iio/lib/fmcomms2_source_impl.cc:230
#5  0x00007ffff7a7579b in gr::sync_block::general_work (this=0x555556f14378, 
    noutput_items=<optimized out>, ninput_items=..., input_items=..., output_items=...)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/sync_block.cc:49
#6  0x00007ffff7a1f771 in gr::block_executor::run_one_iteration (
    this=this@entry=0x7fffa77df010)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/block_executor.cc:622
#7  0x00007ffff7a8825b in gr::tpb_thread_body::tpb_thread_body (this=<optimized out>, 
    block=..., start_sync=..., max_noutput_items=<optimized out>)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:94
#8  0x00007ffff7a73a0c in gr::tpb_container::operator() (this=this@entry=0x55555bc2cd60)
    at /usr/include/c++/9/ext/atomicity.h:96
#9  0x00007ffff7a74977 in gr::thread::thread_body_wrapper<gr::tpb_container>::operator() (
    this=0x55555bc2cd60)
    at /home/acochrane/src/gnuradio/gnuradio-runtime/lib/../include/gnuradio/thread/thread_body_wrapper.h:43
#10 std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) (__functor=...) at /usr/include/c++/9/bits/std_function.h:300
#11 0x00007ffff7f4843b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#12 0x00007ffff4964609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#13 0x00007ffff4889353 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

It looks like gr::iio::device_source_impl::channel_read might be passed a bad address for dst.

acochrane, 2024-02-22:
Well based this snippet from gnuradio/gr-iio/lib/device_source_impl.cc:309

    // Process samples
    unsigned long items = std::min(items_in_buffer, (unsigned long)noutput_items);

    for (size_t i = 0; i < output_items.size(); i++)
        channel_read(channel_list[i], output_items[i], items * sizeof(short));

    items_in_buffer -= items;
    byte_offset += items * iio_buffer_step(buf); 

It looks like that third argument to channel_read(), dst **should** be a small hex value, counting the smaller of the items_in_buffer or noutput_items.

aihamnedal, 2024-06-20:
Has anyone been able to resolve this? I am running into this exact issue as well

carlesfernandez, 2024-09-13:
Can you execute `ldd gnss-sdr` so we know exactly what library versions are you working with? We will try to reproduce that environment. In newest versions, it seems to work fine.

wkazubski, 2024-09-24:
Check the status of PlutoSDR in your GNU Radio. If it is not enabled, the gnss-sdr may crash if you try to use it with Pluto.

djObsidian, 2024-12-14:
PlutoSDR is working fine in gnuradio for me, that shouldn't be the case here
[image]
Same with using FMComms block, no crash.
[image]

My stacktrace looks similar:
```Thread 23 "fmcomms2_source" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd8ff96c0 (LWP 9285)]
0x00007ffff71a2370 in iio_channel_is_enabled (chn=chn@entry=0x55555632ee10) at /home/obsidian/libiio/channel.c:446
446             return chn->index >= 0 && chn->dev->mask &&
(gdb) bt
#0  0x00007ffff71a2370 in iio_channel_is_enabled (chn=chn@entry=0x55555632ee10) at /home/obsidian/libiio/channel.c:446
#1  0x00007ffff71a5f60 in iio_buffer_first (buffer=0x7fffbc001030, chn=0x55555632ee10) at /home/obsidian/libiio/buffer.c:258
#2  0x00007ffff716927f in ?? () from /lib/x86_64-linux-gnu/libgnuradio-iio.so.3.10.9
#3  0x00007ffff7172e0e in ?? () from /lib/x86_64-linux-gnu/libgnuradio-iio.so.3.10.9
#4  0x00007ffff6a28ca0 in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#5  0x00007ffff69ee98c in gr::block_executor::run_one_iteration() () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#6  0x00007ffff6a38fa8 in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) ()
   from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#7  0x00007ffff6a2a93b in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#8  0x00007ffff6a2a9e4 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#9  0x00007ffff7fa499f in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.83.0
#10 0x00007ffff5316a94 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#11 0x00007ffff53a3c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

djObsidian, 2024-12-14:
No surprise, changing `Plutosdr_Signal_Source` to `Fmcomms2_Signal_Source` yields exactly the same result:
```
Thread 23 "fmcomms2_source" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd8ff96c0 (LWP 10605)]
0x00007ffff71a2370 in iio_channel_is_enabled (chn=chn@entry=0x55555633eb90) at /home/obsidian/libiio/channel.c:446
446             return chn->index >= 0 && chn->dev->mask &&
(gdb) bt
#0  0x00007ffff71a2370 in iio_channel_is_enabled (chn=chn@entry=0x55555633eb90) at /home/obsidian/libiio/channel.c:446
#1  0x00007ffff71a5f60 in iio_buffer_first (buffer=0x7fffbc001090, chn=0x55555633eb90) at /home/obsidian/libiio/buffer.c:258
#2  0x00007ffff716927f in ?? () from /lib/x86_64-linux-gnu/libgnuradio-iio.so.3.10.9
#3  0x00007ffff7172e0e in ?? () from /lib/x86_64-linux-gnu/libgnuradio-iio.so.3.10.9
#4  0x00007ffff6a28ca0 in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#5  0x00007ffff69ee98c in gr::block_executor::run_one_iteration() () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#6  0x00007ffff6a38fa8 in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) ()
   from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#7  0x00007ffff6a2a93b in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#8  0x00007ffff6a2a9e4 in ?? () from /lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.9
#9  0x00007ffff7fa499f in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.83.0
#10 0x00007ffff5316a94 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#11 0x00007ffff53a3c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```