gnss-sdr/gnss-sdr #762: Build error in

arch-dawson, 2024-03-22:
When building, I am encountering this error. I am running arch linux and followed the instructions in the readme. 

```
[ 81%] Building CXX object src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/uhd_signal_source.cc.o
In file included from /usr/local/include/uhd/device.hpp:11,
                 from /usr/local/include/uhd/usrp/multi_usrp.hpp:28,
                 from /usr/include/gnuradio/uhd/usrp_block.h:17,
                 from /usr/include/gnuradio/uhd/usrp_source.h:14,
                 from /home/dawson/tmp/gnss-sdr-git/src/algorithms/signal_source/adapters/uhd_signal_source.h:24,
                 from /home/dawson/tmp/gnss-sdr-git/src/algorithms/signal_source/adapters/uhd_signal_source.cc:17:
/usr/local/include/uhd/property_tree.hpp:75:13: error: template-id not allowed for destructor
   75 |     virtual ~property<T>(void) = 0;
      |             ^
/usr/local/include/uhd/property_tree.hpp:186:1: error: no declaration matches ‘uhd::property<T>::~property()’
  186 | property<T>::~property(void)
      | ^~~~~~~~~~~
/usr/local/include/uhd/property_tree.hpp:186:1: note: no functions named ‘uhd::property<T>::~property()’
/usr/local/include/uhd/property_tree.hpp:68:7: note: ‘class uhd::property<T>’ defined here
   68 | class property : uhd::noncopyable
      |       ^~~~~~~~
In file included from /usr/local/include/uhd/property_tree.hpp:263:
/usr/local/include/uhd/property_tree.ipp:24:50: error: expected ‘)’ before ‘mode’
   24 |     property_impl<T>(property_tree::coerce_mode_t mode) : _coerce_mode(mode)
      |                     ~                            ^~~~~
      |                                                  )
/usr/local/include/uhd/property_tree.ipp:31:5: error: template-id not allowed for destructor
   31 |     ~property_impl<T>(void)
      |     ^
make[2]: *** [src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/build.make:300: src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/uhd_signal_source.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1960: src/algorithms/signal_source/adapters/CMakeFiles/signal_source_adapters.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

jwmelto, 2024-03-22:
This looks like a UHD issue, not an GNSS-SDR issue. Our code is just including theirs.

jwmelto, 2024-03-25:
To clarify, as of C++20, the UHD code is invalid. See https://stackoverflow.com/a/65898493/9220132.

I would look for a newer UHD implementation, or downgrade to C++17.

closed 2024-08-09 by carlesfernandez